Marus/cortex-debug #921: Copy from memory view is in base64 and right click copy does not work

jnz86, 2023-08-09:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

**Describe the bug**
I highlight bytes. I use `CTRL-C`. The result in Base64 which is not helpful at all. Most hex viewers would copy to two digit hex.

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug
2. Set a watch to a pointer
3. Click the "VIEW BINARY DATA" box, opens memory.bin hex view with data inspector
4. `CTRL-C` Copy bytes, not HEX as expected but rather BASE64
5. Copy using right click seems to copy from another panel entirely

**Expected behavior**

`00 11 22 33 ...`

**Screenshots**
[image]

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension) `v1.12.0`
 - OS: `Windows 11`
 - GDB Version: `7.6.1`
 - Compiler Toolchain Version: `arm-none-eabi v12.3`

**Please include `launch.json`**
```
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug Microcontroller (JLink)",
            "cwd": "${workspaceRoot}",
            "executable": "./build/application_debug/application.elf",
            "armToolchainPath": "C:/Progra~2/Arm GNU Toolchain arm-none-eabi/12.3 rel1/bin",
            "servertype": "jlink",
            "serverpath": "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe",
            "interface": "swd",
            "device": "STM32F4",
            "runToEntryPoint": "main",
            "rtos": "FreeRTOS"
        },
```

**Additional context**
To be fair... I have the same right click problem with memory-view extension. But the base64 seems to be unique here.

haneefdm, 2023-08-10:
Sorry, wrong extension. That is a feature of Microsoft VSCode. It works across languages and we have nothing to do with that.

https://www.youtube.com/watch?v=biou30QKhUM

Also, they refuse to launch our extension or be able to customize which editor they launch. They hardcoded it to the Microsoft "Hex Editor".

closed 2023-08-10 by haneefdm

jnz86, 2023-08-10:
Sorry! I thought there was a chance it wasn't yours but I couldn't really tell where that window was coming from. 

>Also, they refuse to launch our extension or be able to customize which editor they launch.

Well that was my next question! Where would I request that you think? You have an issue open somewhere I could tag on to?