Marus/cortex-debug #922: GDB executable mangled when using arm-none-eabi-gdb-py

wes8ty, 2023-08-11:
**Describe the bug**
Trying to change the `gdbPath` to be `arm-none-eabi-gdb-py` instead of `arm-none-eabi-gdb` so I can use the python scripting for GDB. Looked at the [debug attributes](https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md) I could not find a variable that would allow me to change the gdb executable without affecting all the other tool path names.

For some reason by changing the executable to `arm-none-eabi-gdb-py` it mangles the other tool names with `-py`. Not sure if I am doing something wrong. Appreciate any insight.

**To Reproduce**
1. Use attached config file to start a debug session
2. View log

**Expected behavior**

I expect all symbols to be loaded properly with gdbPath to be `arm-none-eabi-gdb-py`.

**Environment**

 - Cortex-Debug Version [v1.12.0]
 - OS: [e.g. Linux Ubuntu 22.04 LTS]
 - GDB Version: [GNU gdb (GNU Tools for STM32 11.3.rel1.20230519-1941) 12.1.90.20220802-git]

**`Launch.json`**
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug Microcontroller - STLink-V2",
            "cwd": "${workspaceFolder}",        // Path from where commands are executed
            "type": "cortex-debug",             // Debug 
            "executable": "${command:cmake.launchTargetPath}", //or fixed file path: build/project-name.elf
            "request": "launch",                // Use "attach" to connect to target w/o elf download
            "servertype": "stlink",             // Use stlink setup of cortex-M debug
            "device": "STM32G473VEH6",          // MCU used
            "interface": "swd",                 // Interface setup
            "runToEntryPoint": "main",          // Run to main and stop there
            "svdFile": "STM32G40x.svd",         // SVD file to see registers
            "v1": false,
            "gdbPath": "arm-none-eabi-gdb-py",
            "showDevDebugOutput": "raw",     // Debug output
            "preLaunchCommands": [
                "-enable-pretty-printing",
            ]
        }
    ]
}
```

**`Debug Console` with gdbPath change**
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump-py --syms -C -h -w /workspaces/g4/build/debug/g4-g473vet.elf
Reading symbols from arm-none-eabi-nm-py --defined-only -S -l -C -p /workspaces/g4/build/debug/g4-g473vet.elf
Launching GDB: arm-none-eabi-gdb-py -q --interpreter=mi2
1-gdb-version
Error: arm-none-eabi-nm-py failed! statics/global/functions may not be properly classified: Error: spawn arm-none-eabi-nm-py ENOENT
    Expecting `nm` next to `objdump`. If that is not the problem please report this.
Error: objdump failed! statics/globals/functions may not be properly classified: Error: spawn arm-none-eabi-objdump-py ENOENT    ENOENT means program not found. If that is not the issue, please report this problem.Setting GDB-Server CWD: /opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.1.0.202305091550/tools/bin
Launching gdb-server: /opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.1.0.202305091550/tools/bin/ST-LINK_gdbserver -p 50000 -cp /opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.linux64_2.1.0.202305091550/tools/bin --swd --halt
    Please check TERMINAL tab (gdb-server) for output from /opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.1.0.202305091550/tools/bin/ST-LINK_gdbserver
Finished reading symbols from objdump: Time: 19 ms
Finished reading symbols from nm: Time: 17 ms
```

**`Debug Console` without gdbPath change**
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /workspaces/build/debug/g4-g473vet.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /workspaces/build/debug/g4-g473vet.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Setting GDB-Server CWD: /opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.1.0.202305091550/tools/bin
Launching gdb-server: /opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.1.0.202305091550/tools/bin/ST-LINK_gdbserver -p 50000 -cp /opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.linux64_2.1.0.202305091550/tools/bin --swd --halt
    Please check TERMINAL tab (gdb-server) for output from /opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.1.0.202305091550/tools/bin/ST-LINK_gdbserver
Finished reading symbols from objdump: Time: 24 ms
```

haneefdm, 2023-08-11:
> I expect all symbols to be loaded properly with gdbPath to be arm-none-eabi-gdb-py.

Why did you expect that? Need `objdump` and `nm` to also conform to the same style as gdbPath or you can control those via other methods. Look at the Debug Console, we are having issues with those two programs. What do all three paths look like (gdb, objdump, and nm)?

```
Error: arm-none-eabi-nm-py failed! statics/global/functions may not be properly classified: Error: spawn arm-none-eabi-nm-py ENOENT
    Expecting `nm` next to `objdump`. If that is not the problem please report this.
Error: objdump failed! statics/globals/functions may not be properly classified: Error: spawn arm-none-eabi-objdump-py ENOENT    ENOENT means program not found. If that is not the issue, please report this problem.
```

wes8ty, 2023-08-11:
Thanks @haneefdm. I corrected it by adding `"objdumpPath": "arm-none-eabi-objdump"` to my `launch.json` file.

The output is:
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /workspaces/g4/build/debug/g4-g473vet.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /workspaces/g4/build/debug/g4-g473vet.elf
Launching GDB: arm-none-eabi-gdb-py -q --interpreter=mi2 -iex "set auto-load safe-path /"
```

closed 2023-08-11 by wes8ty