Marus/cortex-debug #923: Local variables with same between functions not tracking or updating context

jnz86, 2023-08-15:
**Describe the bug**
The function I am in defines a local `uint32_t status`. The function I call does the same, inside this function the debugger does not update the address of the new local with the same name. Nothing in the watch window is correct for the new-local, everything references local variable one jump/branch back. Hover over the variable works as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. Defined this:

```
void test_1()
{
    uint32_t status = 0xBEE7BEE7;

    status++;
    if (status == 0x00)         //just nonsense to set a breakpoint on if needed
        return;
}

void test_2()
{
    uint32_t status = 0xC0D3C0D3;
    uint32_t statuA = 0xAAAAAAAA;

    status--;
    if (status == 0x00)
        return;
}

my_function()
{
    uint32_t status = 0xDEADDEAD;

    test_1();
    test_2();
...
}   
```
3. Entering `my_function()` I set `status` and it gets an example address: `0x2400B32C`
4. Entering `test_1()` or `test_2()`, the watch and variables window continues to show `status` value and address as if it's the one from `my_function()`, same address and values.
5. The hover-over shows the context correct version of the local `status`.

**Expected behavior**
That `status` is reflected on it's local state as functions are entered and returned from. 

**Screenshots**

[image]
Status as defined as the local correctly.

[image]
Despite having called `test_1()`, returned, and currently inside `test_2()`, the debug window is still pointing to the original definition of `status`. Notice however that `statuA` works as expected. It's unique name isn't confused when moving around, returning from the function makes the watch window lose context as expected.

**Environment (please complete the following information):**

 - Cortex-Debug Version 1.12.0
 - OS: Windows 11
 - GDB Version: JLink 7.66 & 7.92 (tried separately)
 - Compiler Toolchain Version: arm-none-eabi v11.3 & v12.3

**Launch.json**
```
"configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug Microcontroller (JLink)",
            "cwd": "${workspaceRoot}",
            "executable": "./build/application_debug/application.elf",
            "armToolchainPath": "C:/Progra~2/Arm GNU Toolchain arm-none-eabi/12.3 rel1/bin",
            "servertype": "jlink",
            "serverpath": "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe",
            "interface": "swd",
            "device": "STM32H743VI",
            "runToEntryPoint": "main",
            "rtos": "FreeRTOS",
        },
```

**Additional context**

* `status` is not static or global anywhere
* All code runs as expected on the micro, and hover-over works. 
* I have tried multiple JLinks
* Using FreeRTOS, and stack does not seem to be an issue, plenty of space remaining
* The memory views when pointing to the old context look as expected, and same for the new locations (usually found by looking at the dasm and registers). It's just that the GUI links all point to the original local.
* Optimization is off, volatile does nothing.
* I am not in an ISR context
* This seems to be a recent issue, but I can't say how long with certainty. I recently added `-D_POSIX_C_SOURCE=200809L` to my makefile for `strnlen()` so that's my next check (That's how few options I have left! pleasesendhelp)

haneefdm, 2023-08-25:
Maybe I fixed it. It was hard to reproduce it because all the code above would get optimized out. I created a test case nevertheless and could see the problem

GDB was giving us stale values or said nothing changed when in fact it had. It was looking at values from the previous stack traces and not getting updated for the new stack traces. Very difficult to explain We now force gdb to re-evaluate

Can you please try this out and give me feedback. Pick up the latest from https://github.com/Marus/cortex-debug/releases

and Install from the VSIX file manually. Note that there are other unrelated changes.

jnz86, 2023-08-26:
Thanks Haneef! 

Unfortunately same issue. 

I did double check that I'm using 1.12.0.Pre2.

haneefdm, 2023-08-26:
Ok, that's progress. Trying to reproduce the problem...

haneefdm, 2023-08-26:
Can you try the latest pre-release please. I was able to duplicate and fix the issue. I have no idea if there was a change in gdb that is causing this of if this was always a problem from 2+ years ago.

 https://github.com/Marus/cortex-debug/releases

jnz86, 2023-08-26:
Thanks H!

I’d love to test it, but I’m going to be out of office for the week. I’ll be able to test the day I get back. I’ll won’t abandon this.

I had wondered about when this became a problem. I use status in most of my functions and single return call, so I really assume I would have seen this before! But, you never know.

grum-kin, 2023-08-30:
Experiencing the same issue here.

Can confirm that using the gdb "print" command in the Debug Console outputs the expected value, while the Variables window shows values from previous function calls. Seems to only happen with variables that share the same name.