Marus/cortex-debug #924: Syncing the extension via Git causes "args" of class GDBDebugSession to be undefined

Delzye, 2023-08-16:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

**Describe the bug**
I have a VSCode Installation I'm distributing via Git for work. When copying the vscode folder to different machines, everything works fine. When using a Git repository and cloning that to any machine, I first get:

"Cortex-Debug: VSCode debugger extension version undefined git(d7a405a). <...>".

Then it complains about not finding "undefined-gdb.exe". If I solve that by explicitly setting the gdbPath (The version distributed via Git only sets armToolchainPath), it complains with "GDB Server Console tcp port is undefined.".

All of these issues are in some way related to the 'args' parameter of GDBDebugSession (args.pvtVersion, args.toolchainPath, args.gdbServerConsolePort).

**To Reproduce**
Steps to reproduce the behavior:
1. create an empty Git repository (GitLab in my case, because work and stuff)
2. Add a VSCode Install (1.81.0 for me) and a "data" folder (for portability)
3. Start VSCode and install the cortex-debug extension (newest version, 1.12.0)
4. Try debugging (should work)
5. clone the repository to your PC
6. Start the cloned VSCode and try to debug (should fail)

**Expected behavior**

Since I don't see how the arguments being undefined or not is dependent on the fact if the extension is from git or not, I expected this to work.

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension) [1.12.0]
 - OS: [Windows 10]
 - GDB Version: [10.2.90.20210621-git]
 - Compiler Toolchain Version: [10.4]

**Please include `launch.json`**

NOTE: This works in a not-cloned version.

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "STM32F439",
            "cwd": "${workspaceFolder}",        //Path from where commands are executed
            "type": "cortex-debug",             //Debug 
            "executable": "build/MAIN.elf", //or fixed file path: build/project-name.elf
            "request": "attach",                //Use "attach" to connect to target w/o elf download
            "servertype": "stlink",             //Use stlink setup of cortex-M debug
            "gdbTarget": ":50000",
            "device": "STM32F439RE",            //MCU used
            "interface": "swd",                 //Interface setup
            "runToEntryPoint": "main",          //Run to main and stop there
            "v1": false,
            "showDevDebugOutput": "raw",

            /* Will get automatically detected if STM32CubeIDE is installed to default directory
               or it can be manually provided if necessary.. */
            // "serverpath": "C:\\32-bit\\ToolchainLocalTest\\STM32CubeIDE_1.11.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.0.400.202209281104\\tools\\bin\\ST-LINK_gdbserver.exe",
            "serverpath": "C:\\32-bit\\Toolchain\\stlink-gdbserver\\bin\\ST-LINK_gdbserver.exe",
            // "armToolchainPath": "C:\\32\\ToolchainLocalTest\\STM32CubeIDE_1.11.0\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.100.202210260954\\tools\\bin",
            "armToolchainPath": "C:\\32-bit\\Toolchain\\gnu-tools-10.4\\tools\\bin",
            "stm32cubeprogrammer": "c:\\32-bit\\ToolchainLocalTest\\STM32CubeProgrammer_2.12.0\\bin",
            "svdFile": ""

            /* If you use external loader, add additional arguments */
            //"serverArgs": ["--extload", "path/to/ext/loader.stldr"],
        }
    ]
}
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Cortex-Debug: VSCode debugger extension version undefined git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
GDB executable "undefined-gdb.exe" was not found.
Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.
```

**Additional context**
If you need further Information, please ask!

haneefdm, 2023-08-16:
`GDBDebugSession` is like our main Class which implements the Microsoft Debug Protocol and we are passed args by VSCode using the method `initializeRequest`. We don't control that. It is the callers responsibility (VSCode) to pass proper arguments. Not us.

Whatever you are doing, you will want to talk to Microsoft about that. You may have actually broken VSCode. Not our extension.

You may want to try the mock debug extension to debug the whole issue. Use its source code as well.

https://marketplace.visualstudio.com/items?itemName=ms-vscode.mock-debug

closed 2023-08-16 by haneefdm

Delzye, 2023-08-16:
Hi, I just tested older versions of cortex-debug, because I was pretty sure this used to work, and I have no issues on any version between 1.8.0 and 1.10.0. Only 1.12.0 causes this issuse, which is why I don't think this is a VSCode issue (at least not solely).

Could you reopen the issue please?

haneefdm, 2023-08-16:
What is there for me to do? You created a setup that I cannot recreate and don't want to try. I cannot debug my caller as to why it is passing me an invalid `args`. You can build the extension yourself and debug.

Cortex-Debug now has several dependencies. If you did not also install them then the extension is probably not even loading. Look in our package.json...

https://github.com/Marus/cortex-debug/blob/2afa4d903667841ff1fb856a4dd7a858f52ba095/package.json#L3023

When you install via VSCode, it will automatically install all those extensions in that order. Maybe these extensions are missing in your clones.

We even noted this change in https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md#v1120 which people like you should be reading.