Marus/cortex-debug #926: symbolFiles with more than 2 files does not work.

andyinno, 2023-08-21:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

I looked out for something similar but I did not found anything related to it. What I found was just the base that I used for building the initial launch.json with two files attached.

**Describe the bug**
A clear and concise description of what the bug is.

As the title says, if I add more than two symbolFiles, the debugging does not start.

**To Reproduce**
Steps to reproduce the behavior:

- Add LoadFiles section with 2 files
- Add symbolFiles with the exact 2 files wrapped in an object

- Start the debugging and the debugger is active.

- Add another file in the LoadFiles and by pressing F5 the debugging starts.
- Add the same file to the symbolFiles and by pressing F5 the following output is displayed:

```
[2023-08-21T09:47:18.923Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
[2023-08-21T09:47:18.936Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

**Expected behavior**

I expect to be able to use multiple symbolFiles seen that this is an array object.

**Screenshots**

[comment]: <> If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (this extension) [e.g. 0.2.3] v1.12.0
 - OS: [e.g. Linux Ubuntu 18.04 LTS, Windows 11, etc.] Ubuntu 22.04 LTS 
 - GDB Version: [e.g. 1.11.1] 12.1
 - Compiler Toolchain Version: [e.g. arn-none-eabi V 11.1] arm-none-eabi-gcc 11.3.1

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```

{
            "name": "Remote Full",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/Application/first_executable.elf",
            "loadFiles": [
                "${workspaceRoot}/build/first_executable.elf",
                "${workspaceRoot}/build/second_executable.elf",
                "${workspaceRoot}/build/third_executable.elf"
            ],
            "symbolFiles": [
                {"file": "${workspaceRoot}/build/first_executable.elf"},
                {"file": "${workspaceRoot}/build/second_executable.elf"},
                {"file": "${workspaceRoot}/build/third_executable.elf"}
            ],
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "serverpath": "JLinkGDBServerCLExe",
            "ipAddress": "172.17.0.1",
            "device": "STM32F412ZG",
            "svdFile": "${workspaceRoot}/.vscode/STM32F412.svd",
            "runToEntryPoint": "main",
            "showDevDebugOutput": "both",
            "serverArgs": [
                "-rtos",
                "GDBServer/RTOSPlugin_FreeRTOS"
            ],
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                // "clearSearch": false    // OpenOCD users may have to un-comment this
                "decoders": [
                    {
                        "port": 0,
                        "type": "console"
                    }
                ]
            }
        },
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file

Did it again, the output does not change.
```
[2023-08-21T10:03:31.356Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
[2023-08-21T10:03:31.368Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

**Additional context**
Add any other context about the problem here.

haneefdm, 2023-08-21:
> Attach text from Debug Console

I don't see anything from the `Debug Console`.

andyinno, 2023-08-22:
Hello @haneefdm 

Sorry if I was not clear. That's the point. I cannot add anything from the debug console. It is just empty.

```
[2023-08-21T10:03:31.356Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
[2023-08-21T10:03:31.368Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

as you can see, the SERVER CONSOLE DEBUG output is "connected" and then "closed".

If I use 2 elf files in loadFiles and in symbolFiles everything works as expected. As soon as I add a third elf file, the console is empty and nothing happens. The debug does not start.

haneefdm, 2023-08-22:
If the `Debug Console` is empty, then the debugger failed at startup. This is highly unusual, Can you show a screenshot of the debug console? It should look something like...
[image]

Btw, I tried to duplicate it by just added the same file(s) multiple times and I could not duplicate what you are seeing.

andyinno, 2023-08-22:
Thank you for your reply @haneefdm .

is a gif ok?

at the end, I pressed 3 times F5 for starting the debug session...

[image]

The issue is not related to the fact that the file that I am adding is the same, it happens also if I add the bootloader.
In the gif, you can see that 2 times the same file, it works. If I add a third entry, it does not start the debug session.

Now I am trying with openocd as backend. When I stop the debugger it terminates openocd on the external terminal window

haneefdm, 2023-08-22:
Gif is fine. I added the same file(s) as I know it would be harmless and I don't have a project with three valid loadable files.

I have a feeling that you are unable to see the "Debug Console". It is from Microsoft and it exists even without our extension. It is right next to `GITLENS` in your setup ... and you are saying that is Empty?

The screenshot/gif you are showing me is NOT the debug console. It is the output from the gdb-server and that tab is called TERMINAL. Not Debug Console.

> Now I am trying with openocd as backend. When I stop the debugger it terminates openocd on the external terminal window

What is an external terminal window?

andyinno, 2023-08-22:
well... an externam terminal window, is a linux terminal...
[image]

anyway... I did another gif with the debug console opened

haneefdm, 2023-08-22:
That is what I have been asking for all along. The Debug Console. It is definitely not empty.

I think you have a corrupt GDB. Missing libraries, etc. Make sure you can run gdb properly with all its library dependencies fulfilled. Recent versions of GDB releases are incomplete and there is no way for us to tell on Linux. It just hangs or crashes when it is not properly installed configured.

haneefdm, 2023-08-22:
Okay, that is what I have been asking for all along. The Debug Console. Can you copy/paste (or attach a file) with the entire contents of the console.

andyinno, 2023-08-22:
This is what I get with the 3 files enabled:

```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Finished reading symbols from objdump: Time: 55 ms
Finished reading symbols from nm: Time: 53 ms
```

Nothing else.

andyinno, 2023-08-22:
This with only two files enabled:

```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Finished reading symbols from objdump: Time: 63 ms
Finished reading symbols from nm: Time: 64 ms
Finished reading symbols from objdump: Time: 66 ms
Could not find symbol '_SEGGER_RTT' in executable. Make sure you compile/link with debug ON or you can specify your own RTT address
Finished reading symbols from nm: Time: 67 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Tools for STM32 11.3.rel1.20230519-1941) 12.1.90.20220802-git\n"
-> ~"Copyright (C) 2022 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-linux-gnu --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /home/andrea/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /home/andrea/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-interpreter-exec console "add-symbol-file \"/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf\""
-> ~"add symbol table from file \"/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf\"\n"
add symbol table from file "/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf"
-> ~"(y or n) [answered Y; input not from terminal]\n"
(y or n) [answered Y; input not from terminal]
-> ~"Reading symbols from /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf...\n"
Reading symbols from /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/Secure/secure.elf...
-> 10^done
11-interpreter-exec console "add-symbol-file \"/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf\""
-> ~"add symbol table from file \"/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf\"\n"
add symbol table from file "/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf"
-> ~"(y or n) [answered Y; input not from terminal]\n"
(y or n) [answered Y; input not from terminal]
-> ~"Reading symbols from /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf...\n"
Reading symbols from /home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/build/NonSecure/non_secure.elf...
-> 11^done
12-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
[... 73 lines cut ...]
-> &"warning: Source file is more recent than executable.\n"
warning: Source file is more recent than executable.
-> ~"86\t  HAL_Init();\n"
86	  HAL_Init();
-> *stopped,reason="breakpoint-hit",disp="keep",bkptno="1",frame={addr="0x0c0003ac",func="main",args=[],file="/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/Secure/Core/Src/main.c",fullname="/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/Secure/Core/Src/main.c",line="86",arch="armv8-m.main"},thread-id="1",stopped-threads="all"
mi2.status = stopped
-> =breakpoint-deleted,id="4"
25-thread-list-ids
-> 25^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
26-thread-info 1
-> 26^done,threads=[{id="1",target-id="Remote target",frame={level="0",addr="0x0c0003ac",func="main",args=[],file="/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/Secure/Core/Src/main.c",fullname="/home/andrea/STM32CubeIDE/works [... cut ...] -m.main"}]
29-stack-info-frame --thread 1 --frame 0
-> 29^done,frame={level="0",addr="0x0c0003ac",func="main",file="/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/Secure/Core/Src/main.c",fullname="/home/andrea/STM32CubeIDE/workspace_1.9.0a/TrustZone-app/Secure/Core/Src/main.c",line="86",arch="armv8-m.main"}
30-stack-select-frame --thread 1 0
-> 30^done
31-stack-list-variables --thread 1 --frame 0 --simple-values
-> 31^done,variables=[]

```

I had a breakpoint at the beginning of main.

andyinno, 2023-08-23:
Ok, 

I am unsure about what do you mean with 1. If I comment out the second, I have just two files. I already provided a gif with two files (secure / secure) and (secure / not secure). It loads. Maybe I misunderstood your request.

I did the test of swapping file 2 and 3.
I can confirm the same behavior.

Ordering does not matter. (secure.elf / secure.elf / not_secure.elf ) / (not_secure / secure / not_secure)... I tried more or less all the combinations.

If you have a more "verbose" way to print in the debug console I can try if it can provide more info.

haneefdm, 2023-08-23:
I have no other logs. Since I can't reproduce the problem, we are stuck until I can think of something.

Makes my life easier if you can create an example to share. I will need the *exact* elf files and the launch.json. I don't need any source code.

Or, if you have any *real* experience with Typescript, you can debug our source code available in this repo.

haneefdm, 2023-08-23:
There should be a file /tmp/cortex-debug-server-exiting.log. Can you share the last 20-30 lines of that file?

andyinno, 2023-08-23:
I just send a direct email to you with a project. Sorry no Typescript here.

I will check the log ASAP

andyinno, 2023-08-23:
I cleaned the file before the launch. This is what is is created:

not promising I fear...
```
➜  ~ cat /tmp/cortex-debug-server-exiting.log
[2023-08-23T14:11:31.721Z] ppid=810841 pid=810841 remote: ******* Starting new session request type="launch"
[2023-08-23T14:11:31.746Z] ppid=810841 pid=810856 GDB started ppid=810841 pid=810856
➜  ~ 
```

andyinno, 2023-08-23:
Sorry for the notification spamming. I forgot to mention that from the toolchain file I forgot to re-add the following lines:

```
set(CMAKE_EXECUTABLE_SUFFIX_C .elf)
set(CMAKE_EXECUTABLE_SUFFIX_CXX .elf)
set(CMAKE_EXECUTABLE_SUFFIX_ASM .elf)
```

without those, the launch.json and the files generated by cmake are not aligned.

haneefdm, 2023-08-23:
Unfortunately, I cannot duplicate it with your elf files either. I am trying to download the exact version of GDB you have and I am unable to find it for a Mac.

https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

Can you tell which version you are using from here?

This is what I see with your elf files. Of course, it eventually fails because I don't have your board.
```
Cortex-Debug: VSCode debugger extension version 1.12.0 git(d7a405a). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/hdm/Downloads/TrustZone-app_nogit/secure.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/hdm/Downloads/TrustZone-app_nogit/secure.elf
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/hdm/Downloads/TrustZone-app_nogit/non_secure.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/hdm/Downloads/TrustZone-app_nogit/non_secure.elf
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /Users/hdm/Downloads/TrustZone-app_nogit/secure.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /Users/hdm/Downloads/TrustZone-app_nogit/secure.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
Launching gdb-server: /Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.macos64_2.2.300.202301161003/tools/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.600.202303311036/resources/openocd/st_scripts -s /home/andrea/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.1.0.202306221132/resources/openocd/st_scripts -f /Users/hdm/.vscode/extensions/marus25.cortex-debug-1.12.0/support/openocd-helpers.tcl -f /Users/hdm/Downloads/TrustZone-app_nogit/NonSecure/TrustZone-app_NonSecure.cfg -f /Users/hdm/Downloads/TrustZone-app_nogit/NonSecure/TrustZone-app_bind.cfg
    Please check TERMINAL tab (gdb-server) for output from /Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.macos64_2.2.300.202301161003/tools/bin/openocd
Finished reading symbols from objdump: Time: 31 ms
Finished reading symbols from objdump: Time: 37 ms
Finished reading symbols from objdump: Time: 30 ms
Could not find symbol '_SEGGER_RTT' in executable. Make sure you compile/link with debug ON or you can specify your own RTT address
Finished reading symbols from nm: Time: 46 ms
Finished reading symbols from nm: Time: 43 ms
Finished reading symbols from nm: Time: 41 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git\n"
-> ~"Copyright (C) 2021 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"--host=x86_64-apple-darwin10 --target=arm-none-eabi\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /Users/hdm/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /Users/hdm/.vscode/extensions/marus25.cortex-debug-1.12.0/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-interpreter-exec console "add-symbol-file \"/Users/hdm/Downloads/TrustZone-app_nogit/secure.elf\""
-> ~"add symbol table from file \"/Users/hdm/Downloads/TrustZone-app_nogit/secure.elf\"\n"
add symbol table from file "/Users/hdm/Downloads/TrustZone-app_nogit/secure.elf"
-> ~"(y or n) [answered Y; input not from terminal]\n"
(y or n) [answered Y; input not from terminal]
-> ~"Reading symbols from /Users/hdm/Downloads/TrustZone-app_nogit/secure.elf...\n"
Reading symbols from /Users/hdm/Downloads/TrustZone-app_nogit/secure.elf...
-> 10^done
11-interpreter-exec console "add-symbol-file \"/Users/hdm/Downloads/TrustZone-app_nogit/non_secure.elf\""
-> ~"add symbol table from file \"/Users/hdm/Downloads/TrustZone-app_nogit/non_secure.elf\"\n"
add symbol table from file "/Users/hdm/Downloads/TrustZone-app_nogit/non_secure.elf"
-> ~"(y or n) [answered Y; input not from terminal]\n"
[... cut ...]
-> ~"(y or n) [answered Y; input not from terminal]\n"
(y or n) [answered Y; input not from terminal]
-> ~"Reading symbols from /Users/hdm/Downloads/TrustZone-app_nogit/secure.elf...\n"
Reading symbols from /Users/hdm/Downloads/TrustZone-app_nogit/secure.elf...
-> 12^done
13-target-select extended-remote localhost:50000
-> 13^error,msg="Remote connection closed"
14-interpreter-exec console "monitor reset halt"
Failed to launch GDB: Remote connection closed (from target-select extended-remote localhost:50000)

```

haneefdm, 2023-08-23:
Question: This is the first time I have seen an issue with a user using TFM. May I ask why you are loading the secure elf file twice? Nothing wrong, but curious.

haneefdm, 2023-08-23:
> I am developing an application with bootloader / secure /non_secure.

Yes, that makes sense. Exactly what I would have done. Those are the three apps that would be of most interest.

Are you using the same version of GDB with both CubeIDE and VSCode?

> Or maybe you know better how to configure everything.

I didn't touch anything you sent me. Just the path names to the elf files as I didn't create all the sub directories.

haneefdm, 2023-11-21:
Closing for lack of activity

closed 2023-11-21 by haneefdm