Marus/cortex-debug #927: Cortex-debug requires GDB version >= 9

devprodest, 2023-08-24:
Cortex-debug requires GDB version >= 9. Our team uses version 8.1

Version 9 is needed only for SWO? If you edit the plugin source code and the gdb-swo.init file, then everything works fine.

Is it possible to remove the explicit dependency in future versions and bring these features into the settings.
Is there another prettier way to get around this problem?

haneefdm, 2023-08-24:
Version 9 has very little to do with SWO. It has to do with other features like disassembly, bugs in gdb, incompatible changes in gdb, etc. I am the sole maintainer and I don't have the bandwidth to support such older versions.

Did you know you can use whatever gcc version you want but gdb has to be >= 9.

closed 2023-08-24 by haneefdm

devprodest, 2023-08-29:
What are the important problems and errors in versions less than 9, maybe it will be possible to get around them?