Marus/cortex-debug #929: Static variables don't show

gietljohannes, 2023-09-06:
**Describe the bug**
When debugging an ARM executable, the static variable section in the variable pane always stays empty, even if there are static variables in the source file.

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. Break in any source file with static variables
3. Expand the static variable
4. See issue

**Expected behavior**
Static variables should show up

**Screenshots**
[image]

**Environment:**
 - Cortex-Debug Version v1.12
 - OS: Windows 10
 - GDB Version: 10.1.90.20201028-git
 - Compiler Toolchain Version: arm-none-eabi 10.2_2020q4
 - 
**Please include `launch.json`**

```
{
    "configurations": [
    {
        "cwd": "${workspaceFolder}",
        "executable": "build/demo.axf",
        "name": "Debug with JLink",
        "request": "launch",
        "type": "cortex-debug",
        "servertype": "jlink",
        "serverpath": "${workspaceFolder}/tools/SEGGER/JLink/JLinkGDBServerCL.exe",
        "armToolchainPath": "${workspaceFolder}/tools/10.2_2020q4/bin",
        "device": "EFR32BG22CxxxF512",
        "interface": "swd",
        "serialNumber": "",
        "liveWatch": {
            "enabled": false,
            "samplesPerSecond": 20
        },
        "numberOfProcessors": 1,
        "showDevDebugOutput": "raw",
        "runToEntryPoint": "main",
        "debugServer": 4711
    },
    ]
}
```

**Attach text from `Debug Console`**

```
41-var-create  static_var_'C:\Users\root\git\efr32-basicproject\src\main.c'::StaticVar_8127f1e9ff079ff59d7a58f4fc9935d0eaf0656650939d3f8e31eadd241c5c34 @ "'C:\Users\root\git\efr32-basicproject\src\main.c'::StaticVar"
-> 41^error,msg="-var-create: unable to create variable object"
Could not create global/static variable 'C:\Users\root\git\efr32-basicproject\src\main.c'::StaticVar
Error: -var-create: unable to create variable object (from var-create  static_var_'C:\Users\root\git\efr32-basicproject\src\main.c'::StaticVar_8127f1e9ff079ff59d7a58f4fc9935d0eaf0656650939d3f8e31eadd241c5c34 @ "'C:\Users\root\git\efr32-basicproject\src\main.c'::StaticVar")
```

**Additional context**
I did a bisect and the first bad commit is 2f07c522ca4c349e9874f88875c484ff50cf1d68
I looked at the source code and it seems the issue is in gdb.ts staticVariablesRequest(). Here the exprName which is passed to gdb is prefixed with the filename. This causes the gdb server to throw an error.

haneefdm, 2023-09-06:
Thank you for reporting this issue. And, thank you for doing the detective work. A LOT has changed even since March. It appears that we created an invalid name for a gdb variable (the quotes and the backslashes).

A fix is on the way...

haneefdm, 2023-09-07:
Could you try the latest pre-release and let me know?

https://github.com/Marus/cortex-debug/releases

FYI: We had to prefix statics with file name because disambiguate files have the same name statics. With C++ and auto-generated code, it is becoming fairly common to have name collisions.

This was a Windows only problem/fix. Stupid backslashes.

Again, thank you for reporting this.

If this checks out okay, you can expect a new release within days.

gietljohannes, 2023-09-08:
I don't have hardware to test right now, but it looks good now. (Or is there a way to test without hardware?)

Glad to help out. It started to annoy me, so I kept digging and this is where I came out. I would have fixed it myself, but I didn't know what thoughts went into how exactly the variable should be named.

I can test it again on Monday or Tuesday

gietljohannes, 2023-09-12:
I tested it today and it works now!

closed 2023-09-12 by gietljohannes