Marus/cortex-debug #930: `targetId` is marked as invaild by vscode when `request=attach`, but not for `request=launch`

yyjdelete, 2023-09-14:
**Describe the bug**
`targetId` is marked as invaild by vscode when `request=attach`, but it work for `request=launch`

**To Reproduce**
Steps to reproduce the behavior:
1. Create an launch.json with targetId;
2. When "request": "launch", vscode did't report any error to it;
3. Change request to attach("request": "attach"), and see vscode report an warn: `Incorrect type. Expected "string | number".`

**Expected behavior**

targetId should work for both `request=launch` and `request=attach`

**Screenshots**
[image]

**Environment (please complete the following information):**

 - VSCode: 1.82.1
 - Cortex-Debug Version (this extension): v1.12.1
 - OS: Windows 10

**Please include `launch.json`**
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "type": "cortex-debug",
            "request": "launch",
            "name": "pyocd",
            "servertype": "pyocd",
            "executable": "build\\Debug\\test.elf",
            "runToEntryPoint": "main",
            "targetId": "xxx",
            "toolchainPrefix": "arm-none-eabi"
        },
        {
            "cwd": "${workspaceRoot}",
            "type": "cortex-debug",
            "request": "attach",
            "name": "pyocd attach",
            "servertype": "pyocd",
            "executable": "build\\Debug\\test.elf",
            "runToEntryPoint": "main",
            "targetId": "xxx",
            "toolchainPrefix": "arm-none-eabi"
        }
}
```

**Attach text from `Debug Console`**
none

**Additional context**
none

haneefdm, 2023-09-14:
It will be fixed in the next patch release. While you got an error in the editor, the `targetId` still should have worked when starting an attach type session

haneefdm, 2025-01-23:
Fix is now available from github as a pre-release

closed 2025-01-23 by haneefdm