Marus/cortex-debug #931: SVD file in config, error during microcontroller initialization FDCAN->ARA being set

andyinno, 2023-09-18:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)

**Describe the bug**

I am configuring FDCAN on a STM32 H5 microcontroller.

If I use the launch config file based on this: https://github.com/Marus/cortex-debug/issues/926#issuecomment-1690262780

I have the situation in which adding the following line:

```
"svdFile": "${workspaceRoot}/.vscode/STM32H573.svd",
```
and enabling the interrupts I have continuously the following error set:

[image]

I therefore disabled the interrupt again and while having the XPERIPHERALS windows active I checked in which line the error flag is set.

the flag is set as soon as the __HAL_RCC_FDCAN_CLK_ENABLE macro is called in the initialization code. The interrupt is then always asserted and the program does not proceed.

Seen that I am using openocd for connecting to the nucleo board, (openocd config generated by stm32cube) I loaded the elf file in stm32cube and debugged it from there. I can confirm that also enabling the corresponding XPERIPHERALS window of the stm32cube, the program executes normally. The error is not set.

this means:

- same elf
- same openocd config
- different debugging gui

**To Reproduce**
Steps to reproduce the behavior:
1. Configure FDCAN
2. In launch.json enable the SVD file loading
3. proceed to FDCAN initialization
4. step on __HAL_RCC_FDCAN_CLK_ENABLE macro
5. FDCAN ARA in FDCAN_IR is set

**Expected behavior**

to be able to debug my code while having a SVD file loaded

**Screenshots**

[comment]: <> If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (this extension) [e.g. 0.2.3] v1.12.1
 - OS: [e.g. Linux Ubuntu 18.04 LTS, Windows 11, etc.] ubuntu LTS
 - GDB Version: [e.g. 1.11.1] openocd shipped with stm32cube Open On-Chip Debugger 0.12.0-00017-gb153daa14
 - Compiler Toolchain Version: [e.g. arn-none-eabi V 11.1] 11.3.1 shipped with stm32cube

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
 {
      "executable": "${workspaceRoot}/build/Target/Cross/STM32H573VI/NonSecure/non_secure.elf",
      "servertype": "openocd",
      "serverpath": "/home/andrea/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.linux64_2.2.300.202301161003/tools/bin/openocd",
      "searchDir": [
        "/home/andrea/st/stm32cubeide_1.9.0/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.1.0.202306221132/resources/openocd/st_scripts"
      ],
      "configFiles": [
        "${workspaceRoot}/Src/Target/Cross/STM32H573VI/NonSecure/STM32H573VI.cfg"
      ],
      "name": "hdm",
      "cwd": "${workspaceRoot}",
      "gdbPath": "arm-none-eabi-gdb",
      "runToEntryPoint": "main",
      "postRestartCommands": ["break main", "continue"],
      "request": "launch",
      "type": "cortex-debug",
      "showDevDebugOutput": "raw",
      "loadFiles": [
        "${workspaceRoot}/build/Target/Cross/STM32H573VI/Secure/secure.elf",
        "${workspaceRoot}/build/Target/Cross/STM32H573VI/NonSecure/non_secure.elf"
      ],
      "symbolFiles": [
        {
          "file": "${workspaceRoot}/build/Target/Cross/STM32H573VI/NonSecure/non_secure.elf"
        },
        { "file": "${workspaceRoot}/build/Target/Cross/STM32H573VI/Secure/secure.elf" }
      ],
      // "svdFile": "${workspaceRoot}/.vscode/STM32H573.svd",
       "liveWatch": {
         "enabled": true,
         "samplesPerSecond": 4
       },

      //"rttConfig": {
      //  "enabled": true,
      //  "address": "auto",
      //  "clearSearch": true,
      //  "decoders": [
      //    {
      //      "label": "RTT CH: 0",
      //      "port": 0,
      //      "type": "console"
      //    }
      //  ]
      //},
       "swoConfig": {
         "enabled": true,
         "cpuFrequency": 250000000,
         "swoFrequency": 2500000,
         "source": "probe",
         "decoders": [
           {
             "type": "console",
             "label": "ITM0",
             "port": 0,
             "encoding": "ascii"
           }
         ]
       }
    }
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
I'm not sure which part should be relevant here. 
```

**Additional context**
Add any other context about the problem here.

haneefdm, 2023-09-18:
First of all, the SVD extension is not this one. This is the extension used. You need to follow up over there.

https://github.com/mcu-debug/peripheral-viewer

SVD (XPERIPHERALS) display has nothing to do with the debugger. It uses memory reads to populate registers and nothing else. This occurs when the debugger pauses. Please submit an issue with the relevant repo.

andyinno, 2023-09-18:
Hello @haneefdm 

thank you for your reply. 
will do.