Marus/cortex-debug #933: Oversized log files are generated under Linux systems and occupy the /tmp directory.

yueduz, 2023-09-19:
**Describe the bug**
Oversized log files are generated under Linux systems and occupy the /tmp directory.

**To Reproduce**
Steps to reproduce the behavior:
1. Start jlink debugging
2. We need to let him output more content
3. Check /tmp/gdb-server-console-106259.log. The number is not fixed, it is the process ID.
4. The output file will continue to grow until the /tmp directory is full.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (this extension): v1.12.1
 - OS: archlinux
 - GDB Version: SEGGER J-Link GDB Server V7.92e Command Line Version
 - Compiler Toolchain Version:  arn-none-eabi V 13.2.0

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug Jlink",
            "type": "cortex-debug",
            "request": "launch",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/debug/AT32_Audio.elf",
            "servertype": "jlink",
            "device": "AT32F435CMU7",
            "interface": "swd",
            "runToEntryPoint": "main",
            // "overrideLaunchCommands": [
            //     "monitor halt",
            //     "monitor reset",
            //     "-target-download",
            //     "-enable-pretty-printing"
            // ],
            // "overrideRestartCommands": [
            //     "monitor reset",
            //     "-enable-pretty-printing"
            // ],
            // "showDevDebugOutput": "raw" ,
            "svdFile": "${workspaceRoot}/AT32F435xx_v2.svd",
            "preLaunchTask": "build Debug", //先运行make任务
            "serverArgs": [
                "-nologtofile",
                "-s",
                "-vd",
                "-speed",
                "15000"
            ],
            "rttConfig": {
                "enabled": false,
                "address": "auto",
                // "clearSearch": false    // OpenOCD users may have to un-comment this
                "decoders": [
                    {
                        "port": 0,
                        "type": "console"
                    }
                ]
            }
        }
    ]
}
```
[... 174 lines cut ...]
-> =breakpoint-deleted,id="1"
20-thread-list-ids
-> 20^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
21-thread-info 1
-> 21^done,threads=[{id="1",target-id="Thread 57005",frame={level="0",addr="0x08009d30",func="main",args=[],file="/home/lei/MCU/f435_audio_board/code/src/main.c",fullname="/home/lei/MCU/f435_audio_board/code/src/main.c",line="39",arch="armv7e-m"},state="stopped"}]
22-stack-list-frames --thread 1 0 19
-> 22^done,stack=[frame={level="0",addr="0x08009d30",func="main",file="/home/lei/MCU/f435_audio_board/code/src/main.c",fullname="/home/lei/MCU/f435_audio_board/code/src/main.c",line="39",arch="armv7e-m"}]
23-data-read-memory-bytes "0x40012000" 80
-> 23^done,memory=[{begin="0x40012000",offset="0x00000000",end="0x40012050",contents="0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"}]
24-data-read-memory-bytes "0x40012100" 80
-> 24^done,memory=[{begin="0x40012100",offset="0x00000000",end="0x40012150",contents="0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"}]
25-data-read-memory-bytes "0x40012080" 4
-> 25^done,memory=[{begin="0x40012080",offset="0x00000000",end="0x40012084",contents="00000000"}]
26-data-read-memory-bytes "0x40012180" 4
-> 26^done,memory=[{begin="0x40012180",offset="0x00000000",end="0x40012184",contents="00000000"}]
27-data-read-memory-bytes "0x400120b4" 4
-> 27^done,memory=[{begin="0x400120b4",offset="0x00000000",end="0x400120b8",contents="00000000"}]
28-data-read-memory-bytes "0x400121b4" 4
-> 28^done,memory=[{begin="0x400121b4",offset="0x00000000",end="0x400121b8",contents="00000000"}]
29-exec-continue --thread 1
-> 29^running
-> *running,thread-id="all"
mi2.status = running

```

haneefdm, 2023-09-19:
This file should not be getting that big. Have you been running a session for a very long time? Like months or years? Or you really have run out of disk space because there was any to begin with.

The OS also clears contents of /tmp after 10 days on linux, so I don't understand what is going on.

yueduz, 2023-09-20:
Because when I was debugging, I would get errors, and then J-Link GDB Server would print logs in a loop. Sometimes I was doing other things without paying attention, and then it would output infinitely, freezing my system, and I had to force restart.
I don’t know whether this file is generated by vscode or cortex-debug. I hope that the log file generation can be disabled.

haneefdm, 2023-09-20:
That is our file. You cannot disable it but you can direct it to (hopefully) an unwritable file in the extension settings (not your launch.json)

Create a directory and use the pathname of that directory and it will fail to write to it. Or, give it a filename that does not have write permission.

yueduz, 2023-09-20:
Thanks for the reminder!
Adding the settings.json file solved the problem:
 ``` "cortex-debug.dbgServerLogfile": "/dev/null"```

closed 2023-09-20 by yueduz

haneefdm, 2023-09-20:
Oh yeah. Forgot this is Linux :-) Yop /dev/null is a good one and you won't get an error...