Marus/cortex-debug #935: Disassembly view showing wrong instructions

gietljohannes, 2023-09-26:
**Describe the bug**
When a breakpoint is hit in an editor window, and the user opens the disassembly view, the addresses, offsets and instructions there are all mixed up, meaning in the wrong position. Some instructions are usually in the right place, but then there are seemingly random addresses from other locations of the program text intertwined. Some instructions are even in the wrong order (meaning higher addresses appear above lower addresses and vice versa.

After scrolling a page up or down and then back to the current (highlighted) instruction, the disassembly view is reset to showing all the right instructions again as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. Place a breakpoint anywhere
3. Run program and wait for breakpoint to be reached
4. Open Disassembly view
5. 

**Expected behavior**

All lines of the disassembly are in the correct order.

**Screenshots**
**Before scrolling:** _(notice even the current, highlighted instruction is incorrect)_
[image]
**After scrolling:**
[image]

**Environment (please complete the following information):**

 - Cortex-Debug Version **v1.12.1**
 - OS: **Windows 10**
 - GDB Version: **10.1.90.20201028-git**
 - Compiler Toolchain Version: **arm-none-eabi 10-2020-q4-major**
 
**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "J-Link Debug Application",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/Firmware/build/debug/debug.axf",
            "serverpath": "${workspaceFolder}/tools/SEGGER/JLink/JLinkGDBServerCL.exe",
            "servertype": "jlink",
            "armToolchainPath": "${workspaceFolder}/tools/10.2_2020q4/bin",
            "gdbPath": "${workspaceFolder}/tools/10.2_2020q4/bin/arm-none-eabi-gdb.exe",
            "device": "EFR32BG22CxxxF512",
            "interface": "swd",
            "serialNumber": "",
            "swoConfig": {
                "enabled": true,
                "swoFrequency": 875000,
                "source": "probe",
                "cpuFrequency": 38400000,
                "decoders": [
                    {
                        "port": 0,
                        "type": "console",
                        "label": "SWO output",
                        "encoding": "ascii",
                        "showOnStartup": true
                    }
                ]
            },
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [
                    {
                        "label": "console",
                        "port": 0,
                        "type": "console"
                    }
                ]
            },
            "numberOfProcessors": 1,
            "showDevDebugOutput": "none",
            "runToEntryPoint": "main",
        },
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Suppressing output for '883-data-disassemble -s 0x0001f8fc -e 0x0001f98c -- 5'
Debug: Gdb command: -data-disassemble -s 0x0001f98c -e 0x0001fb3c -- 5      432 bytes  (tftDriverIF_Start_b)
Debug: Gdb command: -data-disassemble -s 0x0001fb3c -e 0x0001fe50 -- 5      788 bytes  (tftDriverIF_StartFade_b)
Suppressing output for '884-data-disassemble -s 0x0001f98c -e 0x0001fb3c -- 5'
Debug: data-disassemble -s 0x0001f8fc -e 0x0001f98c -- 5 => Found 59 instructions. 58 with source code, 1 without
Suppressing output for '885-data-disassemble -s 0x0001fb3c -e 0x0001fe50 -- 5'
Debug: data-disassemble -s 0x0001f98c -e 0x0001fb3c -- 5 => Found 183 instructions. 182 with source code, 1 without
Debug: data-disassemble -s 0x0001fb3c -e 0x0001fe50 -- 5 => Found 324 instructions. 323 with source code, 1 without
Debug-843: Elapsed time for Disassembly Request: 114 ms
Debug-855: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x000084e0","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":855}
Debug-855: Elapsed time for Disassembly Request: 004 ms
```

**Additional context**
Add any other context about the problem here.

haneefdm, 2023-09-26:
> All lines of the disassembly are in the correct order.

Don't know what you mean by correct order? They are ordered by addresses as returned by GDB. We don't do our own disassembly. But we create the requests (which could be wrong) to gdb to do the work.

You can use gdb from the Debug Console to print the disassembly to see what you get for an address range, and does it match what is in the disassembly window. And, does it make sense?

haneefdm, 2023-09-26:
The initial currenty highlighted line is where your `pc` is as shown in the stackframe window.

gietljohannes, 2023-09-26:
Look at the addresses on the left side (1st screenshot), they should be increasing monotonically.
I've marked the ones bold that are 'out of line':

0x82d2
**0x82ce**
0x84d2
0x84d4
0x84d6
**0x82c0**
0x84de
**0x82ba** (shown as `pc`, reading from the register view, the `pc` is actually at 0x84e0!)

The `pc` is actually in the function `LDMA_StartTransfer`, but you see instructions from other functions (`IADC_init`) mixed in

haneefdm, 2023-09-26:
We are expecting instructions in sorted order as well. Can you do the following?

* Set "showDevDebugOutput": "none" in launch.json
* Run until you get to your breakpoint
* In the debug console, enter the following and hit enter
```
disassemble /s 0x811c,0x880a
```
* Attach what you see from the command above. This is kind what we did with gdb. I want to see what it is returning. I notice that you used a compiler other than GCC (nor sure if it is IAR or ARMCC) which is okay but they can generate some very strange instructions when optimization is turned on.

haneefdm, 2023-09-26:
I used the start and end addresses by looking at what was in the Debug Console at the time. If your program has changed, please adjust the `disassemble` command accordingly.

gietljohannes, 2023-10-06:
I'm using GCC for ARM. Why would you think otherwise?

I tried your suggestion:
```
Resetting target

Temporary breakpoint 2, main () at src/main.c:61
61	{
Debug-16: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x00000e1c","offset":0,"instructionOffset":-50,"instructionCount":50,"resolveSymbols":true},"type":"request","seq":16}
Note: We detected the following memory regions as valid using gdb "info mem" and "objdump -h"
    This information is used to adjust bounds only when normal disassembly fails.
================================================================================
  Using following memory regions for disassembly
================================================================================
      Size   VMA Beg   VMA End   LMA Beg   LMA End  Attributes
================================================================================
  Unfortunately, No memory information from gdb (or gdb-server). Will try to manage without
  ------------------------------------------------------------------------------
  00001a68  00000000  00001a68  --------  --------  (.text) contents alloc load readonly code
  00000008  00001a68  00001a70  --------  --------  (.ARM.exidx) contents alloc load readonly data
  00000400  20000000  20000400  --------  --------  (.stack) alloc
  00000860  20000400  20000c60  00001a70  000022d0  (.data) contents alloc load data
  00000060  20000c60  20000cc0  000022d0  00002330  (.bss) alloc
================================================================================
Debug-16: Dequeuing...
Debug: Gdb command: -data-disassemble -s 0x00000d00 -e 0x00000e88 -- 5      392 bytes  (DmaStartTransfer)
Debug: data-disassemble -s 0x00000d00 -e 0x00000e88 -- 5 => Found 150 instructions. 150 with source code, 0 without
Debug-16: Elapsed time for Disassembly Request: 032 ms
Debug-17: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x00000e1c","offset":0,"instructionOffset":0,"instructionCount":50,"resolveSymbols":true},"type":"request","seq":17}
Debug-17: Dequeuing...
Debug: Gdb command: -data-disassemble -s 0x00000e1c -e 0x00000ef8 -- 5      220 bytes  (main)
Debug: data-disassemble -s 0x00000e1c -e 0x00000ef8 -- 5 => Found 83 instructions. 40 with source code, 43 without
Debug-17: Elapsed time for Disassembly Request: 017 ms
disassemble /s 0xd00 0xef8
A syntax error in expression, near `0xef8'.
A syntax error in expression, near `0xef8'. (from interpreter-exec console "disassemble /s 0xd00 0xef8")
disassemble /s 0xd00,0xef8
Dump of assembler code from 0xd00 to 0xef8:
src/dma.c:
16	{
   0x00000d00 <DmaStartTransfer+0>:	push	{lr}
   0x00000d02 <DmaStartTransfer+2>:	sub	sp, #44	; 0x2c
   0x00000d04 <DmaStartTransfer+4>:	mov	r3, sp
   0x00000d06 <DmaStartTransfer+6>:	stmia.w	r3, {r0, r1}

17	    LDMA_TransferCfg_t TxConfig_s = LDMA_TRANSFER_CFG_PERIPHERAL(ldmaPeripheralSignal_USART0_TXBL);
   0x00000d0a <DmaStartTransfer+10>:	movs	r0, #0
   0x00000d0c <DmaStartTransfer+12>:	str	r0, [sp, #24]
   0x00000d0e <DmaStartTransfer+14>:	str	r0, [sp, #28]
   0x00000d10 <DmaStartTransfer+16>:	str	r0, [sp, #32]
   0x00000d12 <DmaStartTransfer+18>:	str	r0, [sp, #36]	; 0x24
   0x00000d14 <DmaStartTransfer+20>:	ldr	r3, [pc, #76]	; (0xd64 <DmaStartTransfer+100>)
   0x00000d16 <DmaStartTransfer+22>:	str	r3, [sp, #24]

18	    LDMA_Descriptor_t TxDescriptor_s = LDMA_DESCRIPTOR_SINGLE_M2P_BYTE(Strip_s.Buffer_ps, USART0->TXDATA, Strip_s.Count_u8 * 3);
   0x00000d18 <DmaStartTransfer+24>:	str	r0, [sp, #8]
   0x00000d1a <DmaStartTransfer+26>:	str	r0, [sp, #12]
   0x00000d1c <DmaStartTransfer+28>:	str	r0, [sp, #16]
   0x00000d1e <DmaStartTransfer+30>:	str	r0, [sp, #20]
   0x00000d20 <DmaStartTransfer+32>:	ldrb.w	r3, [sp]
   0x00000d24 <DmaStartTransfer+36>:	add.w	r3, r3, r3, lsl #1
   0x00000d28 <DmaStartTransfer+40>:	subs	r3, #1
   0x00000d2a <DmaStartTransfer+42>:	ldrh.w	r2, [sp, #8]
   0x00000d2e <DmaStartTransfer+46>:	bfi	r2, r3, #4, #11
   0x00000d32 <DmaStartTransfer+50>:	strh.w	r2, [sp, #8]
   0x00000d36 <DmaStartTransfer+54>:	ldrb.w	r3, [sp, #10]
   0x00000d3a <DmaStartTransfer+58>:	orr.w	r3, r3, #16
   0x00000d3e <DmaStartTransfer+62>:	strb.w	r3, [sp, #10]
   0x00000d42 <DmaStartTransfer+66>:	ldrb.w	r3, [sp, #11]
   0x00000d46 <DmaStartTransfer+70>:	orr.w	r3, r3, #48	; 0x30
[... 278 lines cut ...]
   0x00000eca <_calloc_r+50>:	strd	r3, r3, [r0]
   0x00000ece <_calloc_r+54>:	bls.n	0xee8 <_calloc_r+80>
   0x00000ed0 <_calloc_r+56>:	cmp	r2, #36	; 0x24
   0x00000ed2 <_calloc_r+58>:	strd	r3, r3, [r0, #8]
   0x00000ed6 <_calloc_r+62>:	beq.n	0xeee <_calloc_r+86>
   0x00000ed8 <_calloc_r+64>:	add.w	r2, r0, #16
   0x00000edc <_calloc_r+68>:	b.n	0xeba <_calloc_r+34>
   0x00000ede <_calloc_r+70>:	movs	r1, #0
   0x00000ee0 <_calloc_r+72>:	bl	0x14d4 <memset>
   0x00000ee4 <_calloc_r+76>:	mov	r0, r4
   0x00000ee6 <_calloc_r+78>:	pop	{r4, pc}
   0x00000ee8 <_calloc_r+80>:	add.w	r2, r0, #8
   0x00000eec <_calloc_r+84>:	b.n	0xeba <_calloc_r+34>
   0x00000eee <_calloc_r+86>:	add.w	r2, r0, #24
   0x00000ef2 <_calloc_r+90>:	strd	r3, r3, [r0, #16]
   0x00000ef6 <_calloc_r+94>:	b.n	0xeba <_calloc_r+34>
End of assembler dump.
{"output":"","token":34,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```

I'm using a simpler project now to make debugging easier, hence the different code and addresses.

But considering that the output of the `disassemble` command looks fine, I'm pretty sure the issue lies somewhere in rendering the disassembly. When I scroll a few lines up/down and then in the different direction again, the lines that have moved out of view change to be correct. You can see it happening in the recording below:

[image]

gietljohannes, 2023-10-06:
After hooking the extension up to the debugger, I'm starting to think this may be a bug in VS Code, since I couldn't find anything going wrong in the extension.

haneefdm, 2023-10-06:
Good detective work! It is still possible that we are doing something wrong. Let us verify that we are sending VSCode a proper response. VSCode is not supposed to interpret what we send or do so minimally. If you put the following in your launch.json, we may be able to see what VSCode requested and what we returned.

```
            "showDevDebugOutput": "vscode",
```

It should also match the manual disassembly you have done.

>> After hooking the extension up to the debugger,

When you say that are you actually single-stepping in the extension within VSCode or using the Chrome Debug tools to examine the DOM?

gietljohannes, 2023-10-06:
Here you go:
```
From client: disassemble({"memoryReference":"0x00000e1c","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true})
Debug-18: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x00000e1c","offset":0,"instructionOffset":-200,"instructionCount":400,"resolveSymbols":true},"type":"request","seq":18}
26-interpreter-exec console "show architecture"
-> ~"The target architecture is set to \"auto\" (currently \"armv8-m.main\").\n"
-> 26^done
27-interpreter-exec console "info mem"
-> ~"Using memory regions provided by the target.\n"
-> ~"There are no memory regions defined.\n"
-> 27^done
[memory regions redacted]
Debug-18: Dequeuing...
Debug: Gdb command: -data-disassemble -s 0x00000acc -e 0x00000cb4 -- 5      488 bytes  (LDMA_StartTransfer)
Suppressing output for '28-data-disassemble -s 0x00000acc -e 0x00000cb4 -- 5'
Debug: Gdb command: -data-disassemble -s 0x00000cb4 -e 0x000014d4 -- 5     2080 bytes  (UDELAY_Delay)
Suppressing output for '29-data-disassemble -s 0x00000cb4 -e 0x000014d4 -- 5'
Debug: data-disassemble -s 0x00000acc -e 0x00000cb4 -- 5 => Found 200 instructions. 199 with source code, 1 without
Debug: data-disassemble -s 0x00000cb4 -e 0x000014d4 -- 5 => Found 788 instructions. 185 with source code, 603 without
Debug-18: Elapsed time for Disassembly Request: 516452 ms
To client: {"seq":0,"type":"response","request_seq":18,"command":"disassemble","success":true,"body":{"instructions":[{"address":"0x00000c34","pvtAddress":3124,"instruction":"4ff48061  mov.w\tr1, #1024\t; 0x400","symbol":"USART_InitSync","pvtInstructionBytes":"4f f4 80 61","instructionBytes":"<USART_InitSync+32>","location":{"name":"em_usart.c","path":"C:\\Users\\gietl.jg\\git\\efr32-led\\src\\gecko_sdk_4.0.2\\platform\\emlib\\src\\em_usart.c","sourceReference":0},"line":870,"endLine":871},{"address":"0x00000c38","pvtAddress":3128,"instruction":"0b43      orrs\tr3, r1","symbol":"USART_InitSync","pvtInstructionBytes":"0b 43","instructionBytes":"<USART_InitSync+36>","endLine":871},{"address":"0x00000c3a","pvtAddress":3130,"instruction":"1343      orrs\tr3, r2","symbol":"USART_InitSync","pvtInstructionBytes":"13 43","instructionBytes":"<USART_InitSync+38>","location":{"name":"em_usart.c","path":"C:\\Users\\gietl.jg\\git\\efr32-led\\src\\gecko_sdk_4.0.2\\platform\\emlib\\src\\em_usart.c","sourceReference":0},"line":869,"endLine":869},{"address":"0x00000c3c","pvtAddress":3132,"instruction":"ab60      str\tr3, [r5, #8]","symbol":"USART_InitSync","pvtInstructionBytes":"ab 60","instructionBytes":"<USART_InitSync+40>","endLine":869},{"address":"0x00000c3e","pvtAddress":3134,"instruction":"ab68      ldr\tr3, [r5, #8]","symbol":"USART_InitSync","pvtInstructionBytes":"ab 68","instructionBytes":"<USART_InitSync+42>","location":{"name":"em_usart.c","path":"C:\\Users\\gietl.jg\\git\\efr32-l[...]
From client: disassemble({"memoryReference":"0x00000e1c","offset":0,"instructionOffset":0,"instructionCount":50,"resolveSymbols":true})
Debug-19: Enqueuing {"command":"disassemble","arguments":{"memoryReference":"0x00000e1c","offset":0,"instructionOffset":0,"instructionCount":50,"resolveSymbols":true},"type":"request","seq":19}
Debug-19: Dequeuing...
Debug-19: Elapsed time for Disassembly Request: 72276 ms
To client: {"seq":0,"type":"response","request_seq":19,"command":"disassemble","success":true,"body":{"instructions":[{"address":"0x00000e1c","pvtAddress":3612,"instruction":"00b5      push\t{lr}","symbol":"main","pvtInstructionBytes":"00 b5","instructionBytes":"<main+0>","location":{"name":"main.c","path":"C:\\Users\\gietl.jg\\git\\efr32-led\\src\\main.c","sourceReference":0},"line":60,"endLine":61},{"address":"0x00000e1e","pvtAddress":3614,"instruction":"83b0      sub\tsp, #12","symbol":"main","pvtInstructionBytes":"83 b0","instructionBytes":"<main+2>","endLine":61},{"address":"0x00000e20","pvtAddress":3616,"instruction":"164b      ldr\tr3, [pc, #88]\t; (0xe7c <main+96>)","symbol":"main","pvtInstructionBytes":"16 4b","instructionBytes":"<main+4>","location":{"name":"em_system.h","path":"C:\\Users\\gietl.jg\\git\\efr32-led\\src\\gecko_sdk_4.0.2\\platform\\emlib\\inc\\em_system.h","sourceReference":0},"line":391,"endLine":391},{"address":"0x00000e22","pvtAddress":3618,"instruction":"1b68      ldr\tr3, [r3, #0]","symbol":"main","pvtInstructionBytes":"1b 68","instructionBytes":"<main+6>","endLine":391},{"address":"0x00000e24","pvtAddress":3620,"instruction":"c3f30743  ubfx\tr3, r3, #16, #8","symbol":"main","pvtInstructionBytes":"c3 f3 07 43","instructionBytes":"<main+8>","endLine":391},{"address":"0x00000e28","pvtAddress":3624,"instruction":"012b      cmp\tr3, #1","symbol":"main","pvtInstructionBytes":"01 2b","instructionBytes":"<main+12>","location":{"name":"em_chip.h","path":[...]
```
I'm not sure what to make of this though. Yes, I was setting breakpoints and stepping through the extension code and when I checked e.g. the `instructions` array in `disasm.ts:492` everything was in the right order.

I examined the DOM with Chrome Debug tools in addition to that, but I don't know how to debug the client-side/UI code. I'm trying to build VS Code from source now, but it's giving me a headache 😖

I guess this project is using the `.axf` extension because some developers also use the Silabs IDE, and it's the default there.

haneefdm, 2023-10-06:
Heheh. Yes, building VSCode is a challenge, but hopefully we don't need that.

Lines starting with `From client:` are requests from VSCode. Lines starting with `To client:` are a response from the extension back to VSCodel. This is a feature of VSCode. But, it appears that it is truncating the output because it is very large.

Since you are already debugging this extension, could you do the following.

1. Make sure you are debugging both the extension and the debug adapter following the instructions in https://github.com/Marus/cortex-debug#how-to-debug
2. Make the following changes to the code, and set a breakpoint where indicated
[image]

3. Make sure you are running the task `npm watch` or compile the extension from scratch
4. Start debugging the 'Extension+Debug Adapter' config -- this will launch second VSCode instance
5. In the second instance, Start debugging your FW project
6. Invoke a Disassembly
7. You should hit the breakpoint at `disasm.ts:787`
8. Examine the Console of the first VSCode instance. Since the first instance is running two debuggers, make sure you select the 'Debug Server' console output. You should see something like this....if you scroll a bit
[image]

You can `Copy All` and paste it into a file and attach it here, but also take a look for yourself what is being sent to VSCode

If you make any changes and re-compile, make sure you completely restart the second VSCode instance. You also have to terminate all debug sessions in the first instance before you restart.

gietljohannes, 2023-10-07:
Unfortunately, VS Code is truncating everything that is too large, so this method didn't help. But I found out I can copy the response to the clipboard using `copy(response)`.

Here's the output: https://gist.github.com/gietljohannes/710e0e4d8522d8ab9586f460eb37f380

I also used this code snippet to check if all addresses are in order (they are):
```arr = [0];for(i of response.body.instructions) {if(parseInt(i.address) > parseInt(arr[arr.length-1])){arr.push(parseInt(i.address));}else{console.log("Error!");break;}}```

haneefdm, 2023-10-07:
Aaaarghhh. I would like to not go into the VSCode code base. In this case, I know who authored it, he may accept a PR.

But, it is still hard to believe that the problem is in VSCode because they don't interpret anything other than the `address` field. Anything that starts with `pvt` is for my benefit.

I will review the VSCode implementation tomorrow maybe.

trampas, 2024-12-23:
I might have just ran into this.  I use objdump to dump source code: 
```
00000324 <Reset_Handler>:
Reset_Handler():
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:440
 324:	b580      	push	{r7, lr}
 326:	b084      	sub	sp, #16
 328:	af00      	add	r7, sp, #0
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:446
 32a:	4b27      	ldr	r3, [pc, #156]	@ (3c8 <Reset_Handler+0xa4>)
 32c:	3308      	adds	r3, #8
 32e:	603b      	str	r3, [r7, #0]
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:448
 330:	1d3b      	adds	r3, r7, #4
 332:	3b01      	subs	r3, #1
 334:	3bff      	subs	r3, #255	@ 0xff
 336:	607b      	str	r3, [r7, #4]
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:449
 338:	687b      	ldr	r3, [r7, #4]
 33a:	001a      	movs	r2, r3
 33c:	230f      	movs	r3, #15
 33e:	439a      	bics	r2, r3
 340:	0013      	movs	r3, r2
 342:	607b      	str	r3, [r7, #4]
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:450
 344:	e005      	b.n	352 <Reset_Handler+0x2e>
```

Using Cortex-debug I look at addres 00324 an I see different source files. 
[image]

This is with 1.12.1 and kind of makes the cortex-debug useless as that it shows the wrong source code when stepping through code.

haneefdm, 2024-12-23:
@trampas Not sure what is different/wrong. At 0x324, I see the same stuff from the screenshot vs some text you pasted (guess the output of objdump). I am missing something

Btw, we have to use gdb to disassemble and you can see the gdb requests and what it returns in the Debug Console if you enable debug -- see what other users did above

trampas, 2024-12-25:
In the image, address 32a is showing different source files. The left list is from objdump, the right is the cortex-debug disassemble which shows address 32a being from I2C master file which is incorrect.  The assembly is the same but when stepping through code cortex-debug steps through wrong source file.

haneefdm, 2024-12-25:
Like I said, we rely on GDB. We don't disassemble or calculate source locations. Objdump and gdb may not use the same algorithm either. In your objdump output in the screenshot (right side I guess), I don't see any source information at all. I see some comments that are offsets to other symbols, but that is not source information.

Can you use gdb commands to do the disassembly. That is the one I trust. and the only one we can rely on.

https://visualgdb.com/gdbreference/commands/disassemble

You can enter gdb commands at the bottom of the Debug Console...

Also, we don't step through. We ask gdb to do it and it reports back when/where the next stop happens.

We are not a debugger. We are a front end to a real debugger like GDB.

trampas, 2024-12-25:
Left be is objdump, right is from cortex-debug disassembly. I will see if I can figure out why gdb is not correct.

trampas, 2024-12-26:
After much work I found that arm-none-eabi-gcc version 10.3.1-2.2.1 works and shows the correct source files.  While version 11.2.1-1.1.1 or later versions all have an issue where the source code shown in disassembly or when stepping through files is incorrect.

trampas, 2024-12-26:
I loaded up 11.2.1-1.1.1.  I opened the disassembly in vscode. 
[image]
 If you look at address 0x019c is the `add r7,sp #0`, notice the C source file references around this code.  Now if I then open a gdb via command line and do a `disassemble /s`  I see the correct source code:
 
[image]

I do not do much with the gdb on the command line and not sure what gdb commands are used to show the source files and disassembly, but something seems wrong. 

Note that I have multiple versions of gdb on my computer, and was wondering if cortex-debug/vscode was using one from path?  I do have the following set in the launch.json

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [

        {
            "name": "Cortex Debug",
            "cwd": "${workspaceFolder}",
            "executable": "./build/debug/lora.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "jlink",
            "serverpath": "C:/Program Files/SEGGER/JLink/JLinkGDBServerCL.exe",
            "serverArgs": [],
            "device": "ATSAMR34J18B",
            "interface": "swd",
            "svdFile": "${workspaceFolder}/src/CMSIS/svd/ATSAMR34J18B.svd",
            "armToolchainPath": "${workspaceFolder}/xpacks/@xpack-dev-tools/arm-none-eabi-gcc/.content/bin",
            "showDevDebugOutput": "raw"

        }
    ]
}
```

trampas, 2024-12-26:
I have also noticed when I use 10.3.1 toolchain, I can not toggle the disassembly view, not sure why. 

I also dumped the cortex-debug log for both configurations if it helps anyone. 
[11.2.1.txt](https://github.com/user-attachments/files/18254881/11.2.1.txt)
[10.3.1.txt](https://github.com/user-attachments/files/18254882/10.3.1.txt)

haneefdm, 2024-12-26:
The disassembly view and source code toggling is handled by VSCode

https://github.com/Marus/cortex-debug/wiki/Disassembly-Debugging

trampas, 2024-12-26:
I have found that the `objdump -dlr` produces different listing files. 

10.3.1
```
build/debug/lora.elf:     file format elf32-littlearm

Disassembly of section .text:

00000000 <exception_table>:
_sfixed():
   0:	ff 7f 00 20 45 04 00 00 0d 05 00 00 4d 05 00 00     ... E.......M...
	...
  2c:	0d 05 00 00 00 00 00 00 00 00 00 00 0d 05 00 00     ................
  3c:	55 08 00 00 0d 05 00 00 75 0b 00 00 09 06 00 00     U.......u.......
  4c:	0d 05 00 00 0d 05 00 00 0d 05 00 00 0d 05 00 00     ................
  5c:	0d 05 00 00 29 07 00 00 4d 07 00 00 71 07 00 00     ....)...M...q...
  6c:	95 07 00 00 b9 07 00 00 dd 07 00 00 0d 05 00 00     ................
  7c:	0d 05 00 00 0d 05 00 00 79 09 00 00 b9 09 00 00     ........y.......
  8c:	f9 09 00 00 39 0a 00 00 79 0a 00 00 0d 05 00 00     ....9...y.......
  9c:	0d 05 00 00 0d 05 00 00 0d 05 00 00 0d 05 00 00     ................
  ac:	0d 05 00 00 00 00 00 00                             ........

000000b4 <__do_global_dtors_aux>:
__do_global_dtors_aux():
  b4:	b510      	push	{r4, lr}
  b6:	4c06      	ldr	r4, [pc, #24]	; (d0 <__do_global_dtors_aux+0x1c>)
  b8:	7823      	ldrb	r3, [r4, #0]
  ba:	2b00      	cmp	r3, #0
  ```
  
11.2.1
```
d:\Projects\SECA\LoRa\firmware\build\debug\lora.elf:     file format elf32-littlearm

Disassembly of section .text:

00000000 <exception_table>:
getStackSize():
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:244
   0:	ff 7f 00 20 95 04 00 00 5d 05 00 00 9d 05 00 00     ... ....].......
	...
getStackUsed():
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:254
  2c:	5d 05 00 00 00 00 00 00 00 00 00 00 5d 05 00 00     ]...........]...
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:257
  3c:	c5 08 00 00 5d 05 00 00 f9 0b 00 00 65 06 00 00     ....].......e...
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:250
  4c:	5d 05 00 00 5d 05 00 00 5d 05 00 00 5d 05 00 00     ]...]...]...]...
D:\Projects\SECA\LoRa\firmware/src/CMSIS/wlr089/source/gcc/startup_wlr089.c:267
  5c:	5d 05 00 00 99 07 00 00 bd 07 00 00 e1 07 00 00     ]...............
  6c:	05 08 00 00 29 08 00 00 4d 08 00 00 5d 05 00 00     ....)...M...]...
_ZN10I2C_MASTER4syncEv():
D:\Projects\SECA\LoRa\firmware/src/drivers/i2c_master/i2c_master.cpp:90
  7c:	5d 05 00 00 5d 05 00 00 e9 09 00 00 29 0a 00 00     ]...].......)...
_ZN10I2C_MASTER18setCommandBitsWireEh():
D:\Projects\SECA\LoRa\firmware/src/drivers/i2c_master/i2c_master.cpp:90
  8c:	69 0a 00 00 a9 0a 00 00 e9 0a 00 00 5d 05 00 00     i...........]...
_sfixed():
D:\Projects\SECA\LoRa\firmware/src/drivers/i2c_master/i2c_master.cpp:91
  9c:	5d 05 00 00 5d 05 00 00 5d 05 00 00 5d 05 00 00     ]...]...]...]...
_ZN10I2C_MASTER9read_byteEPh():
D:\Projects\SECA\LoRa\firmware/src/drivers/i2c_master/i2c_master.cpp:93
  ac:	5d 05 00 00 00 00 00 00                             ].......
```

[11.2.1_lst.txt](https://github.com/user-attachments/files/18255102/11.2.1_lst.txt)
[10.3.1_lst.txt](https://github.com/user-attachments/files/18255103/10.3.1_lst.txt)

trampas, 2024-12-26:
I have done some looking and it does appear to be a objdump issue.  I am trying to track down the correct place to post the issue now.