Marus/cortex-debug #938: "serverpath" ignored when "servertype" set to "external"

razvan1t, 2023-10-06:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)
Please visit the correct repo to file an issue. If this is an issue with
* Peripherals/SVD Viewer: [https://github.com/mcu-debug/peripheral-viewer](https://github.com/mcu-debug/peripheral-viewer)
* Memory Viewer: [https://github.com/mcu-debug/memview](https://github.com/mcu-debug/memview)
* RTOS Viewer: [https://github.com/mcu-debug/rtos-views](https://github.com/mcu-debug/rtos-views)
* the debugger itself, continue below

**Describe the bug**
A clear and concise description of what the bug is.

I am trying to use a custom made debugger which may make it's way into the market. 
I set"servertype" to "external"
I set "serverpath" to "firelink"
I set "serverArgs" to ["--target", "stm32h7", "--speed", "30000", "--gdbserver"]

My understanding was that when launching a debug session, Cortex-Debug was supposed to launch the server command using
the supplied arguments. It is not doing that.

 - Cortex-Debug Version (this extension) 1.12.1
 - OS: Linux Ubuntu 12.04 LTS
 - GDB Version: irrelevant
 - Compiler Toolchain Version: irrelevant

    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/bin/app_v0.03.elf",
            "name": "Application Debug",
            "servertype": "external",
			"serverpath": "firelink",
			"serverArgs": ["--target", "stm32h7", "--speed", "30000", "--gdbserver"],
			"gdbTarget": "localhost:4444",
            "armToolchainPath": "/opt/gcc-arm-none-eabi/bin",
            "runToEntryPoint": "main",
            "svdFile": "",
			"showDevDebugOutput": "raw"
        }
    ]

*Note: We are unlikely to look at the issue if you do not supply this*
```
Paste launch.json contents here
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Paste Debug Console contents here
```

**Additional context**
Add any other context about the problem here.

haneefdm, 2023-10-06:
Sorry, `external` means you launch it. You also configure the server ... there is no universal configuration method. Every server is unique and has its own lingo/].

closed 2023-10-06 by haneefdm

razvan1t, 2023-10-06:
> From "Haneef Mohammed" ***@***.***>
> To "Marus/cortex-debug" ***@***.***>
> Cc "razvan1t" ***@***.***>; "Author"
> Date 10/6/2023 6:25:57 PM
> Subject Re: [Marus/cortex-debug] "serverpath" ignored when "servertype"
> set to "external" (Issue #938)

Hi Haneef,

Would be nice if cortex-debug would just launch the executable specified 
in serverpath with the arguments in serverArgs.
This would make life so much easier. Shouldn't be too much work to do 
that.

Regards,
Raz

haneefdm, 2023-10-06:
Yeah, you can still do that. Fake it. Use one of the existing server types (like JLink) and override EVERYTHING that we do by default. Just don't expect things like RTT/SWO/etc. to work. There are just too many things specific to each server. Debug should work fine. You can use your own script or exe to intercept and do creative things with the command line as well. See this link and search for `override`, `pre`, and `post`. Understand the flow as well to customize it better.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md
https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood

You can also use VSCode features like preLaunchCommand, postDebugXXX etc., to launch and terminate your server.