Marus/cortex-debug #939: Unable to start JLink with VSCOde

vennelag, 2023-10-10:
Hi,

I have been unable to start debug on JLink in VScode. I am using an Ubuntu Ubuntu 22.04.2 LTS and Visual studio code 1.81.0.

JLink version is JLink_V792g (7.92g)

To reproduce, Just start a debug session and gdb crashes

Below is my debug console:

SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 Unspecified
Target device parameters:      none
Target interface:              JTAG
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Failed to get index for device name 'Unspecified'.GDBServer will be closed...
Shutting down...
Could not connect to target.
Please check power, connection and settings.Error: Unable to start GDB even after 5 seconds or it couldn't even start Make sure you can start gdb from the command-line and run any command like "echo hello".
    If you cannot, it is most likely because "libncurses" or "python" is not installed. Some GDBs require these
GDB session ended unexpectedly. exit-code: 253
GDB could not start as expected. Bad installation or version mismatch. See if you can start gdb from a shell prompt and check its version (Must be >= 9)

Below is my Launch.json:
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
      {
      "type": "cortex-debug",
      "request": "launch",
      "name": " J-Link launch",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/build/vas2/config_x1/artifacts/VAS2.elf",    
      "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe", 
      "servertype": "jlink",
      "device": "KW45B41Z83",
      "runToEntryPoint": "main",
      "interface": "swd",
      "serialNumber": "", //If you have more than one J-Link probe, add the serial number here.
      },
      {
        "type": "cortex-debug",
        "request": "attach",
        "name": " J-Link attach",
        "cwd": "${workspaceRoot}",
        "executable": "${workspaceRoot}/build/vas2/config_x1/artifacts/VAS2.elf",    
        "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe", 
        "servertype": "jlink",
        "device": "KW45B41Z83",
        "runToEntryPoint": "main",
        "interface": "swd",
        "serialNumber": "", //If you have more than one J-Link probe, add the serial number here.

        }
      ]
}

haneefdm, 2023-10-10:
There are lots of errors, that are happening and we are reporting them to you. Did you look into this? If you cannot run the tools manually the way we run them (all instructions in the Debug Console), neither can we.
```
J-Link is connected.
Failed to get index for device name 'Unspecified'.GDBServer will be closed...
Shutting down...
Could not connect to target.
Please check power, connection and settings.Error: Unable to start GDB even after 5 seconds or it couldn't even start Make sure you can start gdb from the command-line and run any command like "echo hello".
If you cannot, it is most likely because "libncurses" or "python" is not installed. Some GDBs require these
GDB session ended unexpectedly. exit-code: 253
GDB could not start as expected. Bad installation or version mismatch. See if you can start gdb from a shell prompt and check its version (Must be >= 9)
```

Also, I cannot make heads or tails of what you posted. That is not what our entire Debug Console looks like. It is barely recognizable.

I will close this if you do not respond or do not do your investigation.

vennelag, 2023-10-10:
Hi, attaching the entire snapshot of the debug console on Visual studio code:
[image]

haneefdm, 2023-10-10:
Please learn to use markdown, so you can paste unformatted text. Here i can at least some relevant stuff. Having text form, we'll allow me to copy/paste stuff. Are you sure you are showing me that actual configuration from launc.json. Nothing matches.

You need to read the whole output in the Debug Console to make sure things make sense. From what I see, we think you are debugging the JLink executable. Not your firmware. Pay attention to every error message in your Debug Console

haneefdm, 2023-10-10:
Oops, close by mistake.

I see no effort from you to debug what you are looking at and how to configure things. Please do so.

jnz86, 2023-11-10:
@haneefdm 

I have the same issue, created a better topic for it if you want to close this one. It could be a version issue.

https://github.com/Marus/cortex-debug/issues/954

closed 2023-11-11 by haneefdm