Marus/cortex-debug #941: Target HALT after debugger disconnect (STLink with openOCD)

OptifySudarshanPatil, 2023-10-18:
**Describe the bug**
Whenever I disconnect debugger, target halt code execution.

**To Reproduce**
Steps to reproduce the behavior:
1. Open serial port terminal
2. Start debug session
3. wait for few second
4. disconnect
5. observe serial terminal

**Expected behavior**

Controller should keep executing running code as it is.

Code-

```
#include <Arduino.h>

String timestamp = "";

int time_of_blinky = 0;
int period_blinky = 1000;

bool async_delay(int * t, int v)
{
  if((millis() - *t) > v)
  {
    *t = millis();
    return true;
  }
  return false;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  timestamp = String(__DATE__) + " " + String(__TIME__);
  Serial.println(timestamp);
  pinMode(LED_BUILTIN,OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  if(async_delay(&time_of_blinky,period_blinky))
  {
    digitalWrite(LED_BUILTIN,!digitalRead(LED_BUILTIN));
    Serial.println(String(millis()/1000) + " Blink");
  }
}

```

**Environment (please complete the following information):**

 - Cortex-Debug Version v1.12.1
 - OS: Windows 10 Pro
 - GDB Version:  0.12.0 (2023-10-02)
 - Compiler Toolchain Version: GNU gdb (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.1.90.20221210-git

**Please include `launch.json`**
```
{
    "cwd": "${workspaceFolder}",
    "executable": ".pio/build/due/firmware.elf",
    "name": "Debug with OPENOCD",
    "request": "launch",
    "type": "cortex-debug",
    "device": "ATSAM3X8E",
    "showDevDebugOutput": "vscode",
    "servertype": "openocd",
    "openOCDPreConfigLaunchCommands": [
        "set CHIPNAME at91sam3X8E",
        "set CPUTAPID 0x2ba01477"
    ],
    "configFiles": [
        "C:/Users/jaysa/.platformio/packages/tool-openocd/openocd/scripts/interface/stlink.cfg",
        "C:/Users/jaysa/.platformio/packages/tool-openocd/openocd/scripts/target/at91sam3ax_8x.cfg"
    ],
    "interface": "swd",
    "showDevDebugTimestamps": true,
    "liveWatch": {
        "enabled": true,
        "samplesPerSecond": 4
    },
    "preLaunchTask": {
        "type": "PlatformIO",
        "task": "Pre-Debug",
    },
    "postLaunchCommands": [
        "monitor reset init",
        "monitor sleep 200"
    ],
},```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
[showDevDebugOutput_raw.json](https://github.com/Marus/cortex-debug/files/13024759/showDevDebugOutput_raw.json)

haneefdm, 2023-10-18:
There are a few things you need to know
* Your gdb-server HAS to support a proper detach. Stock openocd does not the last time I tried. Even JLink did not do the right thing and we had that fixed thanks to SEGGERs response.
* Make sure you are using the right button. If you hover over the Terminate button, it will give you two options. Disconnect vs Detach or something like that. The wording kept changing. Make sure you select the right one

We issue all the right generic commands to gdb to do the right thing for the type of disconnect/detach. Your gdb-server also has to obey the protocol.

haneefdm, 2023-10-18:
From your log file, you also seemed to have used the wrong button variation. You asked to simply disconnect.

closed 2023-10-18 by haneefdm