Marus/cortex-debug #944: Avoid automatic issue of "monitor" commands to the GDB server

jebso, 2023-10-20:
**Describe the bug**
By default, cortex-debug issues `monitor` commands to the remote GDB server, such as e.g. `monitor halt` upon connecting or `monitor reset halt` when clicking the restart button. I want to use cortex-debug with our system simulator which does not support `monitor` commands. The debug session hangs as soon as these commands are issued.

* Is there a way to disable any sending of monitor commands by default?

I found there is a way to override the default behaviors. Setting `"overrideAttachCommands": []` in my  `launch.json`, I can sucessfully launch a debug session and step through the code. I wonder though:
* since I can override the attach commands, what is the default value for this?
* what else do I need to override? clicking the restart button issues the monitor command and I find there are also `overrideResetCommands` and `overrideRestartCommands`. Is there an easy way to restrict the plugin to standard GDB commands?

**To Reproduce**
The simulator provides a standard GDB server so I configured the plugin with `"servertype" : "external",`. I can launch the debug session and can see the debugger connection in the simulator. But then the debug session hangs in VS Code. Looking at the verbose debug console, I see the session hangs at the point where cortex-debug issues the `monitor halt` command after establishing the connection. From this point on, the communication between the client and the GDB server no longer works.

I tried to reproduce this manually, using a commandline GDB client instead of VS Code and the cortex-debug plugin. The behavior is the same: If I issue `monitor halt`, the connection is broken. So put simply, our simulator doesn't support any `monitor` commands.
```
(gdb) monitor halt
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

**Expected behavior**

No automatic issue of `monitor X` commands.

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension): v1.12.1
 - OS: Ubuntu 20.04 LTS on WSL2/Win10
 - GDB & Compiler Toolchain Version: arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi

**Please include `launch.json`**

This is the version with the _attach_ override to `[]`. that works to start debugging. I wonder though what else I might have cut out. What is the default? What would you recommend?
```
        {
            "name": "Using cortex-debug", // marus25.cortex-debug
            "type": "cortex-debug",
            "cwd" : "${workspaceRoot}",
            "executable": "${workspaceFolder}/build_hello_world/hello-world",
            "gdbPath": "/opt/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gdb",
            "servertype" : "external",
            "gdbTarget": "localhost:31627",
            "request" : "attach",
            "showDevDebugOutput": "vscode",
            "overrideAttachCommands": []
         }
```

**Attach text from `Debug Console`**

as shown above from the plain GDB session, the issue is the `monitor halt` command which is not supported by our simulator / GDB server.

Thanks a lot

haneefdm, 2023-10-20:
Maybe reading our documentation can help? We print some info at the start in the Debug Console itself.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md
https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood

You can also look at our source code. Every gdb-server has its own defaults. For external, you cal look at this

https://github.com/Marus/cortex-debug/blob/master/src/external.ts

> as shown above from the plain GDB session, the issue is the monitor halt command which is not supported by our simulator / GDB server.

Not sure where `monitor halt` is used.

Closing this as there is nothing actionable here. You have not shown the contents of the Debug Console.

closed 2023-10-20 by haneefdm

jebso, 2023-10-20:
Thank you for providing the link to the gdb-server source. This is really helpful.

> Not sure where monitor halt is used.

It's right there in the `external.ts` you pointed me towards:
https://github.com/Marus/cortex-debug/blob/master/src/external.ts#L56C28-L56C28
```
    public attachCommands(): string[] {
        const commands = [
            'interpreter-exec console "monitor halt"'
        ];

        return commands;
    }
```

In the source I see it's used in `launchCommands`, `attachCommands`, and `restartCommands`. There is `overrideAttachCommands` and `overrideRestartCommands` in launch.json so I can remove the `monitor`. That's great.

Thanks a lot. Problem solved :-)