Marus/cortex-debug #947: Cannot modify variables while debugging

rohinidr, 2023-11-03:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)
Please visit the correct repo to file an issue. If this is an issue with
* Peripherals/SVD Viewer: [https://github.com/mcu-debug/peripheral-viewer](https://github.com/mcu-debug/peripheral-viewer)
* Memory Viewer: [https://github.com/mcu-debug/memview](https://github.com/mcu-debug/memview)
* RTOS Viewer: [https://github.com/mcu-debug/rtos-views](https://github.com/mcu-debug/rtos-views)
* the debugger itself, continue below

**Describe the bug**
VS Code version 1.83.1
Cortex debug 
After downloading the code to STM device , I am trying to modify one of the static variables during debugging . It fails to updated the variables from Variable window but can modify this variable using terminal . Attached is the snapshot of the error 
[image]

 - Cortex-Debug Version  : v1.12.1
 - OS:  Linux Ubuntu 22.04 
 - Compiler Toolchain Version:  arn-none-eabi V 11.1

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "liveWatch": {
                "enabled": true,
                "samplesPerSecond": 4
            },
            "name": "Debug Microcontroller - STLink-V2",
            "cwd": "${workspaceFolder}",        // Path from where commands are executed
            "type": "cortex-debug",             // Debug 
            "executable": "${command:cmake.launchTargetPath}", //or fixed file path: build/project-name.elf
            "request": "launch",                // Use "attach" to connect to target w/o elf download
            "servertype": "stlink",             // Use stlink setup of cortex-M debug
            "device": "STM32G473VEH6",          // MCU used
            "interface": "swd",                 // Interface setup
            "runToEntryPoint": "main",          // Run to main and stop there
            "svdFile": "STM32G473xx.svd",         // SVD file to see registers
            "v1": false,
            "gdbPath": "arm-none-eabi-gdb-py",
            "objdumpPath": "arm-none-eabi-objdump",
            "showDevDebugOutput": "none",     // Debug output
            "debuggerArgs": [
                "-iex",
                "set auto-load safe-path /",
            ],
            "preLaunchCommands": [
                "-enable-pretty-printing",
            ]
        }
    ]
}

```
`Could not set variable: Variable object not found (from var-assign stopped_state_status 0x01)`

haneefdm, 2023-11-21:
Thank you for reporting this issue. We can duplicate this and we are actively working on a fix.

starball5, 2023-11-21:
Related on Stack Overflow: [Cannot modify variable value while debugging C code in VS Code with the Cortex Debug extension](https://stackoverflow.com/q/77355590/11107541).

haneefdm, 2023-11-22:
The fix has been pushed to this repo. If you have the time to try it out, I can make a private build you can download from GitHub.

Note that it has other unrelated changes we are testing. We will make an official release after the holidays (late Nov)

Thanks again for reporting this issue

rohinidr, 2023-11-22:
Thanks for taking care of the issue . I can test it out in couple of days . Let me know when you have the build ready . Thanks

phb98, 2024-02-01:
Hello, Has the fix for this issue been released officially ? I am encountering the same problem.

haneefdm, 2025-01-17:
There are custom builds available now that address this. Download any of the latest from here...

https://github.com/Marus/cortex-debug/releases

It will not make it to the marketplace for a while since we are doing some major features and restructuring of the code.

closed 2025-01-17 by haneefdm