Marus/cortex-debug #950: Cortex A support

Upsylonbare, 2023-11-07:
Hello, I've seen in the cortex-debug documentation that this extension targets only Cortex M.
I was only using Cortex M in my embedded projects for now but I have some projects with Cortex A.

This extension is the best one I found in the VSCode marketplace that suits my needs as an embedded engineer.
So I was wondering if, any support for Cortex A can be planned whith the same usecase covered with Cortex M ?

haneefdm, 2023-11-07:
We have several users using this extension for non Cortex-M devices. Even non-ARM devices. We are mostly agnostic but, unfortunately, we cannot change the extension's name or the description. We know of RISC-V/Xtensa/Cortex-A/R, etc.

You may run into issues with 64-bit devices so just let us know and help us fix them, wherever we print 64-bit addresses.

I alone maintain this extension for free and have a day job that pays bills. I can fix issues as they are reported. I also don't have the means to test other devices, so you have to work with me to get what you want.

Upsylonbare, 2023-11-07:
Thank you for your answer. No problem and thank you for your work.
I will try to report issues if I encounter some.

closed 2023-11-07 by Upsylonbare