Marus/cortex-debug #954: Fails to load SEGGER GDB from container. 'Unspecified'.GDBServer will be closed...

jnz86, 2023-11-10:
**Describe the bug**
GDB fails to run in container. Nearly certain this is not a cortex-debug issue as running Segger's GDB (`JLinkGDBServer`) from terminal presents the error on it's own. I am aware this is unlikely to be cortex-debug / an @haneefdm problem, but I'm posting here for other people that will have the same issue.

arm-none-eabi-gdb: OK. Runs from terminal 

JLink: OK. Runs and connects from terminal

Launching a session with VSCode: Not OK. Presents:
[image]

**To Reproduce**
Start a debug session as defined in launch.json. Fails quickly.

**Environment (please complete the following information):**
Ubuntu 22.4 container
Linux Mint 21.2 desktop
Privledged mode, USB working
Versions listed in logs
Launch json contents more accurately shown in log.

Debug console after trying to launch a session:
```
"configuration": {
    "type": "cortex-debug",
    "request": "launch",
    "name": "Debug Microcontroller (JLink)",
    "cwd": "/workspaces/firmware",
    "executable": "/workspaces/firmware/build/application_debug/application.elf",
    "showDevDebugOutput": "raw",
    "servertype": "jlink",
    "serverpath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
    "interface": "swd",
    "device": "STM32H743VI",
    "runToEntryPoint": "main",
    "rtos": "GDBServer/RTOSPlugin_FreeRTOS.so",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "toolchainPath": "/opt/arm-gnu/bin/",
    "toolchainPrefix": "arm-none-eabi",
    "gdbPath": "/opt/SEGGER/JLink/JLinkGDBServer",
    "objdumpPath": "/opt/arm-gnu/bin/arm-none-eabi-objdump",
    "extensionPath": "/root/.vscode-server/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "2418f5bc-bbcb-416d-b179-d2e2c8e8cac6",
    "pvtShowDevDebugOutput": "raw"
}

```

Running just the JLinkGDBServer from terminal itself:
```
SEGGER J-Link GDB Server V7.92k Command Line Version

JLinkARM.dll V7.92k (DLL compiled Oct 18 2023 15:03:51)

-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               off
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 Unspecified
Target device parameters:      none
Target interface:              JTAG
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Failed to get index for device name 'Unspecified'.GDBServer will be closed...
Shutting down...
Could not connect to target.
```

Output from the gdb-server terminal in VSCode
```
Waiting for gdb server to start...[2023-11-10T22:21:12.271Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/opt/SEGGER/JLink/JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32H743VI -rtos GDBServer/RTOSPlugin_FreeRTOS.so
SEGGER J-Link GDB Server V7.92o Command Line Version

JLinkARM.dll V7.92o (DLL compiled Nov  8 2023 15:46:18)

Command line: -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32H743VI -rtos GDBServer/RTOSPlugin_FreeRTOS.so
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32H743VI
Target device parameters:      none
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link Ultra V4 compiled Sep 22 2022 15:00:10
Hardware: V4.00
S/N: XXXXXXXXX
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.34 V
Listening on TCP/IP port 50000
Connecting to target...
Halting core...
Connected to target
Waiting for GDB connection...Restoring target state and closing J-Link connection...
Shutting down...
[2023-11-10T22:21:13.308Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

The cortex-debug log:
```
  GNU nano 6.2                                                                                                          /tmp/cortex-debug-server-exiting.log                                                                                                                   
[2023-11-10T22:21:12.256Z] ppid=2627 pid=2627 Debug Microcontroller (JLink): ******* Starting new session request type="launch"
[2023-11-10T22:21:12.267Z] ppid=2627 pid=2636 GDB started ppid=2627 pid=2636
[2023-11-10T22:21:13.106Z] ppid=2627 pid=2636 GDB: exited
[2023-11-10T22:21:13.106Z] ppid=2627 pid=2636 Debug Microcontroller (JLink): quitEvent: Killing server
[2023-11-10T22:21:13.106Z] ppid=2627 pid=2636 GDBServer(2637): forcing an exit with kill()
[2023-11-10T22:21:13.118Z] ppid=2627 pid=2636 Debug Microcontroller (JLink): quitEvent: sending VSCode TerminatedEvent
[2023-11-10T22:21:13.163Z] ppid=2627 pid=2636 Debug Microcontroller (JLink): Begin disconnectRequest
[2023-11-10T22:21:13.302Z] ppid=2627 pid=2636 GDBServer(2637): exited code=0 signal=null
```

References https://github.com/Marus/cortex-debug/issues/939 with more info.

@haneefdm I'm not sure how to fix this yet. I verified I have libncursesw5 and python3 installed in the container. Searching. Any advice you have would be great.

jnz86, 2023-11-11:
I had tried `apt install libncursesw5` and apt install libncurses5` to the same effect.

jnz86, 2023-11-11:
The JLink I'm using is latest release V7.92o. Went back to try V6.96 (randomly selected, 2021) to the same effect.

haneefdm, 2023-11-11:
Your arm tool chan and gdb paths are messed up. Your final launch.json has this
```
    "gdbPath": "/opt/SEGGER/JLink/JLinkGDBServer",
```

You just replaced gdb, which is our debugger.  The following should be all you need to pick up GNU tools
```
    "cortex-debug.armToolchainPath": "/opt/arm-gnu/bin/",
```

All of the following are not needed
```
    "cortex-debug.armToolchainPrefix": "arm-none-eabi",
    "cortex-debug.gdbPath": "/opt/SEGGER/JLink/JLinkGDBServerCLExe",
    "cortex-debug.objdumpPath": "/opt/arm-gnu/bin/arm-none-eabi-objdump",
```

jnz86, 2023-11-11:
Ok, I heard you.

Here is my launch json now: 

```
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug Microcontroller (JLink)",
            "cwd": "${workspaceRoot}",
            "executable": "./build/application_debug/application.elf",
            "showDevDebugOutput": "raw",
            "servertype": "jlink",
            "interface": "swd",
            "device": "STM32H743VI",
            "runToEntryPoint": "main",
            "rtos": "FreeRTOS"
        },
```

Now... When I build the container, I get this: 

[image]

**Debug Output:**
```
Cortex-Debug: VSCode debugger extension version undefined git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "type": "cortex-debug",
    "request": "launch",
    "name": "Debug Microcontroller (JLink)",
    "cwd": "/workspaces/firmware",
    "executable": "/workspaces/firmware/build/application_debug/application.elf",
    "showDevDebugOutput": "raw",
    "servertype": "jlink",
    "interface": "swd",
    "device": "STM32H743VI",
    "runToEntryPoint": "main",
    "rtos": "FreeRTOS",
    "__configurationTarget": 6,
    "__sessionId": "8e493b8f-ea95-4e87-a4a7-263a7b43bff6",
    "pvtShowDevDebugOutput": "raw",
    "graphConfig": []
}
GDB executable "undefined-gdb" was not found.
Please configure "cortex-debug.armToolchainPath" or "cortex-debug.gdbPath" correctly.

```

If I click the **Reload Window** (regarding my other post at https://github.com/Marus/cortex-debug/issues/793) then this error does not come up anymore.

**Debug Output:**
```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "type": "cortex-debug",
    "request": "launch",
    "name": "Debug Microcontroller (JLink)",
    "cwd": "/workspaces/firmware",
    "executable": "/workspaces/firmware/build/application_debug/application.elf",
    "showDevDebugOutput": "raw",
    "servertype": "jlink",
    "interface": "swd",
    "device": "STM32H743VI",
    "runToEntryPoint": "main",
    "rtos": "GDBServer/RTOSPlugin_FreeRTOS.so",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "toolchainPath": "/opt/arm-gnu/bin/",
    "toolchainPrefix": "arm-none-eabi",
    "extensionPath": "/root/.vscode-server/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "d69a726c-9559-4f3a-9141-8370becde9c3",
    "pvtShowDevDebugOutput": "raw"
}
Reading symbols from /opt/arm-gnu/bin/arm-none-eabi-objdump --syms -C -h -w /workspaces/firmware/build/application_debug/application.elf
Reading symbols from /opt/arm-gnu/bin/arm-none-eabi-nm --defined-only -S -l -C -p /workspaces/firmware/build/application_debug/application.elf
Launching GDB: /opt/arm-gnu/bin/arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
```

However, nothing happens. It's still not running. I get no terminal view. Maybe I shouldn't use workspace settings? Is it possible they aren't being loaded until after I click Reload?

haneefdm, 2023-11-11:
It appears that the first time, none of your settings are applying. They are undefined. We don't control your settings. VSCode tells us. I also feel like the first time, the extension is not even fully loaded. Again, we don't control this. See, we can't even tell what version we are. We are not container+vscode experts. This may be something you have to bring up with Microsoft.

```
Cortex-Debug: VSCode debugger extension version undefined git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
...
GDB executable "undefined-gdb" was not found.
```

jnz86, 2023-11-11:
As per the issue of getting extensions not loading on the first go without a reload - I solved that (for me at least) at https://github.com/Marus/cortex-debug/issues/793#issuecomment-1806900057. 

No more error on load / reload. No GDB undefined. That is solved. The extensions were loading into the container, but seem to have been conflicting with their local versions. I marked them as workspace in settings and there is no more issue there.

However, I'm still not up and could use some advice.

**Describe the bug:**
Press DEBUG, no error no GDB is started.

**Debug tracker with most verbose:**
```
mcu-debug.debug-tracker-vscode-0.0.15: Session 'cortex-debug:Debug Microcontroller (JLink)': Status 'initializing', id = 8ca9a786-f875-4797-85b4-ffc421fad4aa
<-- {"command":"initialize","arguments":{"clientID":"vscode","clientName":"Visual Studio Code","adapterID":"cortex-debug","pathFormat":"path","linesStartAt1":true,"columnsStartAt1":true,"supportsVariableType":true,"supportsVariablePaging":true,"supportsRunInTerminalRequest":true,"locale":"en","supportsProgressReporting":true,"supportsInvalidatedEvent":true,"supportsMemoryReferences":true,"supportsArgsCanBeInterpretedByShell":true,"supportsMemoryEvent":true,"supportsStartDebuggingRequest":true},"type":"request","seq":1}
--> {"seq":1,"type":"response","request_seq":1,"command":"initialize","success":true,"body":{"supportsHitConditionalBreakpoints":true,"supportsConfigurationDoneRequest":true,"supportsConditionalBreakpoints":true,"supportsLogPoints":true,"supportsFunctionBreakpoints":true,"supportsEvaluateForHovers":true,"supportsSetVariable":true,"supportsRestartRequest":true,"supportsGotoTargetsRequest":true,"supportSuspendDebuggee":true,"supportTerminateDebuggee":true,"supportsDataBreakpoints":true,"supportsDisassembleRequest":true,"supportsSteppingGranularity":true,"supportsInstructionBreakpoints":true,"supportsReadMemoryRequest":true,"supportsWriteMemoryRequest":true}}
mcu-debug.debug-tracker-vscode-0.0.15: Session 'cortex-debug:Debug Microcontroller (JLink)': event 'capabilities', id = 8ca9a786-f875-4797-85b4-ffc421fad4aa
<-- {"command":"launch","arguments":{"type":"cortex-debug","request":"launch","name":"Debug Microcontroller (JLink)","cwd":"/workspaces/firmware","executable":"/workspaces/firmware/build/application_debug/application.elf","showDevDebugOutput":"raw","servertype":"jlink","interface":"swd","device":"STM32H743VI","runToEntryPoint":"main","rtos":"GDBServer/RTOSPlugin_FreeRTOS.so","__configurationTarget":6,"gdbServerConsolePort":55878,"pvtAvoidPorts":[],"chainedConfigurations":{"enabled":false},"debuggerArgs":[],"swoConfig":{"enabled":false,"decoders":[],"cpuFrequency":0,"swoFrequency":0,"source":"probe"},"rttConfig":{"enabled":false,"decoders":[]},"graphConfig":[],"preLaunchCommands":[],"postLaunchCommands":[],"preAttachCommands":[],"postAttachCommands":[],"preRestartCommands":[],"postRestartCommands":[],"preResetCommands":[],"postResetCommands":[],"toolchainPath":"/opt/arm-gnu/bin/","toolchainPrefix":"arm-none-eabi","extensionPath":"/root/.vscode-server/extensions/marus25.cortex-debug-1.12.1","registerUseNaturalFormat":true,"variableUseNaturalFormat":true,"pvtVersion":"1.12.1","__sessionId":"8ca9a786-f875-4797-85b4-ffc421fad4aa"},"type":"request","seq":2}
--> {"seq":2,"type":"event","event":"output","body":{"category":"stdout","output":"Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage"}}
--> {"seq":3,"type":"event","event":"output","body":{"category":"stderr","output":"\"configuration\": {\n    \"type\": \"cortex-debug\",\n    \"request\": \"launch\",\n    \"name\": \"Debug Microcontroller (JLink)\",\n    \"cwd\": \"/workspaces/firmware\",\n    \"executable\": \"/workspaces/firmware/build/application_debug/application.elf\",\n    \"showDevDebugOutput\": \"raw\",\n    \"servertype\": \"jlink\",\n    \"interface\": \"swd\",\n    \"device\": \"STM32H743VI\",\n    \"runToEntryPoint\": \"main\",\n    \"rtos\": \"GDBServer/RTOSPlugin_FreeRTOS.so\",\n    \"__configurationTarget\": 6,\n    \"gdbServerConsolePort\": 55878,\n    \"pvtAvoidPorts\": [],\n    \"chainedConfigurations\": {\n        \"enabled\": false\n    },\n    \"debuggerArgs\": [],\n    \"swoConfig\": {\n        \"enabled\": false,\n        \"decoders\": [],\n        \"cpuFrequency\": 0,\n        \"swoFrequency\": 0,\n        \"source\": \"probe\"\n    },\n    \"rttConfig\": {\n        \"enabled\": false,\n        \"decoders\": []\n    },\n    \"graphConfig\": [],\n    \"preLaunchCommands\": [],\n    \"postLaunchCommands\": [],\n    \"preAttachCommands\": [],\n    \"postAttachCommands\": [],\n    \"preRestartCommands\": [],\n    \"postRestartCommands\": [],\n    \"preResetCommands\": [],\n    \"postResetCommands\": [],\n    \"toolchainPath\": \"/opt/arm-gnu/bin/\",\n    \"toolchainPrefix\": \"arm-none-eabi\",\n    \"extensionPath\": \"/root/.vscode-server/extensions/marus25.cortex-debug-1.12.1\",\n    \"registerUseNaturalFormat\": true,\n    \"variableUseNaturalFormat\": true,\n    \"pvtVersion\": \"1.12.1\",\n    \"__sessionId\": \"8ca9a786-f875-4797-85b4-ffc421fad4aa\",\n    \"pvtShowDevDebugOutput\": \"raw\"\n}\n"}}
--> {"seq":4,"type":"event","event":"output","body":{"category":"stderr","output":"Reading symbols from /opt/arm-gnu/bin/arm-none-eabi-objdump --syms -C -h -w /workspaces/firmware/build/application_debug/application.elf\n"}}
[... cut ...]
Reading symbols from /opt/arm-gnu/bin/arm-none-eabi-nm --defined-only -S -l -C -p /workspaces/firmware/build/application_debug/application.elf
Launching GDB: /opt/arm-gnu/bin/arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
```
**Terminal output:**
None. There is nothing but my bash. It doesn't load the re-usable GDB window at all.

Terminal output if I run the command manually:
```
[snip]

Connecting to J-Link...
J-Link is connected.
Failed to get index for device name 'Unspecified'.GDBServer will be closed...
Shutting down...
Could not connect to target.
Please check power, connection and settings.root@be390ce83434:/# 
```

jnz86, 2023-11-11:
Ok, well... Nothing you haven't seen before! libcurses... again. 

**From my `Dockerfile` to bring in dependencies:**
```
RUN apt-get install -y git cmake ninja-build clang-format clang-tidy udev bsdextrautils libncursesw5
```
At some point I removed `libncursesw5` and tried `libncurses5` based on an issue someone else had. Forgot to switch it back. So... Don't do that! 

That's it. I'm up and working. I wish I had something cleaner for the next person. So instead I'll just attach the relavent portions of what is working for me:

**From `devcontainer.json`:**
```
  "customizations": {
    "vscode": {
      "extensions": [
        "ms-vscode.cpptools-extension-pack",
        "mcu-debug.debug-tracker-vscode",
	"mcu-debug.memory-view",
	"mcu-debug.rtos-views",
	"mcu-debug.peripheral-viewer",
        "marus25.cortex-debug",
      ]
    }
```

**From `launch.json`:**
```
{
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug Microcontroller (JLink)",
            "cwd": "${workspaceRoot}",
            "executable": "./build/application_debug/application.elf",
            "showDevDebugOutput": "raw",
            "servertype": "jlink",
            "interface": "swd",
            "device": "STM32H743VI",
            "runToEntryPoint": "main",
            "rtos": "FreeRTOS"
        },
```

**From VS Code user settings `settings.json` at `~/.config/Code/User/`:**
```
    "remote.extensionKind": {
        "mcu-debug.debug-tracker-vscode": ["workspace"],
	"mcu-debug.memory-view": ["workspace"],
	"mcu-debug.rtos-views": ["workspace"],
	"mcu-debug.peripheral-viewer": ["workspace"],
        "marus25.cortex-debug": ["workspace"],
```

closed 2023-11-14 by jnz86