Marus/cortex-debug #955: Unexpected "monitor reset halt" after "-target-download"

ethpch, 2023-11-11:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)
Please visit the correct repo to file an issue. If this is an issue with
* Peripherals/SVD Viewer: [https://github.com/mcu-debug/peripheral-viewer](https://github.com/mcu-debug/peripheral-viewer)
* Memory Viewer: [https://github.com/mcu-debug/memview](https://github.com/mcu-debug/memview)
* RTOS Viewer: [https://github.com/mcu-debug/rtos-views](https://github.com/mcu-debug/rtos-views)
* the debugger itself, continue below

**Describe the bug**
I'm new to embedded development. I'm debugging hpm6750evk dev board from HPMicro. Following its UserGuide everything goes well in shell.
Then I want to use VSCode to debug it. After editing cmake-kits, launch. settings and press F5, cmake and openocd work as expected but breakpoints not hit. I have googled voluminous data about embedded debugging and execute every command in "showDevDebugOutput=raw" output, I find an unexpected "monitor reset halt" after "-target-download". If I type every command in shell except the "monitor reset halt" after "-target-download", everything works.

**To Reproduce**
Steps to reproduce the behavior:
1. Copy sample "hello_world" project from hpm_sdk/samples
2. Edit cmake-kits, launch, settings and environment variables
3. Press F5
4. See issue

**Expected behavior**

gdb runs well and debugger breaks at main

**Screenshots**

[image]

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension) [1.12.1 git(652d042)]
 - OS: [Windows 11 Professional 22H2, 22621.2428]
 - GDB Version: [10.1] (distributed with hpm_sdk)
 - Compiler Toolchain Version: [riscv32-unknown-elf V 11.1.0] (distributed with hpm_sdk)

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
launch.json
{
    "version": "0.2.0",
    "configurations": [
        {
            "configFiles": [
                "probes/jlink.cfg",
                "soc/hpm6750-single-core.cfg",
                "boards/hpm6750evk.cfg"
            ],
            "cwd": "${workspaceFolder}",
            "device": "HPM6750",
            "executable": "${command:cmake.launchTargetPath}",
            "interface": "jtag",
            "name": "HPM6750 Debug",
            "request": "launch",
            "runToEntryPoint": "main",
            "searchDir": [
                "${env:OPENOCD_SCRIPTS}"
            ],
            "servertype": "openocd",
            "showDevDebugOutput": "raw",
            "type": "cortex-debug"
        }
    ]
}
settings.json
{
    "cmake.configureOnOpen": true,
    "cmake.configureSettings": {
        "BOARD": "hpm6750evk"
    },
    "cmake.generator": "Ninja",
    "cortex-debug.gdbPath": "riscv32-unknown-elf-gdb.exe",
    "cortex-debug.variableUseNaturalFormat": false,
}
cmake-kits.json
[
    {
        "name": "rv32imac-ilp32-multilib-win",
        "compilers": {
            "C": "riscv32-unknown-elf-gcc.exe",
            "CXX": "riscv32-unknown-elf-g++.exe"
        },
        "isTrusted": true
    }
]
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file

see 14, it's the unexpected command

```
10-file-exec-and-symbols "C:/Users/13891/source/repos/HPMSDKProject/hello_worldTest/build/output/demo.elf"
-> 10^done
11-target-select extended-remote localhost:50000
-> =thread-group-started,id="i1",pid="42000"
-> =thread-created,id="1",group-id="i1"
-> ~"0x20016284 in ?? ()\n"
0x20016284 in ?? ()
-> *stopped,frame={addr="0x20016284",func="??",args=[],arch="riscv:rv32"},thread-id="1",stopped-threads="all"
mi2.status = stopped
Program stopped, probably due to a reset and/or halt issued by debugger
-> 11^connected
12-interpreter-exec console "monitor reset halt"
-> @"JTAG tap: hpm6750.cpu tap/device found: 0x1000563d (mfg: 0x31e (Andes Technology Corporation), part: 0x0005, ver: 0x1)\n"
JTAG tap: hpm6750.cpu tap/device found: 0x1000563d (mfg: 0x31e (Andes Technology Corporation), part: 0x0005, ver: 0x1)
-> 12^done
13-target-download
-> 13+download,{section=".start",section-size="54",total-size="350103"}
-> 13+download,{section=".start",section-sent="54",section-size="54",total-sent="54",total-size="350103"}
-> 13+download,{section=".vectors",section-size="1024",total-size="350103"}
-> 13+download,{section=".text",section-size="46392",total-size="350103"}
-> 13+download,{section=".data",section-size="472",total-size="350103"}
-> 13+download,{section=".data",section-sent="472",section-size="472",total-sent="47942",total-size="350103"}
-> 13^done,address="0x00000000",load-size="47942",transfer-rate="601152",write-rate="7990"
14-interpreter-exec console "monitor reset halt"
-> @"JTAG tap: hpm6750.cpu tap/device found: 0x1000563d (mfg: 0x31e (Andes Technology Corporation), part: 0x0005, ver: 0x1)\n"
JTAG tap: hpm6750.cpu tap/device found: 0x1000563d (mfg: 0x31e (Andes Technology Corporation), part: 0x0005, ver: 0x1)
-> 14^done
15-break-insert "c:\\Users\\13891\\source\\repos\\HPMSDKProject\\hello_worldTest\\src\\hello_world.c:25"
-> 15^done,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x000058ea",func="main",file="C:/Users/13891/source/repos/HPMSDKProject/hello_worldTest/src/hello_world.c",fullname="C:\\Users\\13891\\source\\repos\\HPMSDKProject\\hello_worldTest\\src\\hello_world.c",line="30",thread-groups=["i1"],times="0",original-location="c:\\Users\\13891\\source\\repos\\HPMSDKProject\\hello_worldTest\\src\\hello_world.c:25"}
Returning dummy thread-id to workaround VSCode issue with pause button not working
Returning dummy stack frame to workaround VSCode issue with pause button not working: {"threadId":1,"startFrame":0,"levels":20}
16-break-insert -t --function main
-> 16^done,bkpt={number="2",type="breakpoint",disp="del",enabled="y",addr="0x000058ca",func="main",file="C:/Users/13891/source/repos/HPMSDKProject/hello_worldTest/src/hello_world.c",fullname="C:\\Users\\13891\\source\\repos\\HPMSDKProject\\hello_worldTest\\src\\hello_world.c",line="17",thread-groups=["i1"],times="0",original-location="-function main"}
17-exec-continue --all
-> 17^running
-> *running,thread-id="all"
mi2.status = running
```

**Additional context**
I'm sure I execute every command from "showDevDebugOutput=raw" output in shell. If I don't type command 14, everything goes well and gdb breaks at main. If I type it, everythings goes like the extension does.
Do I use this extension correctly? Please give me some suggestions.

haneefdm, 2023-11-11:
We have to reset after FW download, because we have no idea where you PC/SP are. Many devices use temporary FW to program the device. Besides, a 'reset' should always work. This is how 99% of devices work. If a reset causes a problem, should take it up with your device manufacturer and their openocd scripts. Btw, the Reset/Restart buttons also will not work.

But, that is what we do by default. You can override almost everything and use your own commands.

https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood
https://github.com/Marus/cortex-debug/wiki
https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

Of course, no one reads our documentation. We even provide a link in the Debug console.

ethpch, 2023-11-11:
I'm new to embedded development so not familiar with it's run logic. I just read the `Readme` and then try to use it. I will try to use `overrideLaunchCommands` to handle it. Thanks for your reply, and sorry for my impatient.

closed 2023-11-11 by ethpch

haneefdm, 2023-11-11:
Yes, a big learning curve indeed. Every device in the embedded world has its own quirks. We can only take care of the 90% use case.

We are not commercial. I alone maintain this extension and I have a day job. We want people to do their own homework a little bit, Embedded is so much different from desktop

Good luck.