Marus/cortex-debug #956: Suspend watchdog counters while halted on a breakpoint

marco-polo-github, 2023-11-16:
Hello, because I could not find anything watchdog related I just want to ask here.

**Is your feature request related to a problem? Please describe.**
In my STM32U5 setup the code does stop on a breakpoints while the watchdog is activated, so stopping once is possible but then the watchdog reset appears.

Squinting to the CubeIDE, there is a feature in the menu below "Debug Configurations" which says
Device settings -> Suspend watchdog counters while halted -> {No configuration, Enable, Disable}
I have just checked the GDB options for that, but it is not related to that.

Would this be a feature for the cortex-debug plugin? If not, do you have a hint where to look at?

haneefdm, 2023-11-16:
You have to ask them what they do under the hood to make that work. We can't help you with this if they are setting some registers -- which you can do as well using post one of the pre/post/override settings n launch.json

This is not a gdb feature and I have not seen anything in their gdb-server.

Do these options exist if you select JLink or OpenOCD as a gdb-server?

haneefdm, 2023-11-16:
You can also do this in FW. You can query if the debugger is connected and if so, disable the watchdogs.

https://stackoverflow.com/questions/59765691/arm-cortex-m-detect-if-debugger-is-in-run-or-debug-mode

marco-polo-github, 2023-11-16:
> You have to ask them what they do under the hood to make that work. We can't help you with this if they are setting some registers -- which you can do as well using post one of the pre/post/override [...]

I guess one has to manipulate the IWDG register to do that.
Jlink and OpenOCD do not have such an option:
[image]
[image]
[image]

haneefdm, 2023-11-16:
> I guess one has to manipulate the IWDG register to do that.
Jlink and OpenOCD do not have such an option:

Yes, neither can we. But, it should be easy to do in FW.