Marus/cortex-debug #957: A serious error occurred with gbd: unable to contininue or interrupt

basprins, 2023-11-17:
I often get this error when the debugger hits a breakpoint. 

In the terminal output I see

Setting breakpoint @ address 0x6003008E, Kind = 2, Type = THUMB, BPHandle = 0x0000
WARNING: No more breakpoint resources left
ERROR: Failed to set breakpoint at 0x6003008E
Setting breakpoint @ address 0x600300EE, Kind = 2, Type = THUMB, BPHandle = 0x0000
WARNING: No more breakpoint resources left
ERROR: Failed to set breakpoint at 0x600300EE

Which is weird, I have a JLink Plus with unlimited breakpoints. I have currently set two. So I don't know if this actually causes the GDB error, but I thought to mention it anyway.

In the debug console it seems that this is the cause too:

12 received signal SIGTRAP, Trace/breakpoint trap.
Cannot remove breakpoints because program is no longer writable.
Further execution is probably impossible.
0xfffffffe in ?? ()
warning: Error removing breakpoint 2
Warning:
Cannot insert breakpoint 3.
Cannot access memory at address 0x6003008e
Cannot insert breakpoint 4.
Cannot access memory at address 0x600300ee

GDB Error? continue command is reported as error after initially succeeding. token '52'
Error: A serious error occurred with gdb, unable to continue or interrupt We may not be able to recover from this point. You can try continuing or ending session. Must address root cause though
Command aborted.

I am using:
-  arm toolchain 12.2

bp@953ccf69bc5d:/home/dev/app$ arm-none-eabi-gdb --version
GNU gdb (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.1.90.20221210-git

- cortex-debug 1.12.1 (running inside docker container).
- ubuntu 23.04 as host
- jlink 7.92

bp@953ccf69bc5d:/home/dev/app$ JLinkExe --version
SEGGER J-Link Commander V7.92c (Compiled Aug 30 2023 14:58:04)

Any ideas? Would be a happy camper if this no longer occurred :)

basprins, 2023-11-17:
Another occurrence in this scenario:

- launch application (uploads binary to 0x60020000)
- target gets reset by debugger
- custom bootloader jumps to App B at 0x60110000)
- application starts up normally, cortex debug session active and OK
- I send an mqtt command to my board to swap to AppA
- App restarts (cortex debugger still happy)
- Bootloader jumps to AppA at 0x60020000
- Serious GDB error occurred:

Command aborted.
Resetting target
Warning:
Cannot insert breakpoint 3.
Cannot access memory at address 0x60048538
Cannot insert breakpoint 2.
Cannot access memory at address 0x6004853e
Cannot insert breakpoint 1.
Cannot access memory at address 0x60048ac8

GDB Error? continue command is reported as error after initially succeeding. token '32'
Error: A serious error occurred with gdb, unable to continue or interrupt We may not be able to recover from this point. You can try continuing or ending session. Must address root cause though
Command aborted.

The weird thing now is, I only have ONE break point active. Who is making up these other two breakpoints? JLink probe bug? GDB bug? Cortex bug?

basprins, 2023-11-17:
Another case where I am just launching the app as normal:
- press F5, app uploaded
- NO breakpoints active
- A slightly longer time for the debugger to respond (5-10 sec)
- GDB error

ortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /home/dev/app/build-app/app.axf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /home/dev/app/build-app/app.axf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: /usr/bin/JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MIMXRT1021xxx5A -rtos GDBServer/RTOSPlugin_FreeRTOS.so
    Please check TERMINAL tab (gdb-server) for output from /usr/bin/JLinkGDBServerCLExe
Finished reading symbols from objdump: Time: 101 ms
Output radix now set to decimal 10, hex a, octal 12.
Input radix now set to decimal 10, hex a, octal 12.
Finished reading symbols from nm: Time: 389 ms
0x6004b540 in prvIdleTask (pvParameters=0x0 <ulTaskGetIdleRunTimePercent>) at /home/dev/app/thirdparty/FreeRTOS/Source/tasks.c:3441
3441	{
Program stopped, probably due to a reset and/or halt issued by debugger
2
Resetting target
Warning:
Cannot insert breakpoint 1.
Cannot access memory at address 0x6004853c

GDB Error? continue command is reported as error after initially succeeding. token '17'
Error: A serious error occurred with gdb, unable to continue or interrupt We may not be able to recover from this point. You can try continuing or ending session. Must address root cause though
Command aborted.
[New Remote target]

After disassembling the elf (.axf) file, I can confirm that the address mentioned is the entry point of my application.

6004853c <main>:
6004853c:	b580      	push	{r7, lr}

basprins, 2023-11-17:
I just disabled the rtos option in launch config which seems to make a huuuuuge difference.

  "configurations": [
    {
      "name": "app-debug",
      "type": "cortex-debug",
      "request": "launch",
      "cwd": "${workspaceRoot}",
      "executable": "${workspaceRoot}/build-app/app.axf",
      "device": "MIMXRT1021xxx5A",
      "serverpath": "/usr/bin/JLinkGDBServerCLExe",
      "servertype": "jlink",
      "preLaunchTask": "app - build - debug",
      "interface": "swd",
      "runToEntryPoint": "main",
      // "rtos": "FreeRTOS",
      // "showDevDebugOutput": "raw"
    },
    {

I will report back here when I have updates. But maybe you can already say something about this based on the given input. If the rtos option is a bad idea to use, then I will just remove it from the config. Can live without :)

haneefdm, 2023-11-17:
```
Launching gdb-server: /usr/bin/JLinkGDBServerCLExe -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device MIMXRT1021xxx5A -rtos GDBServer/RTOSPlugin_FreeRTOS.so
Please check TERMINAL tab (gdb-server) for output from /usr/bin/JLinkGDBServerCLExe
```

Does the JLink command look okay to you? If not please path the actual path to the free-rtos plugin. Did you check the TERMINAL tab to see if JLink crash/errors?

> "runToEntryPoint": "main",

This will cause one temporary breakpoint to be inserted.

I need full contents of Debug Console (with debug enabled as the instructions say) as well as the contents of the TERMINAL tab (JLink output). If the rtos is making a difference, then it is a problem with JLink server

haneefdm, 2023-11-21:
Duplicate of #958

In either case, we don't think this is an issue with Cortex-Debug. Closing this one. I will wait for you to update #958 with details or I will close that one too

closed 2023-11-21 by haneefdm