Marus/cortex-debug #958: Stepping through code kills the debug session

basprins, 2023-11-17:
Often when I am stepping through code the debug session dies. It takes a couple of seconds before the debugger responds after hitting F10 (so i already know "it happened again"). 

The only thing I can see in the debug console is:

[Switching to Thread 538996992]

Thread 12 hit Breakpoint 1, startup_task (parameter=0x0 <vPortEndScheduler>) at /home/dev/app/source/iobox-app/main.cpp:354
354	    if (pulse_configuration.exists())
warning: Error removing breakpoint -7
[New Remote target]
Cannot remove breakpoints because program is no longer writable.
Further execution is probably impossible.

That doesn't make much sense to me. Maybe it does to you?

- ubuntu 23.04
- cortex 1.12.1 (running in docker container)
- arm toolchain 12.2
- JLink plus (JLinkExe 7.92c)

basprins, 2023-11-17:
Same here, removing the RTOS option from config makes a huge difference. No GDB weirdness since... fingers crossed for me...

haneefdm, 2023-11-17:
>> "rtos": "FreeRTOS",

We don't do anything with the RTOS. We simply pass it to JLink. You can see that in the Debug Console. All your RTOS magic is between GDB and JLinkGDBServerCLExe. We have nothing to do with it.

For JLink, you have to actually pass the path to the DLL/so. I think the path can be relative to JLink install dir. We attempt to convert a simple name to the name of a DLL/so but best way is for you to supply that yourself.

[image]

You did not attach the contents of the Terminal where JLinkServer output exists. JLink may have crashed. You can probably see what is going on over there. YOu also did not provide the contents of Debug Console.

Please contact SEGGER if you are having issues if the JLink command line looks okay

haneefdm, 2023-11-21:
I am closing this as it is not a problem with this extension. Since no relevant information from the Debug Console or from the TERMINAL(gdb-server) is provided, nothing here is actionable.

Please use triple back-quotes for things that look like code/XML/JSON or things that are not a conversation (logs, output, etc.). It is part of markdown.

closed 2023-11-21 by haneefdm