Marus/cortex-debug #959: RTT logs are not printed to terminal

HaoboGu, 2023-11-19:
**Describe the bug**

I'm using openocd + RTT + daplink to debug my board, RTT console doesn't print logs to VSCode terminal. gdb reports it fails to find RTT control block

However, after I paused the program, I do find the control block and the log content which should be printed in the MCU memory:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. Start the program
3. Check rtt log at Terminal - RTT channel 0

**Expected behavior**

Log should be printed here

**Screenshots**

[image]

**Environment (please complete the following information):**

 - Cortex-Debug Version (this extension): v1.12.1
 - OS: Windows 11
 - GDB Version: 12.2
 - Compiler Toolchain Version: rustc 1.76.0-nightly 

**Please include `launch.json`**

```json
 {
            "name": "Cortex-Debug",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/target/thumbv7em-none-eabihf/debug/firmware",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "showDevDebugOutput": "parsed",
            "runToEntryPoint": "main",
            "configFiles": [
                "${workspaceFolder}/openocd.cfg"
            ],
            "interface": "swd",
            "svdFile": "${workspaceFolder}/STM32H7B0x.svd",
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "clearSearch": false,
                "polling_interval": 20,
                "rtt_start_retry": 20000,
                "decoders": [
                    {
                        "label": "RTT channel 0",
                        "port": 0,
                        "timestamp": true,
                        "type": "console"
                    }
                ]
            },
        },
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Cortex-Debug - stm32h7",
[... 262 lines cut ...]
Info : Bank (0) size is 128 kb, base address is 0x08000000
Info : No id from flash1
Info : No id - set flash parameters manually
Info : New GDB Connection: 1, Target stm32h7b0xx.cpu0, state: halted
Info : SWD DPIDR 0x6ba02477
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080002ac msp: 0x20020000
Info : No id from flash1
Info : No id - set flash parameters manually
Info : SWD DPIDR 0x6ba02477
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080002ac msp: 0x20020000
Info : No id from flash1
Info : No id - set flash parameters manually
Info : SWD DPIDR 0x6ba02477
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080002ac msp: 0x20020000
Info : SWD DPIDR 0x6ba02477
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080002ac msp: 0x20020000
Info : Listening on port 60000 for rtt connections
Info : rtt: Searching for control block 'SEGGER RTT'
Info : rtt: No control block found
Info : accepting 'rtt' connection on tcp/60000
```

haneefdm, 2023-11-19:
Unlike JLink, openocd has a stupid implementation of RTT. We documented this. We have to attempt to start RTT with OpenOCD until it works, it could take many tries. But, why did you set `"rtt_start_retry": 20000`? That is 20 seconds. That is when we request openocd to look for the control block with the command
```
monitor rtt start
```
You can even do this yourself from the Debug Console.

closed 2023-11-19 by haneefdm

HaoboGu, 2023-11-19:
> But, why did you set `"rtt_start_retry": 20000`? That is 20 seconds. That is when we request openocd to look for the control block with the command

I tried both 2000/20000, neither of them worked. I thought 20000 would make cortex-debug keep trying to detect RTT control blocks in 20 seconds with a polling interval 20. Am I right?

HaoboGu, 2023-11-19:
@haneefdm It worked when I manually execute `monitor rtt start` after the session starts and pauses after the rtt initialization code. Do you have any idea about how to automate this?

haneefdm, 2023-11-19:
You can use a openocd tcl script to start rtt every time you see a pause. See their doc for events. You are not giving us a chance to do repeated retries for a start at all and you need to get past the RTT initialization which happens after your main. You can stop at the start of main all day long but RTT will still not initialize  The polling interval is for what OpenOCD and how it polls the device, but that is in effect after a successful start. `rtt_start_retry` is us polling OpenOCD trying to start the darn thing.

You can read OpenOCD docs as well.

I would set `rtt_start_retry` to a lower number not a higher one. But, you need to be running -- has no effect if you are paused.