Marus/cortex-debug #963: V1.12.1 does not kill openocd process after pressing stop button (Inside Devcontainer)

nojas15, 2023-12-01:
**Describe the bug**
I use VSCode on a Windows Machine. I have setup a Devcontainer within which I debug and develop.
When I start the first Debugging session everything seems to work (First debugging session after building the devcontainer). I'm not 100% sure if I get some errors already in the console (console content posted further down).
Then I stop the debugging session and it stops as expected.
If I want to start a second debugging session I get following error message from openocd:

```
[2023-12-01T17:24:17.793Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
extern/start_openocd.sh -c "gdb_port 50003" -c "tcl_port 50004" -c "telnet_port 50005" -s /workspaces/hybrid-control-unit/extern/ -f /home/vscode/.vscode-server/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f /usr/share/openocd/scripts/interface/stlink.cfg -f /usr/share/openocd/scripts/target/stm32h7x.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDLiveWatchSetup
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50004 for tcl connections
Info : Listening on port 50005 for telnet connections
Info : clock speed 1800 kHz

[2023-12-01T17:24:17.862Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

When I check with "top" I can see there is still a openocd process running. If I kill it, it works as expected.

If I click "Reset" or "Restart" everything also works as expected (in the debug console)

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. Click on '....'
3. Scroll down to '....'
4. See issue

**Expected behavior**

OpenOCD process should be stopped and I can start a second debug session.

**Screenshots**

-

**Environment (please complete the following information):**

 - VSCode Version: 1.84.2
 - Cortex-Debug Version (this extension): v1.12.1
 - OS: Win11 -> Docker (Devcontainer) in WSL2 Backend
 - GDB Version: GNU gdb (Debian 13.1-3) 13.1
 - Compiler Toolchain Version: gcc version 12.2.1 20221205 (15:12.2.rel1-1)

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
{
            "name": "HCU v04 Debug launch",
            "type": "cortex-debug",
            "request": "launch",
            "executable": "${workspaceRoot}/eft-h1x-hcu/build/nucleo-h7a3_v0-4-0.out",
            "interface": "swd",
            "servertype": "openocd",
            //"serverpath": "openocd",
            "serverpath": "extern/start_openocd.sh",
            "cwd": "${workspaceRoot}",
            "armToolchainPath": "/usr/bin",
            "gdbPath": "gdb-multiarch",
            "searchDir": [
                "${workspaceRoot}/extern/"
            ],
[... 306 lines cut ...]
-> 34^error,msg="-var-create: unable to create variable object"
watch -var-create: unable to create variable object (from var-create --thread 1 --frame 0 watch_885dc0e3b20f3cb56302e233f846016097318015556985d6de2bd1ab211794c3 @ "O2_dac_data[1]")35-var-create --thread 1 --frame 0 watch_dd1c7dc7a36ebb5a804967480a7c97605feca1ff606fe0df05ddc9976e743d26 @ "O2_dac_data[2]"
-> 35^error,msg="-var-create: unable to create variable object"
watch -var-create: unable to create variable object (from var-create --thread 1 --frame 0 watch_dd1c7dc7a36ebb5a804967480a7c97605feca1ff606fe0df05ddc9976e743d26 @ "O2_dac_data[2]")36-var-create --thread 1 --frame 0 watch_11da499dbd2ac05b80c0e15f6c2ad2d0a14b4c0aa0aaaa9fba81fa3732f53619 @ "out_val"
-> 36^error,msg="-var-create: unable to create variable object"
watch -var-create: unable to create variable object (from var-create --thread 1 --frame 0 watch_11da499dbd2ac05b80c0e15f6c2ad2d0a14b4c0aa0aaaa9fba81fa3732f53619 @ "out_val")37-var-create --thread 1 --frame 0 hover_a0caade21aee976e19237a7c24094af6def1d2acf887dea47c867fd543f48cc2 @ "pxReadyTasksLists"
-> 37^error,msg="-var-create: unable to create variable object"
hover -var-create: unable to create variable object (from var-create --thread 1 --frame 0 hover_a0caade21aee976e19237a7c24094af6def1d2acf887dea47c867fd543f48cc2 @ "pxReadyTasksLists")38-var-create --thread 1 --frame 0 hover_88a8e759d0b575f6f113f5badafcf28ed3aeb200cbe075b3bf4af9a8ae216f56 @ "_track_list_k_mutex"
-> 38^error,msg="-var-create: unable to create variable object"
hover -var-create: unable to create variable object (from var-create --thread 1 --frame 0 hover_88a8e759d0b575f6f113f5badafcf28ed3aeb200cbe075b3bf4af9a8ae216f56 @ "_track_list_k_mutex")39-var-create --thread 1 --frame 0 hover_3a036eb0c8fd1a02d65dd10c96ee568e6997b1105097f5efe0bf86d1c1fd3176 @ "_kernel_thread_info_offsets"
-> 39^error,msg="-var-create: unable to create variable object"
hover -var-create: unable to create variable object (from var-create --thread 1 --frame 0 hover_3a036eb0c8fd1a02d65dd10c96ee568e6997b1105097f5efe0bf86d1c1fd3176 @ "_kernel_thread_info_offsets")40-var-create --thread 1 --frame 0 hover_88a8e759d0b575f6f113f5badafcf28ed3aeb200cbe075b3bf4af9a8ae216f56 @ "_track_list_k_mutex"
-> 40^error,msg="-var-create: unable to create variable object"
hover -var-create: unable to create variable object (from var-create --thread 1 --frame 0 hover_88a8e759d0b575f6f113f5badafcf28ed3aeb200cbe075b3bf4af9a8ae216f56 @ "_track_list_k_mutex")41-stack-info-frame --thread 1 --frame 0
-> 41^done,frame={level="0",addr="0x0800fb84",func="main",file="/workspaces/hybrid-control-unit/eft-h1x-hcu/stm32-lib/Core_v0-4-0/Src/main.c",fullname="/workspaces/hybrid-control-unit/eft-h1x-hcu/stm32-lib/Core_v0-4-0/Src/main.c",line="120",arch="armv7e-m"}
42-stack-list-variables --thread 1 --frame 0 --simple-values
-> 42^done,variables=[]
43-var-create --thread 1 --frame 0 hover_f22a05b5213e928b72190291eb8249c9542a6b73250b63f7d7d5be972bc55d88 @ "Initialize"
-> 43^error,msg="-var-create: unable to create variable object"
hover -var-create: unable to create variable object (from var-create --thread 1 --frame 0 hover_f22a05b5213e928b72190291eb8249c9542a6b73250b63f7d7d5be972bc55d88 @ "Initialize")

```

**Additional context**
I found following ticket which seems to be a similar / the same problem, but I couldnt find a solution for myself.

nojas15, 2023-12-01:
As already stated above the user within the devcontainer is my issue.

As stated above:

I use a user inside the devcontainer because I had issues with git when using root to commit everything.
Therefore I have to use a start_openocd.sh script which starts openocd for me as sudo. Maybe this is the issue?

```
#!/bin/sh
sudo openocd "$@"
```

Do you know of any way to start the openocd server within the devcontainer with sudo?
I think you could close this issue, as I feel this is an issue how I have set it up. But I'm also happy if I get some feedback whats the best setup for a devcontainer on windows and debugging.

Thanks for all your great work! And sorry for bothering

haneefdm, 2025-10-19:
Closing due to lack of activity.

closed 2025-10-19 by haneefdm