Marus/cortex-debug #965: `cortex-debug.openocdPath` Written to `settings.json` or `.code-workspace` file when VSCode Opens

cvanbeek13, 2023-12-04:
My company uses shared `.code-workspace` files to share workspace-specific settings, launch configurations, etc.  Every time I open VSCode, it adds the following line to my `.code-workspace` file:

```json
"cortex-debug.openocdPath": "C:\\USERS\\CVANBEEK\\APPDATA\\ROAMING\\CODE\\USER\\GLOBALSTORAGE\\BMD.STM32-FOR-VSCODE\\@XPACK-DEV-TOOLS\\OPENOCD\\0.12.0-1.1\\.CONTENT\\BIN\\OPENOCD.EXE"
```

I have an identical line in my VSCode user `settings.json` file.  Since the `openocdPath` is absolute, it is different for everyone and cannot be checked into our source control.  I would prefer if the `cortex-debug` extension just used the setting from my personal `settings.json` file so that I do not have to manually revert the `.code-workspace` file every time I open VSCode.

I have the same issue with another project that has a folder-level `settings.json` file instead of a `.code-workspace` file.  The `cortex-debug.openocdPath` is written into the folder-level `settings.json` file every time VSCode is opened.

**To Reproduce**

1. Install openOCD and add the `cortex-debug.openocdPath` variable to the user `settings.json` file.
2. Open a project that uses `"servertype": "openocd"` in the launch configuration
3. Create a `.vscode/settings.json` or `.code-workspace` file for the folder
4. Restart VSCode to reopen the folder or workspace

**Expected behavior**

If the `openocdPath` is specified in a user-scoped `settings.json` file, the extension should use that setting without writing it to the workspace level settings.

**Environment (please complete the following information):**

 - Cortex-Debug Version: 1.12.1
 - OS: Windows 10
 - OpenOCD Version: 0.12.0-01004-g9ea7f3d64-dirty

```
	"launch": {
		"version": "0.2.0",
		"configurations": [
			{
				"showDevDebugOutput": "parsed",
				"cwd": "${workspaceFolder:Project}",
				"executable": "./build/app.elf",
				"name": "Debug STM32",
				"request": "launch",
				"type": "cortex-debug",
				"servertype": "openocd",
				"preLaunchTask": "CMake Build STM",
				"device": "stm32l496",
				"configFiles": [
					"openocd.cfg"
				],
				"rtos": "FreeRTOS",
				"svdFile": "stm32l4x6.svd",
				"liveWatch": {
					"enabled": true,
					"samplesPerSecond": 4
				}
			},
			{
				"showDevDebugOutput": "parsed",
				"cwd": "${workspaceFolder:Project}",
				"executable": "./build/app.elf",
				"name": "Attach STM32",
				"request": "attach",
				"type": "cortex-debug",
				"servertype": "openocd",
				"preLaunchTask": "CMake Build STM",
				"device": "stm32l496",
				"configFiles": [
					"openocd.cfg"
				],
				"rtos": "FreeRTOS",
				"svdFile": "stm32l4x6.svd"
			}
		],
	},
```

haneefdm, 2023-12-04:
We don't do that. We have no involvement in your settings management. It is all done by VSCode. Please take it up with them. We don't even know where a given setting is coming from. It is all VSCode. We just get the final setting as decided by VSCode.

haneefdm, 2023-12-04:
We don't know what other extensions you are using. But we don't modify settings - not these kinds - and have no reason to. We only read them.

I will close this as this is 99% not our issue. Thanks for submitting though.

closed 2023-12-04 by haneefdm

cvanbeek13, 2023-12-04:
You are correct, I tracked this down to a bug in the `STM32forVSCode` extension.  Sorry for incorrectly bringing this up and thank you for pointing me in the right direction.

haneefdm, 2023-12-04:
Thanks for getting back to me.