Marus/cortex-debug #967: elf not refreshed when restarting debugger after build

JayToltTech, 2023-12-07:
**Describe the bug**

When restarting a debug session after rebuilding my .elf file, the previous build is started by the debugger unless I clean the build folder first.

**To Reproduce**

Build file use VSCode CMake Build or terminal `cmake -S . -B build -G Ninja; ninja -C build -j 10`

VSCode 'Run and Debug' with below config:
```
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceFolder}",
            "executable": "./build/adr-prototype-2307.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "jlink",
            "device": "ATSAME54P20A",
            "showDevDebugOutput": "none"
        }
```

View output using JLinkRTTViewer which is show build date from `printf("Built on %s at %s\r\n", __DATE__, __TIME__);`

I have confirmed that the ELF is modified by viewing the file system.  For whatever odd reason, it will not flash until I clean `rm -rf build` and then rebuild first.

**Expected behavior**

After build, new ELF is flashed before debugger session is started.

**Environment (please complete the following information):**

VSCode 1.85.0
Cortex-Debug v1.12.1
Segger JLink Plus Compact
Microchip SAME54Xplained
MacOS 14.1.2
gcc version 13.2.1 20231009 (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7))
GNU gdb (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.90.20231008-git

launch.json
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceFolder}",
            "executable": "./build/adr-prototype-2307.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "jlink",
            "device": "ATSAME54P20A",
            "showDevDebugOutput": "none"
        }
    ]
}
```

**Attach text from `Debug Console`**

```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
[... 121 lines cut ...]
Returning dummy thread-id to workaround VSCode issue with pause button not working
Returning dummy stack frame to workaround VSCode issue with pause button not working: {"threadId":1,"startFrame":0,"levels":20}
17-break-insert -t --function main
-> 17^done,bkpt={number="2",type="breakpoint",disp="del",enabled="y",addr="0x000002f4",func="main",file="/Users/jay/Code/adr_prototype_2307/main.c",fullname="/Users/jay/Code/adr_prototype_2307/main.c",line="6",thread-groups=["i1"],times="0",original-location="-function main"}
18-exec-continue --all
-> 18^running
-> *running,thread-id="all"
mi2.status = running
-> =breakpoint-modified,bkpt={number="2",type="breakpoint",disp="del",enabled="y",addr="0x000002f4",func="main",file="/Users/jay/Code/adr_prototype_2307/main.c",fullname="/Users/jay/Code/adr_prototype_2307/main.c",line="6",thread-groups=["i1"],times="1",original-location="-function main"}
-> ~"\n"

-> ~"Temporary breakpoint 2, main () at /Users/jay/Code/adr_prototype_2307/main.c:6\n"
Temporary breakpoint 2, main () at /Users/jay/Code/adr_prototype_2307/main.c:6
-> ~"6\t    SYS_Initialize(NULL);\n"
6	    SYS_Initialize(NULL);
-> *stopped,reason="breakpoint-hit",disp="del",bkptno="2",frame={addr="0x000002f4",func="main",args=[],file="/Users/jay/Code/adr_prototype_2307/main.c",fullname="/Users/jay/Code/adr_prototype_2307/main.c",line="6",arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
-> =breakpoint-deleted,id="2"
19-thread-list-ids
-> 19^done,thread-ids={thread-id="1"},current-thread-id="1",number-of-threads="1"
20-thread-info 1
-> 20^done,threads=[{id="1",target-id="Thread 57005",frame={level="0",addr="0x000002f4",func="main",args=[],file="/Users/jay/Code/adr_prototype_2307/main.c",fullname="/Users/jay/Code/adr_prototype_2307/main.c",line="6",arch="armv7e-m"},state="stopped"}]
21-stack-list-frames --thread 1 0 19
-> 21^done,stack=[frame={level="0",addr="0x000002f4",func="main",file="/Users/jay/Code/adr_prototype_2307/main.c",fullname="/Users/jay/Code/adr_prototype_2307/main.c",line="6",arch="armv7e-m"}]
```

JayToltTech, 2023-12-08:
Restart is Stop/Start.

The log shows that the elf is being flashed, but the code from the older elf is what is running after the Start.

haneefdm, 2023-12-08:
We don't control that. We don't even cache ANYTHING across a Stop and Start. In fact, our debugger is killed and restarted. So, I am not sure what is wrong with your OS -- are you using a network drive?

Also, if we are reloading, where did we get the older elf from?

JayToltTech, 2023-12-08:
It's a mystery to me too!  Using a local drive on MacOS with a single disk/volume, no mystery mounts or links or such...

JayToltTech, 2023-12-08:
The logs don't show more detail about how Cortex-Debug goes from:

 "executable": "/Users/jay/Code/adr_prototype_2307/build/adr-prototype-2307.elf",

to

Resetting target
Downloading 16112 bytes @ address 0x00000000
...

Perhaps something in that path is using a stale open file handle to read bytes?!?

JayToltTech, 2023-12-08:
OK, I moved to Segger Tools V7.94a from V7.86g and the issue seems to have gone away after a few light tests...

haneefdm, 2023-12-08:
No, it can't even be a bug in JLinkGDBServer. Something else is at play.

For instance, on a Mac, if you move a file to trash, it is not really gone. If you use the same path name, it will serve things from the trash without you even knowing it.

I don't have a clue what is exactly happening on your computer. I am going to close this because I am 99.9% sure it is not our problem.

closed 2023-12-08 by haneefdm

tim-hilt, 2024-10-25:
@JayToltTech did you ever find out why this is happening for you? Sounds a lot like a problem I have. We are using Windows though.