Marus/cortex-debug #971: STM32 device. Flash and debug an application with code in internal and external flash

davepregan, 2024-01-04:
Thank you for this superb project.
I have an application that uses internal and external flash on an STM32F746 processor. I can flash this using STM32CubeIDE using an external loader I created for STM32CubeProgrammer.
My colleague is very dedicated to vscode and I am very new to it. I am trying to understand the steps involved in flashing and debugging this app in vscode. I think that my steps would be
* Use STM32CubeProgrammer in cli mode to download code to internal and external flash having broken the elf into 2 separate bin files
* Load symbols from elf (how?)
* Start GDB over STLink using STLinkGDB (again how?)

I would really appreciate your advice on how to do this and how to create the launch.json (if that is how it should be done)

davepregan, 2024-01-16:
I managed to get something working. Here is my launch.json

```
{  
    // Use IntelliSense to learn about possible attributes.  
    // Hover to view descriptions of existing attributes.  
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387  
    "version": "0.2.0",  
    "configurations": [  
        {  
            "name": "<myapp Debug>",  
            "cwd": "${workspaceFolder}",  
            "executable": "./debug/<myapp>.elf",  
            "request": "launch",  
            "type": "cortex-debug",  
            "runToEntryPoint": "main",  
            "servertype": "stlink",  
            "serverArgs":   
            [  
                "-s",  
                "-m 0",  
                "-k",  
                "--frequency 21000",  
                "-el C:\\Program Files\\STMicroelectronics\\STM32Cube\\STM32CubeProgrammer\\bin\\ExternalLoader\\MT25Q128A_STM32F746_KAL_AB.stldr"  
            ],  
            "armToolchainPath": "C:/ST/STM32CubeIDE_1.6.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin",  
            "device":"STM32F746IGK7" ,  
            "svdFile": "C:/ST/STM32CubeIDE_1.6.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.productdb.debug_2.1.100.202311191741/resources/cmsis/STMicroelectronics_CMSIS_SVD/STM32F746.svd"  
        }  
    ]
}
```

closed 2024-01-16 by davepregan

haneefdm, 2024-01-16:
I am surprised this worked for you. For `serverArgs` you cannot combine arguments. For instance `"-m 0"` should be `"-m", "0"`. Same for --frequency and -el. For command line args, every argument is a distinct argument. This is true for all OSes. stlink may be parsing it differently if it is working. This will not work with most other command-line tools

davepregan, 2024-01-16:
You are correct. Well spotted. I hadn't noticed the quotes in the debug console and was unsure as to how serverArgs were being processed.

I've changed my launch.json to
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "<myAypp Debug>",
            "cwd": "${workspaceFolder}",
            "executable": "./debug/<myApp>.elf",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "servertype": "stlink",
            "serverArgs": 
            [
                "-s",
                "-m", "0",
                "-k",
                "--frequency", "21000",
                "-el", "C:\\Program Files\\STMicroelectronics\\STM32Cube\\STM32CubeProgrammer\\bin\\ExternalLoader\\MT25Q128A_STM32F746_KAL_AB.stldr"
            ],
            "armToolchainPath": "C:/ST/STM32CubeIDE_1.6.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.win32_1.1.100.202309141235/tools/bin",
            "device":"STM32F746IGK7" ,
            "svdFile": "C:/ST/STM32CubeIDE_1.6.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.productdb.debug_2.1.100.202311191741/resources/cmsis/STMicroelectronics_CMSIS_SVD/STM32F746.svd"
        }
    ]
}
```
Which gave a debug console output of
```
Launching gdb-server: "C:\\ST\\STM32CubeIDE_1.6.1\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.win32_2.1.100.202310302101\\tools\\bin\\ST-LINK_gdbserver.exe" -p 50000 -cp "C:\\ST\\STM32CubeIDE_1.6.1\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.1.100.202311100844\\tools\\bin" --swd --halt -s -m 0 -k --frequency 21000 -el "C:\\Program Files\\STMicroelectronics\\STM32Cube\\STM32CubeProgrammer\\bin\\ExternalLoader\\MT25Q128A_STM32F746_KAL_AB.stldr"
```