Marus/cortex-debug #974: No option to disable a monitor reset after download

vasanthmurari, 2024-01-14:
Type: Bug

I would like to use JTAG to write the application directly to SRAM rather than the external flash. When writing binary to SRAM, the MCU will loose the binary written via JTAG if a reset is invoked. And launch.json file doesn't have option to perform monitor RESET only when needed, and in my case, I want it before download only. Please see the highlighted section in bold.

In the excerpts below, there is a monitor reset before and after downloading the application binary and I want to avoid the reset which happens after downloading the file. 

Without this option, I can't use VS Code for development purpose.

Below is the excerpts:

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V11 compiled Aug 2 2023 10:34:01
Hardware: V11.00
S/N: 851001477
Feature(s): RDI, FlashBP, FlashDL, JFlash, GDB
Checking target voltage...
Target voltage: 3.32 V
Listening on TCP/IP port 50000
Connecting to target...
Halting core...
Connected to target
Waiting for GDB connection...Connected to 0000:0000:0000:0000:0000:0000:0000:0001
GDB client (conn. 824) requested target.xml from GDB Server
Reading common registers: Read register 'r0' (4 bytes) from hardware: 0x00000000
Read register 'r1' (4 bytes) from hardware: 0xC4AA0624
Read register 'r2' (4 bytes) from hardware: 0x00000000
Read register 'r3' (4 bytes) from hardware: 0x00000000
Read register 'r4' (4 bytes) from hardware: 0xC9120008
Read register 'r5' (4 bytes) from hardware: 0x00000000
Read register 'r6' (4 bytes) from hardware: 0x55555555
Read register 'r7' (4 bytes) from hardware: 0xC8B40624
Read register 'r8' (4 bytes) from hardware: 0x55555555
Read register 'r9' (4 bytes) from hardware: 0x55555555
Read register 'r10' (4 bytes) from hardware: 0x55555555
Read register 'r11' (4 bytes) from hardware: 0x55555555
Read register 'r12' (4 bytes) from hardware: 0x000000E1
Read register 'sp' (4 bytes) from hardware: 0xC8B40624
Read register 'lr' (4 bytes) from hardware: 0x13040008
Read register 'pc' (4 bytes) from hardware: 0xD0120008
Read register 'xpsr' (4 bytes) from hardware: 0x00000041
Read 4 bytes @ address 0x080012D0 (Data = 0xE7FDBF00)
Read 2 bytes @ address 0x080012D0 (Data = 0xBF00)
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x080012D0)
Received monitor command: reset
Resetting target
Downloading 672 bytes @ address 0x24000000
Downloading 15856 bytes @ address 0x240002A0
Downloading 15744 bytes @ address 0x24004090
Downloading 15728 bytes @ address 0x24007E10
Downloading 15648 bytes @ address 0x2400BB80
Downloading 15888 bytes @ address 0x2400F8A0
Downloading 15856 bytes @ address 0x240136B0
Downloading 15808 bytes @ address 0x240174A0
Downloading 16160 bytes @ address 0x2401B260
Downloading 15872 bytes @ address 0x2401F180
Downloading 15632 bytes @ address 0x24022F80
Downloading 16048 bytes @ address 0x24026C90
Downloading 15888 bytes @ address 0x2402AB40
Downloading 16048 bytes @ address 0x2402E950
Downloading 15968 bytes @ address 0x24032800
Downloading 15952 bytes @ address 0x24036660
Downloading 15568 bytes @ address 0x2403A4B0
Downloading 15872 bytes @ address 0x2403E180
Downloading 16000 bytes @ address 0x24041F80
Downloading 15920 bytes @ address 0x24045E00
Downloading 15968 bytes @ address 0x24049C30
Downloading 11940 bytes @ address 0x2404DA90
Downloading 36 bytes @ address 0x24050934
Downloading 16344 bytes @ address 0x24050958
Downloading 16288 bytes @ address 0x24054930
Downloading 16176 bytes @ address 0x240588D0
Downloading 15544 bytes @ address 0x2405C800
Downloading 8 bytes @ address 0x240604B8
Downloading 5188 bytes @ address 0x240604C0
Downloading 13096 bytes @ address 0x24061904
Writing register 'pc' = 0x240003E4

> **Received monitor command: reset
> Resetting target**

Read 4 bytes @ address 0x240003E4 (Data = 0xF000E75C)
Read 2 bytes @ address 0x240003E4 (Data = 0xE75C)
Read 4 bytes @ address 0x240003E4 (Data = 0xF000E75C)
Read 2 bytes @ address 0x240003E4 (Data = 0xE75C)
Starting target CPU...
Debugger requested to halt target...
...Target halted (PC = 0x24018318)
Reading common registers: Read register 'r0' (4 bytes) from hardware: 0x00000000

VS Code version: Code 1.84.2 (https://github.com/microsoft/vscode/commit/1a5daa3a0231a0fbba4f14db7ec463cf99d7768e, 2023-11-09T10:51:52.184Z)
OS version: Windows_NT x64 10.0.19045

haneefdm, 2024-01-14:
Please read the documentation. There are dozens of ways to customize/override what we do by default.

closed 2024-01-14 by haneefdm

vasanthmurari, 2024-01-17:
Hi @haneefdm , thanks, I ran through the documentation and fixed it. 

All  I had to do was to create overrideLaunchCommands and update as required. Nice documentation by the way. 

            "overrideLaunchCommands": ["monitor reset halt", "load", "c"]

The only difference in lauch.json between bootloader and main application is the overrideLaunchCommands and breakpoints for some reasons works on bootloader application but doesn't work on main application. Can you please confirm if my observation is true?

haneefdm, 2024-01-17:
At this point you are basically talking to gdb directly. breakpoints can only work if your elf file has all the appropriate symbols and debug info. This is why the elf file(s) is important. You can also load multiple elf files using `symbolFiles`