Marus/cortex-debug #975: set up stlink debugger: can not change arguments for gdbpath

domikun, 2024-01-18:
**Describe the bug**
I use VSCode on windows and try to set up a debug session to my STM32F407VETx controller. For this I want to use a ST-Link GDB Server, which was shipped with Atollic IDE v9.0.0. 

After running debug session it shows following output on the debug console:

`Launching gdb-server: "C:\\Program Files (x86)\\Atollic\\TrueSTUDIO for STM32 9.0.0\\Servers\\ST-LINK_gdbserver\\ST-LINK_gdbserver.exe" -p 50000 -cp "C:\\ST\\STM32CubeIDE_1.10.1\\STM32CubeIDE\\plugins\\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_2.0.301.202207041506\\tools\\bin" --halt`

"gdbTarget" and "debuggerArgs" seems to be not defined for servertype "stlink" (check launch.json below). 
**Where can I define arguments? And where are the arguments for -p and -cp coming from?** 

I also tried "servertype "external". The output is:

`Launching GDB: "C:\\Program Files (x86)\\Atollic\\TrueSTUDIO for STM32 9.0.0\\Servers\\ST-LINK_gdbserver\\ST-LINK_gdbserver.exe" -q --interpreter=mi2 "-f C:\\Tmp\\TEST.log" "-p 6123"`

Here the arguments are sent, but also -q and --interpreter, which are not defined for this version of gdbserver. 
**Is it possible to remove the arguments -q and --interpreter?**

**Expected behavior**

GDB server starts. 

**Environment (please complete the following information):**

 - Cortex-Debug Version v1.12.1
 - OS: Windows 10
 - GDB Version: Atollic TrueSTUDIO gdbserver for ST-Link. Version 4.2.0
 - Compiler Toolchain Version: arm-atollic-eabi

My launch.js:

`{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
    {
            "cwd": "${workspaceFolder}",
            "executable": "./main.elf",
            "name": "Debug with ST-Link",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "showDevDebugOutput": "both",
            "servertype": "stlink",
            "interface": "jtag",
            "gdbTarget": "localhost:6123",
            "gdbPath": "C:\\Program Files (x86)\\Atollic\\TrueSTUDIO for STM32 9.0.0\\Servers\\ST-LINK_gdbserver\\ST-LINK_gdbserver.exe",
            "debuggerArgs":["-f C:\\Tmp\\TEST.log","-p 6123"],
            "device": "STM32F407xx",
            
        }
    ]
}`

My setting.json:

```
{
    "files.associations": {
        "*.log": "log",
        "stdio.h": "c"
    },
    "cortex-debug.armToolchainPath": "C:\\Program Files (x86)\\Atollic\\TrueSTUDIO for STM32 9.0.0\\ARMTools\\bin\\arm-atollic-eabi-gcc.exe",
    "cortex-debug.gdbPath": "C:\\Program Files (x86)\\Atollic\\TrueSTUDIO for STM32 9.0.0\\ARMTools\\bin\\arm-atollic-eabi-gdb.exe",
    "cortex-debug.armToolchainPrefix": "arm-atollic-eabi",
    "cortex-debug.objdumpPath": "C:\\Program Files (x86)\\Atollic\\TrueSTUDIO for STM32 9.0.0\\ARMTools\\bin\\arm-atollic-eabi-objdump.exe",
    "cortex-debug.stlinkPath": "C:\\Program Files (x86)\\Atollic\\TrueSTUDIO for STM32 9.0.0\\Servers\\ST-LINK_gdbserver\\ST-LINK_gdbserver.exe", 
    "cortex-debug.stlinkPath.windows": "C:\\Program Files (x86)\\Atollic\\TrueSTUDIO for STM32 9.0.0\\Servers\\ST-LINK_gdbserver\\ST-LINK_gdbserver.exe",

}
```

haneefdm, 2024-01-18:
You are confusing `gdbPath` with `serverPath`. The later is the gdb-server. The former is the path to GDB. Not the server

haneefdm, 2024-01-18:
Your settings already seem correct. You just added some confusion by overriding `gdbPath` to something that is not GDB

closed 2024-01-18 by haneefdm