Marus/cortex-debug #977: Possibility to show integer values in hex, normal, or both?

basprins, 2024-02-04:
Hi,
I was wondering if it's possible to set cortex debug in a way where it will show both hex value and "natural" value representation?
I often want to see a bunch of variables in hex, and others in base10.  Now I am either constantly switching between hex/base10, or am I adding a bunch of watch variables as `address,x` to get the hex value while I am using the base10/natural representation. 

If "hex mode" would show both, like the base10 value and the hex value between brackets e.g. that would be great (for me :))

basprins, 2024-02-09:
And if anything can be changed here, would it also be possible to move the setting 
```
"cortex-debug.registerUseNaturalFormat": false,
```
to some other place than settings.json? 

I need the settings.json in git so that we can share settings. Now I constantly have to revert a file before committing.

haneefdm, 2025-01-17:
You can use the watch window to display the same variable/expression in multiple formats. See

https://github.com/Marus/cortex-debug/issues/1077

closed 2025-01-17 by haneefdm