Marus/cortex-debug #978: Support explicit hardware breakpoints (send "hbreak" instead of "break" to GDB)

recursivenomad, 2024-02-07:
My hardware has 4 hardware breakpoints, so I pass `set remote hardware-breakpoint-limit 4` to GDB during launch via my launch.json.  However, if I set more than 4 breakpoints, GDB does not guard me from setting too many breakpoints as expected; instead it reports an error from the remote target that it "Cannot insert hardware breakpoint 5", and that "A serious error occurred with gdb, unable to continue or interrupt".

If I instead explicitly set 4 hardware breakpoints by manually passing `hbreak` to GDB, then attempting to set a 5th hardware breakpoint this way, GDB prevents me from setting this breakpoint and gracefully reports "Hardware breakpoints used exceeds limit", with no fatal errors.

When setting breakpoints in the UI, I can confirm that once they are sent from GDB to the gdbserver; seen with `set debug remote 1`, GDB is appropriately passing `Z1` commands to automatically assign hardware breakpoints.  But oddly enough, these automatic hardware breakpoints don't appear to count towards the hardware-breakpoint-limit.  Upon querying `info break`, all breakpoints are listed as `breakpoint`, not `hw breakpoint`.  This is contrary to what `info break` reports when manually passing `hbreak` to GDB.

The ideal solution for my use-case would be to allow users to specify how Cortex-Debug communicates breakpoints to GDB (specifically so I can set it to send `hbreak` instead of `break`).

This is already [a feature in Microsoft's cpptools extension](https://github.com/microsoft/vscode-cpptools/issues/4745#issuecomment-990174224), however I would prefer not to rely on cpptools as it is largely closed-source, and I am attempting to maintain a fully FOSS development environment.  I also do not know if that feature of the cpptools extension is even compatible with Cortex-Debug.

The following alternative solutions are not ideal workarounds for my use-case:
- Microsoft's cpptools extension (closed-source)
- Manually submitting `hbreak` to GDB (cumbersome, doesn't update UI)
- Intentionally avoiding setting more than 4 breakpoints (cumbersome, risks errors, unintuitive to new users)

For context, I am using Cortex-Debug with arm-none-eabi-gdb through OpenOCD to a Raspberry Pi Pico, communicating through a Picoprobe (an additional Pico running as a CMSIS-DAP debugger).

Thank you for developing and maintaining these extensions :)

litui, 2024-09-01:
I'd like to bump this request as well. In my particular case, debugging firmware running in RAM on a Cortex-A target, using a Black Magic Probe, BMP sends the correct memory map to gdb which includes RAM areas and this causes 'break' to select software breakpoints by default. The catch, however, is that software breakpoints have not yet been implemented for Cortex-A targets on BMP (work-in-progress).

This leads to breakpoints not being set at all when "break" is called. I've temporarily worked around the issue in a BMP fork by forcing Cortex-A software breakwatch calls to use hardware instead, but I'm not sure this is the right direction for a long-term fix.

It would simply be nice to be able to set the break command to be used by cortex-debug somewhere in the launch file.

recursivenomad, 2025-04-24:
Chiming in a year later that I can all-but confirm this is caused by a bug in GDB itself.  You can follow that bug report [here](https://sourceware.org/bugzilla/show_bug.cgi?id=31403).

In the meantime, an option to explicitly send `hbreak` instead of `break` to GDB would be a helpful workaround to prevent fatal errors, as I expect it will be a year or two before the bug is addressed, and then longer before a fix is shipped into the Arm GNU Toolchain releases.

Although I'm not familiar with VSCode extension development, I'm happy to help with a PR if given some initial direction.

recursivenomad, 2025-11-06:
Thanks for pinning @haneefdm.  I have plans to submit a 1-of-2 patch to GDB when I'm able to find the time.

Can you clarify what's wrong with the VSCode UI, and how that relates to providing users with an option to send alternative GDB commands (ie. `hbreak` instead of `break`) when the UI generates a breakpoint, such as a setting in `launch.json`?

haneefdm, 2025-11-07:
If we enabled HW breakpoints what does the GUI do with it. Last time I checked, nothing. There has to be something that says this is HW and this is NOT and what the default is

Normally, a debug adapter like ours advertises what it supports via Capabilities. See Capabilities. BreakpointMode[]

https://microsoft.github.io/debug-adapter-protocol/specification#Types_Capabilities
[BreakpointMode](https://microsoft.github.io/debug-adapter-protocol/specification#Types_BreakpointMode)
[BreakpointModeApplicability](https://microsoft.github.io/debug-adapter-protocol/specification#Types_BreakpointModeApplicability)

Then the UI and VSCode will make proper requests with additional info tagged on to the breakpoint request(s). If we never see the requests, it won't work. If there is no way for the user to specify breakpoint time...the story goes on

Do you see a way for the user to tag which bkpt is which? Or an exmple of some other debugger doing it? I wasted a few days on it and then gave up because the GUI did nothing.

You can use the Microsofts VSCode MockDebug extension to see what it does. I didn't check but when they out in a new feature, they first try it out over there. It is easy to experiment with that rather than with cortex-debug

haneefdm, 2025-11-07:
> Well, on my target (Raspberry Pi Pico), I cannot place a SW breakpoint on code which is run from the flash

And, rightfully, you shouldn't try to use a SW breakpoint for code in FLASH

But, this is the most common case for nearly all embedded. Between gdb and the gdb-server, they know the memory map, max HW bkpts, and do the right thing. If it is in FLASH, you should automatically get HW breakpoints. Guess, I am not understanding.

So, what could be wrong here? Your exe mamory regions don't have right attributes, yur gdb-server does not know? So, I don't know waht is going on. Can you attach the gdb-server output found in the TERMINAL tab? This aspect is a combination of your gdb-server and gdb. GDB has no way of knowing how many HW breakpoints there are.

recursivenomad, 2025-11-07:
> So all you are looking for a graceful rejection of a failed breakpoint?

Ideally, for my specific situation, what I would like is to be *unable* to add a breakpoint, or at least have *immediate feedback* (such as a warning) if I have exceeded the `hardware-breakpoint-limit` (ie. in a perfect world, GDB returns its warning, and the breakpoint does not get set as a red dot in the UI).  I can only exceed the `hardware-breakpoint-limit` if `hbreak` is sent to GDB, otherwise the error is deferred to the target's response via openOCD.

> Either way, the command we issue to gdb will fail and you will a very similar outcome in the Debug Console.

The handling is different.  If `hardware-breakpoint-limit` is reached through `hbreak`, GDB catches this *immediately* and doesn't ever send the command to the server (openOCD).  Whereas for exceeding that limit through `break`, the command is sent through openOCD *only after the user continues execution* because GDB fails to log it as a HW breakpoint, and it fails via a response from the target.  This would be more graceful to immediately reject the breakpoint the user just set, rather than waiting for the user to continue.

I have different needs from @litui, which they describe above; but both of our issues would be resolved with the ability to send the `hbreak` command.  I agree with their justificiation that even though the problem stems from GDB, offering more versatility within cortex-debug allows more people to be able to use it for debugging.

---

Example showing that `hbreak` is different from `break` in the context of `hardware-breakpoint-limit`:

**Within launch.json:**
```json
"postLaunchCommands": [
    "set remote hardware-breakpoint-limit 4",
    "set remote hardware-watchpoint-limit 2",
],
```

**GDB commands:**
```
interrupt
hbreak main.c:17
hbreak main.c:19
hbreak main.c:20
hbreak main.c:26
hbreak main.c:27
continue
info break
```

**DEBUG CONSOLE (GDB) output:**
```
interrupt
{"output":"","token":28,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}

Thread 
1 "rp2040.core0" received signal SIGINT, Interrupt.
sleep_until (t=94605449) at C:\tools\sdk\pico\pico-sdk\src\common\pico_time\time.c:408
408	            while (!time_reached(t_before)) {
hbreak main.c:17
Hardware assisted breakpoint 1 at 0x100003ac: file C:\tools\pico-testing\hello-world\main.c, line 17.
{"output":"","token":39,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
hbreak main.c:19
Hardware assisted breakpoint 2 at 0x100003b0: file C:\tools\pico-testing\hello-world\main.c, line 19.
{"output":"","token":46,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
hbreak main.c:20
Hardware assisted breakpoint 3 at 0x100003b6: file C:\tools\pico-testing\hello-world\main.c, line 20.
{"output":"","token":53,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
hbreak main.c:26
Hardware assisted breakpoint 4 at 0x100003be: file C:\tools\pico-testing\hello-world\main.c, line 26.
{"output":"","token":60,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
hbreak main.c:27
Hardware breakpoints used exceeds limit.
Hardware breakpoints used exceeds limit. (from interpreter-exec console "hbreak main.c:27")
continue
Continuing.
{"output":"","token":74,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}

Thread 1 "rp2040.core0" hit Breakpoint 4, main () at C:\tools\pico-testing\hello-world\main.c:26
26	    gpio_put(PIN_LED, true);
info break
Num     Type           Disp Enb Address    What
1       hw breakpoint  keep y   0x100003ac in main at C:\tools\pico-testing\hello-world\main.c:17
2       hw breakpoint  keep y   0x100003b0 in main at C:\tools\pico-testing\hello-world\main.c:19
3       hw breakpoint  keep y   0x100003b6 in main at C:\tools\pico-testing\hello-world\main.c:20
4       hw breakpoint  keep y   0x100003be in main at C:\tools\pico-testing\hello-world\main.c:26
	breakpoint already hit 1 time
{"output":"","token":81,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```

haneefdm, 2025-11-07:
Ahhh, I see @recursivenomad You are saying it looks like an exception occurs which is like hitting a breakpoint and the debugger stops right there, requiring a manual continue? Is this what is happening?

Btw, we have been cross posting furiously and I got lost in the thread.

recursivenomad, 2025-11-07:
Yes, this has turned into quite the conversation :)  Thank you for your responsiveness.

To clarify, the warning you sent above is from the current implementation, which sends `break`.  That warning only occurs *after* you "continue" the execution.  VSCode highlights a random (or perhaps the closest) breakpoint, and shows this:

[image]

I can continue debugging if I remove one of the breakpoints and "continue" again, but that does not address the issue of immediacy, which I describe in my [previous](https://github.com/Marus/cortex-debug/issues/978#issuecomment-3503930778) message - nor does it help @litui's situation, which to the best of my understanding is an issue where their target does not accept SW breakpoints, even in RAM (but somehow does handle HW breakpoints in RAM), and so sending `hbreak` would help their unique situation as well, just in a different way.

Ultimately, the request is for an additional point of versatility in cortex-debug, allowing us to override what command is sent to GDB when a breakpoint is set in the UI (with both of our cases benefiting from changing that command from `break` to `hbreak`).  Having this as an option would allow us to fix these issues ourselves, instead of bugging you in the future 🙂

litui, 2025-11-07:
> to the best of my understanding is an issue where their target does not accept SW breakpoints, even in RAM (but somehow does handle HW breakpoints in RAM), and so sending hbreak would help their unique situation as well, just in a different way.

You got it. It's a Cortex-A feature.

> This is the part I am not getting. Workaround for what?. Workaround to a faulty gdb-server? Or how that gdb-server is configured?

In this particular case, it's a hardware JTAG/SWD debug adapter (Blackmagic Debug) that runs a native gdb instance (over serial UART) that does not (yet) support software breakpoints on some hardware. You're not wrong about how a *fully functioning* gdb instance should behave in its stock behaviour. I recognize this is an edge case, but it's one that gdb itself allows one to work around through explicit calls to hbreak. I'm just desiring the ability to make use of that gdb workaround via cortex-debug so I'm not stuck using the commandline for it.

recursivenomad, 2025-11-07:
@haneefdm:

> But doing it for the wrong reasons is not a good thing.

I don't believe these are "wrong reasons" to include a broad feature - the goal is enabling user control of more complete GDB command support.  As of right now, cortex-debug does not provide any way to support GDB's `hbreak` command, and the discussed solution here would bring that support to cortex-debug.

For example, look at `"overrideLaunchCommands"`, `"overrideRestartCommands"`, and `"overrideResetCommands"`.  In my experience, they each allow for users to override default behaviour in cortex-debug, enabling us to implement GDB commands that are unique to our hardware & software situation.

In my mind, adding `"overrideBreakCommand"` (or similar) is no different.  It is an option which would override default behaviour and enable users to implement GDB commands which are currently missing from cortex-debug.

> Lets keep the issue faced by @recursivenomad separate from @litui

I still believe the implementation of something like `"overrideBreakCommand"` could resolve both (as well as prevent future issues), but I respect your desire to keep the topics separate moving forward.

> I will look into what can be done (if any) about @recursivenomad issue. Wondering if this is just OpenOCD or it is all gdb-servers.

As my specific issue stems from GDB being allowed to pass an automatic hardware breakpoint request which exceeds the supported number of HW breakpoints on the target (which could be avoided by explicitly sending `hbreak` instead), I assume this would affect all gdb-servers; however, I assume the way in which they each respond to this bad request is likely different.

I am happy to continue providing any logs or ideas as you need them.

litui, 2025-11-07:
Alright. I've stated my case, you've stated yours. Bowing out of this thread. All the best.

haneefdm, 2025-11-10:
Thanks for pointing the typo. But, I cannot see the issue with the limit. Not sure how the limit enforcement interferes with getting HW breakpoints ... according to the code I have. I added the limit not just to mirror cpptools. Without it, when you exceed the number of breakpoints, GDB is not well behaved....unless you set the `"set remote hardware-breakpoint-limit 4",` This is the whole reason I did this. Most people would not know to use that gdb feature. But, I made another mistake. I lumped breakpoints and watchpoints into the same category.

While your way of enforcing limits is clearly better, I don't want to revisit this when people see GDB messing up and me explaining oh...you need set these gdb options in preLaunchTask....

I see these commands go to gdb. Can you check your output again?
```
launch.json:
            "hardwareBreakpoints": {
                "require": true,                 // I fixed this typo
                "limit": 6
            },
```
Debug Console
```
18-break-insert -h "/Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:66" 
19-break-insert -h "/Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:71" 
20-break-insert -h "/Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:72" 
21-break-insert -h "/Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:73" 
22-break-insert -h "/Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:74" 
23-break-insert -h "/Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:75" 
```
and the following for the info: They are all HW breakpoints (I cleaned up the output)
```
info breakpoints
42-interpreter-exec console "info breakpoints"
1       hw breakpoint  keep y   0x10004744 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:66
2       hw breakpoint  keep y   0x10004762 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:71
3       hw breakpoint  keep y   0x10004768 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:72
4       hw breakpoint  keep y   0x1000476e in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:73
5       hw breakpoint  keep y   0x10004774 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:74
6       hw breakpoint  keep y   0x1000477a in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:75
```

recursivenomad, 2025-11-10:
I can provide a more thorough reproduction later, but I'll quickly note that the bug with `"limit"` only occurs after you try to set *more* than the `"limit"` value.  So in your above example, if you try setting a 7th breakpoint, and then submit `info break`, I believe it will show that all the breakpoints have been converted to automatic breakpoints.

haneefdm, 2025-11-10:
I am seeing some unexplainable behavior from GDB. I have no limits of any sort. I was stopped at line 66 with total of 4 bkpts. My HW allows 6. I kept adding more breakpoints beyond the limit and no complaints from gdb at all. I can keep on adding. I have eight below but it will keep going without complaints. Normally, it barfs on the 7th one. I have been able to do this more than once. However, while gdb seems to be responding normally...I can neither continue/single-step/proceed in anyway. It is stuck

[image]

```
107-interpreter-exec console "info breakpoints"
-> ~"Num     Type           Disp Enb Address    What\n"
Num     Type           Disp Enb Address    What
-> ~"23      hw breakpoint  keep y   0x10004744 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:66\n"
23      hw breakpoint  keep y   0x10004744 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:66
-> ~"24      hw breakpoint  keep y   0x1000474a in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:67\n"
24      hw breakpoint  keep y   0x1000474a in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:67
-> ~"25      hw breakpoint  keep y   0x10004750 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:68\n"
25      hw breakpoint  keep y   0x10004750 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:68
-> ~"26      hw breakpoint  keep y   0x10004756 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:69\n"
26      hw breakpoint  keep y   0x10004756 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:69
-> ~"27      hw breakpoint  keep y   0x1000475c in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:70\n"
27      hw breakpoint  keep y   0x1000475c in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:70
-> ~"28      hw breakpoint  keep y   0x10004762 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:71\n"
28      hw breakpoint  keep y   0x10004762 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:71
-> ~"29      hw breakpoint  keep y   0x10004768 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:72\n"
29      hw breakpoint  keep y   0x10004768 in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:72
-> ~"30      hw breakpoint  keep y   0x1000476e in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:73\n"
30      hw breakpoint  keep y   0x1000476e in main at /Users/hdm/mtw35/Dual-CPU_Empty_PSOC6_App/proj_cm4/main.c:73
```

When I try to single-step, I see the normal crash that GDB has. But before I tried that, I was able to list variables, examine the stack, etc. Execution is also where it went sideways and complains about breakpoint 30 which it earlier said was fine. Well No. 29 is also a bad one but it barfed before it could complain about that.
```
-exec-next --thread 1
-> 112^running
-> *running,thread-id="all"
mi2.status = running
-> &"Warning:\n"
Warning:
-> &"Cannot insert hardware breakpoint 30.\n"
Cannot insert hardware breakpoint 30.
-> &"Could not insert hardware breakpoints:\n"
Could not insert hardware breakpoints:
-> &"You may have requested too many hardware breakpoints/watchpoints.\n"
You may have requested too many hardware breakpoints/watchpoints.
-> &"\n"

-> 112^error,msg="Command aborted."
Internal Error? Multiple results or no handler for query token '112'
Command aborted.
```

recursivenomad, 2025-11-10:
> I kept adding more breakpoints beyond the limit and no complaints from gdb at all.

With no limits set (of any kind), this is normal behaviour in my experience.  GDB doesn't know there is a problem until you "continue".  Once you continue, openOCD (or other gdb-server) will attempt to set breakpoints, but then receive a bad response from the target.  So only once you "continue" is when I would expect to see an error.

> I was able to list variables, examine the stack, etc.

I think this is consistent with the above; I don't believe (by default) GDB will send the excessive breakpoints to the gdb-server until you "continue", so until that point you should be able to query & examine the stack without issue; only to error out later.

> complains about breakpoint 30 which it earlier said was fine.

I have found that the breakpoint GDB will complain about when you "continue" changes, perhaps based on the location of current execution.

Let me know if any of that helps.