Marus/cortex-debug #981: Missing SWO Timestamp in logfile

J-Rios, 2024-02-10:
Hello,

Same as in issue:
["Missing RTT Timestamp in Logfile"](https://github.com/Marus/cortex-debug/issues/740)

Now I notice it is also happening for SWO. If you setup to use SWO with timestamp and log to file, the timestamp info is not store in the log file.

Best Regards.

J-Rios, 2024-02-28:
Created [Pull Request](https://github.com/Marus/cortex-debug/pull/991) with a fix for this.

linuxguy123, 2024-03-12:
Does this mean that cortex-debug can now receive data from STLink devices via SWO ?

Are you aware that the STLink driver (https://github.com/stlink-org/stlink) has a new driver called st-trace that allows SWO communication ?  st-util does not.    How would I set up uScope to use st-trace ?

closed 2024-07-06 by haneefdm