Marus/cortex-debug #982: Unable to use st-util under linux. Cortex-Debug tries to use the st-util as if it is a windows 'st-link' gdb server

daantimmer, 2024-02-12:
### Please make you search through our existing [issues](https://github.com/Marus/cortex-debug/issues?q=type:issue) (both open and closed)
It may help to look at these instructions in `Preview` mode. Please visit the correct repo to file an issue. If this is an issue with
* Peripherals/SVD Viewer: [https://github.com/mcu-debug/peripheral-viewer](https://github.com/mcu-debug/peripheral-viewer)
* Memory Viewer: [https://github.com/mcu-debug/memview](https://github.com/mcu-debug/memview)
* RTOS Viewer: [https://github.com/mcu-debug/rtos-views](https://github.com/mcu-debug/rtos-views)
* the debugger itself, continue below

Please read our documentation as well. You have a lot of control over how Cortex-Debug works. Besides the top level [README.md](https://github.com/Marus/cortex-debug/blob/master/README.md), we have the following

https://github.com/Marus/cortex-debug/wiki
https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood
https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

*Finally, make sure all your external tools are configured properly and working. We print all the commands in the Debug Console. Most of the bug reports are from Linux users with improperly installed GNU tools. If you can't run those tools, neither can we.*

Thank you for helping reduce the number of issues which is becoming overwhelming as most of them are not issues at all. You can delete all the above text and start filing the issue

**Describe the bug**
I am using Visual Studio Code with a (linux) devcontainer on windows. We are able to bind our usb devices directly to WSL through usbipd-win. This works flawlessly for a j-link. I am now trying to create a setup for st-links instead.

I've downloaded the st-link sources and build from source to get them installed (1.8 version).

I've setup my launch configuration like:

```
            "servertype": "stlink",
            "serverpath": "/usr/local/bin/st-util",
```

But, this results in the following way that st-util is executed:

```
Setting GDB-Server CWD: /usr/local/bin
Launching gdb-server: /usr/local/bin/st-util -p 50000 -cp /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin --swd --halt
```

Which is an incorrect, this is the help menu from st-util:

```
st-util --help
st-util - usage:

  -h, --help            Print this help
  -V, --version         Print the version
  -vXX, --verbose=XX    Specify a specific verbosity level (0...99)
  -v, --verbose         Specify generally verbose logging
  -p 4242, --listen_port=1234
                        Set the gdb server listen port. (default port: 4242)
  -m, --multi
                        Set gdb server to extended mode.
                        st-util will continue listening for connections after disconnect.
  -n, --no-reset, --hot-plug
                        Do not reset board on connection.
  -u, --connect-under-reset
                        Connect to the board before executing any instructions.
  -F 1800k, --freq=1M
                        Set the frequency of the SWD/JTAG interface.
  --semihosting
                        Enable semihosting support.
  --serial <serial>
                        Use a specific serial number.
```

There is no need to set the the cube programmer, or `--swd` or `--halt`.

**To Reproduce**
Steps to reproduce the behavior:
1. install stlink-tools in a ubuntu devcontainer
2. configure a launch command with `servertype`: `stlink`
3. configure the `serverpath`: to whereever st-util is installed
4. start a debug session

**Expected behavior**
able to start debugging through st-util

**Environment (please complete the following information):**
- cortex-debug: 1.12.1
- os: (host) windows 10, (devcontainer) ubuntu 22.04
- gdb-multiarch: 12.1-0ubuntu1~22.04
- st-util: 1.8.0-2-g65f062f (build/installed from source)

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
        {
            "name": "Debug stm32g474re stlink",
            "cwd": "${workspaceFolder}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "device": "STM32G474RE",
            "servertype": "stlink",
            "serverpath": "/usr/local/bin/st-util",
            "runToEntryPoint": "main",
            "svdFile": "Keil::STM32G4xx_DFP@1.5.0",
            "deviceName": "STM32G474RETx"
        }
```

**Attach text from `Debug Console`**
```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from arm-none-eabi-objdump --syms -C -h -w <redacted>.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p <redacted>.elf
Launching GDB: gdb-multiarch -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Setting GDB-Server CWD: /usr/local/bin
Launching gdb-server: /usr/local/bin/st-util -p 50000 -cp /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin --swd --halt -p 50000
    Please check TERMINAL tab (gdb-server) for output from /usr/local/bin/st-util
Finished reading symbols from objdump: Time: 35 ms
ST-LINK: GDB Server Quit Unexpectedly. See gdb-server output in TERMINAL tab for more details.
```

**Attach text from `Terminal**
```[2024-02-12T08:26:04.033Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/usr/local/bin/st-util -p 50000 -cp /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin --swd --halt -p 50000
/usr/local/bin/st-util: invalid option -- 'c'
Invalid port /usr/local/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin
[2024-02-12T08:26:04.070Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

**Additional context**
Install st-link tools from source did not work when I followed the build instructions. What I did instead:

```
1. make a temp dir
2. git clone the st-link repo: https://github.com/stlink-org/stlink.git
3. install the required dependencies (https://github.com/stlink-org/stlink/blob/testing/doc/compiling.md)
4. this is where i diverged:
5. `make clean`
6. `make release` (this fails, but is required for step 7 as it configures the cmake project)
7. `cmake --build build/Release` (this does work)
8. `cmake --install build/Release`
9. `ldconfig`
```

daantimmer, 2024-02-12:
There is a workaround:

launch `st-util` manually and change `servertype` to `external`:
```
        {
            "name": "Debug stm32g474re stlink",
            "cwd": "${workspaceFolder}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "device": "STM32G474RE",
            "servertype": "external",
            "gdbTarget": "localhost:4242",
            "overrideLaunchCommands": [
                "monitor reset",
                "-target-download",
                "monitor reset"
            ],
            "overrideRestartCommands": [
                "monitor reset"
            ],
            "runToEntryPoint": "main",
            "svdFile": "Keil::STM32G4xx_DFP@1.5.0",
            "deviceName": "STM32G474RETx"
        }
```

afezza, 2024-06-21:
If you use `stutil` instead of `stlink` as server type it will do the trick

` "servertype": "stutil"`

haneefdm, 2024-06-21:
The following is wrong

```
            "servertype": "stlink",
            "serverpath": "/usr/local/bin/st-util",
```

`stlink` and `st-util` are very different gdb-servers, incompatible, supplied by different people. `stlink` from ST Micro is my preferred gdb-server.

closed 2024-06-21 by haneefdm

daantimmer, 2024-06-21:
Thanks for responding, and pointing me at an error on my side!. However, I can't seem to find this `stlink` tool. (Specifically for Linux).

The only thing that I can find is https://github.com/stlink-org/stlink, which, when installed, only installs:
- st-flash
- st-info
- st-util
- st-trace

haneefdm, 2024-06-21:
Like I said, stlink is distributed by ST Micro and is part of their install of STMCube32 IDE.

https://www.st.com/en/development-tools/stm32cubeide.html

If you install in the default location, Cortex-Debug will find it automatically. You can also install STMicro's VSCode extension that will auto-generate the launch.json for you

https://community.st.com/t5/stm32-vscode-extension-mcus/new-stm32-vscode-extension-v2-0-0-released/td-p/657415

There are hundreds of examples on the web.

https://stackoverflow.com/questions/71066091/problem-understanding-cortex-debug-configurations-in-visual-studio-code-stm32
https://forum.electro-smith.com/t/st-link-and-cortex-debugger-on-ubuntu-24-04/5260

But it all starts by installing the official ST Micro tools. You are taking a scenic route to getting this done. While we don't discourage that, we don't recommend it either.

daantimmer, 2024-06-22:
Right, we don't depend on CubeIDE _because_ we are in a devcontainer (you can call it scenic, we call it modern).
We target multiple devices from multiple vendors and our container supports all of them with the same tools :-).

If you search for "stlink Ubuntu" you get pointed at the st-utils repository and not at CubeIDE. There is no mention of Cube whatsoever.

Regardless, thank you for the information regarding `stlink` and CubeIDE. That was information that I missed. Also thank you for taking time to respond and pointing me at my error/mistake in a friendly manner :-)