Marus/cortex-debug #992: debug error

jieBieber, 2024-03-04:
I hope to use the Cortex-debug tool to debug STM32. My system is Ubuntu 20.04 and I have chosen gdb-multiarch as the debugging too.

The content of my launch.json file is as follows：
```json
       {
            "cwd": "${workspaceRoot}",
            "executable": "./build/FreeRtos_Hao.elf",
            "name": "Debug Microcontroller",
            "request": "launch",
            "type": "cortex-debug",
            "interface": "swd",
            "runToEntryPoint": "main",
            "servertype": "openocd",
            "configFiles": [
                "/usr/share/openocd/scripts/interface/stlink-v2.cfg",
                "/usr/share/openocd/scripts/target/stm32f1x.cfg"
            ],
            "armToolchainPath": "/usr/bin/",
            "gdbPath": "/usr/bin/gdb-multiarch",
            "showDevDebugOutput": "raw"
        }
```

After clicking debug, an error message appears as follows：
```c
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "cwd": "/home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS",
    "executable": "/home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS/build/FreeRtos_Hao.elf",
    "name": "Debug Microcontroller",
    "request": "launch",
    "type": "cortex-debug",
    "interface": "swd",
    "runToEntryPoint": "main",
    "servertype": "openocd",
    "configFiles": [
        "/usr/share/openocd/scripts/interface/stlink-v2.cfg",
        "/usr/share/openocd/scripts/target/stm32f1x.cfg"
    ],
    "armToolchainPath": "/usr/bin/",
    "gdbPath": "/usr/bin/gdb-multiarch",
    "showDevDebugOutput": "raw",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "searchDir": [],
    "toolchainPath": "/usr/bin/",
    "toolchainPrefix": "arm-none-eabi",
    "extensionPath": "/home/topeet/.vscode/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "fe91856c-764c-410f-af8b-d9b3ee08880a",
    "pvtShowDevDebugOutput": "raw"
}
Reading symbols from /usr/bin/arm-none-eabi-objdump --syms -C -h -w /home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS/build/FreeRtos_Hao.elf
Reading symbols from /usr/bin/arm-none-eabi-nm --defined-only -S -l -C -p /home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS/build/FreeRtos_Hao.elf
Launching GDB: /usr/bin/gdb-multiarch -q --interpreter=mi2
1-gdb-version
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS -f /home/topeet/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f1x.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
Finished reading symbols from objdump: Time: 64 ms
Finished reading symbols from nm: Time: 82 ms
-> =thread-group-added,id="i1"
-> ~"GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2\n"
-> ~"Copyright (C) 2020 Free Software Foundation, Inc.\n"
-> ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
-> ~"\nType \"show copying\" and \"show warranty\" for details.\n"
-> ~"This GDB was configured as \"x86_64-linux-gnu\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
2-gdb-set mi-async on
-> 2^done
3-interpreter-exec console "set print demangle on"
-> 3^done
4-interpreter-exec console "set print asm-demangle on"
-> =cmd-param-changed,param="print asm-demangle",value="on"
-> 4^done
5-enable-pretty-printing
-> 5^done
6-interpreter-exec console "source /home/topeet/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdbsupport.init"
-> 6^done
7-interpreter-exec console "source /home/topeet/.vscode/extensions/marus25.cortex-debug-1.12.1/support/gdb-swo.init"
-> =cmd-param-changed,param="language",value="c"
-> =cmd-param-changed,param="language",value="auto"
-> 7^done
8-interpreter-exec console "set output-radix 0xa"
-> ~"Output radix now set to decimal 10, hex a, octal 12.\n"
Output radix now set to decimal 10, hex a, octal 12.
-> 8^done
9-interpreter-exec console "set input-radix 0xa"
-> ~"Input radix now set to decimal 10, hex a, octal 12.\n"
Input radix now set to decimal 10, hex a, octal 12.
-> 9^done
10-file-exec-and-symbols "/home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS/build/FreeRtos_Hao.elf"
-> 10^done
Failed to launch OpenOCD GDB Server: Timeout.
```

No matter how I change the configuration file, there will always be an error message`Failed to launch OpenOCD GDB Server: Timeout`.
But what surprised me was that it could be debugged and downloaded normally from the command line.

I am very much looking forward to receiving your help. Thank you！

haneefdm, 2024-03-04:
```
Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS -f /home/topeet/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f1x.cfg
    Please check TERMINAL tab (gdb-server) for output from openocd
```

Well, did you check the TERMINAL tab to see why openocd is not starting? Also make sure you can launch OpenOCD from the command line just like we do. Every command we run, you can also run from the command line to make sure your GNU tools are properly installed, although that does not seem to be the problem.

jieBieber, 2024-03-05:
> ```
> Launching gdb-server: openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS -f [...]

hi,
Output the content on the tab(gdb-server) page : 

```c
[2024-03-05T11:53:43.118Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/topeet/Desktop/gcc_stm32-master/gcc_stm32-master/Template/STM32F103_FreeRTOS -f /home/topeet/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f1x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDLiveWatchSetup
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v38 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.231196
Error: init mode failed (unable to connect to the target)
in procedure 'init' 
in procedure 'ocd_bouncer'

[2024-03-05T11:53:43.916Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

I can't understand what the output here means? I think this is a normal output. What do you think？

kossnikita, 2024-03-05:
It looks like you just connected your stlink to the device incorrectly. Check connection first and retry with OpenOCD standalone

```bash
openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f1x.cfg
```

You should have no errors. The extension has no effect on this.

jieBieber, 2024-03-05:
> 看起来您只是将stlink错误地连接到了设备。首先检查连接，然后使用OpenOCD单机版重试
> 
> ```shell
> openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f1x.cfg
> ```
> 
> 你应该没有错误。扩展对此没有影响。

Excuse me, is this command being used on the command line? I can connect to stm32 using this command on the command line. I have been really troubled for many days and I really hope to get your help。

The result after executing the command is:
```shell
topeet@ubuntu:~$ openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f1x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v38 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.231196
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints

```

haneefdm, 2025-01-17:
Please use proper version of openocd and proper versions of stling config files. These are released by ST and I don't trust what get from the internet or opt-get, etc. They are generally old and wrong.

closed 2025-01-17 by haneefdm