Marus/cortex-debug #997: "serialNumber" property located in local user settings instead of launch.json

anders-rasm, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
When working multiple people on the same project using SCM, it is annoying to have to resolve merge conflicts every time a new commit is made due to the serialNumber property being in launch.json.

**Describe the solution you'd like**
The "serialNumber" property should be located in user settings.

selectstriker2, 2024-03-14:
Have you tried removing the "serialNumber" property altogether? I'm using a J-Link and I've never had to include the "serialNumber" property in my launch.json

anders-rasm, 2024-03-14:
sorry, I should have noted that I have multiple debuggers connected to my computer, so I need to specify which one to use.

anders-rasm, 2024-03-19:
I found out that you can put custom settings in the user settings.json and reference it from the launch.json. Example:
In settings.json
``"serialNumber": <your serial number>``
In launch.json:
``"serialNumber": "${config:serialNumber}",``

haneefdm, 2025-01-17:
There are multiple ways to get external things into launch.json. Having it in the User settings also doesn't make sense because if I have multiple boards, I would have multiple serial numbers.

closed 2025-01-17 by haneefdm