Marus/cortex-debug #1000: Transport eclipse jlink debug configuration to vscode+cortex-debug failed.

googlehim, 2024-03-26:
**Describe the bug**
Debug configurations on eclipse works OK, now I am moving to vscode+cortex-debug; 

**Expected behavior**
both can debug my program.

**Screenshots**
Here is configuration on eclipse
[image]

Here is my launch.json
[image]

In EPC6450-AWI.gdb
[image]

googlehim, 2024-03-27:
Hi @PhilippHaefele 

Thanks for your kind help!

gdb scripts was originaly  shipped by sdk used for eclipse jlink debug, in it  connect to remote gdb server, setup pll clock and many other  hardware initial sequence; if cortex-debug don't do things this way, how I embed this gdb scripts to cortex-debug? I think the most convient way is as gdb -X args as is, like eclips do, overried cortex-debug sequence?

I tried launch gdb-server and gdb manually, here are logs, seems gdb can connect to gdb-server, however "bfd requires flen 8, but target has flen 4" error occur, I have already send "set architecture riscv:rv32" like in eclipse commands.

JLinkGDBServerCL.log
[JLinkGDBServerCL.log](https://github.com/Marus/cortex-debug/files/14766567/JLinkGDBServerCL.log)

riscv32-unknown-elf-gdb.log
[riscv32-unknown-elf-gdb.log](https://github.com/Marus/cortex-debug/files/14766560/riscv32-unknown-elf-gdb.log)

PhilippHaefele, 2024-03-28:
Mhh interesting error.

First of all it's in general ok to use gdb scripts, but there are thinks that are not allowed or make no sense (as the extension already does do them).

For your error message I found following https://doc.nucleisys.com/nuclei_sdk/faq.html -> case 3

So it seems that the binary and MCU floating point bit-width do not match. What kind of floating point unit does your MCU have?

Is there any reason why you set architecture and processor two times?

googlehim, 2024-03-28:
> First of all it's in general ok to use gdb scripts, but there are thinks that are not allowed or make no sense (as the extension already does do them).

There must be some commands in gdb scripts violate cortex-debug extension work flow, but I don't know which and how.

Since both debug methods (eclipse debugger/coretex-debug) use same toolchains, eclipse seems more flexiable and straight forward, you can see "target remote locahost:2331" in gdb scripts, can set GDBServer port. Cortex-debug extension already do something 
you should very familar with and look into it you can custom. Will you consider this?

> So it seems that the binary and MCU floating point bit-width do not match. What kind of floating point unit does your MCU have?

For error message, I have checked compile flags -march -mabi, they are correct, plus, compiled elf file can load run and debug on my hardware with eclipse jlink debug configuration. Weird error message to me.

PhilippHaefele, 2024-03-28:
Looked into the logs again and noticed that you haven't removed the gdb script from the debugger args. Please attach logs without it.

On a first guess i would have said that neither the debuggerArgs nor the preLaunchCommands are needed at all. Have you tried it with both elements removed in the launch.json?

googlehim, 2024-03-29:
I attached all logs with eclipse's for compare:

[cortex-debug(DEBUG CONSOLE).txt](https://github.com/Marus/cortex-debug/files/14798680/cortex-debug.DEBUG.CONSOLE.txt)
[cortex-debug(JLinkGDBServerCL).txt](https://github.com/Marus/cortex-debug/files/14798681/cortex-debug.JLinkGDBServerCL.txt)
[cortex-debug(launch.json).txt](https://github.com/Marus/cortex-debug/files/14798682/cortex-debug.launch.json.txt)
[eclipse(gdb traces).txt](https://github.com/Marus/cortex-debug/files/14798683/eclipse.gdb.traces.txt)
[eclipse(JLinkGDBServerCL).txt](https://github.com/Marus/cortex-debug/files/14798685/eclipse.JLinkGDBServerCL.txt)

closed 2024-04-03 by googlehim

PhilippHaefele, 2024-04-03:
Sorry that I couldn't help further. Was out of time during eastern.

How did you fix the issue?