Marus/cortex-debug #1005: Fail source level debugging Zephyr Blinky build on NUCLEO-WL55JCx

VeniaminGH, 2024-04-13:
Dear All,

Please find issue report I'm experiencing with coretx-debug v1.12.1 using it with
VSCode v1.88.1 under Ubuntu LTS v22.04 with NUCLEO-WL55JCx board
when trying to run \sample\basic\blinky demo from zephyr-sdk-0.16.5-1

**Describe the bug**

NUCLEO-WL55JCx is working fine with STM32CudeIDE, have updated STLINK debugger to V3J14M5.
Zephyr-SDK installation is working fine with FRDM-K64F boards with VSCode using MCUXpresso plugin.
I can also flash and manually GDB debug Blinky example on NUCLEO-WL55JCx board with `west`.
`west flash` and `west debug` all work are expected using "stlink".

But I can't do source level debugging using cortex-debug neither with "stlink" nor with "openocd"

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. See issue

_VSCode Terminal Output for "openocd" servertype_
```
24-04-13T20:07:43.230Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/XXXXXXX/zephyrproject/zephyr/samples/basic/blinky -f /home/XXXXXXX/.vscode/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f /home/XXXXXXX/zephyrproject/zephyr/boards/st/nucleo_wl55jc/support/openocd.cfg
[2024-04-13T20:07:45.362Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

_VSCode Termial Output for "stlink" servertype_
```
[2024-04-13T20:14:10.553Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/opt/st/stm32cubeide_1.15.0/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.1.200.202401191126/tools/bin/ST-LINK_gdbserver -p 50000 -cp /opt/st/stm32cubeide_1.15.0/plugins/com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.linux64_2.1.200.202311302303/tools/bin --swd --halt

STMicroelectronics ST-LINK GDB server. Version 7.6.0
Copyright (c) 2024, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 31
        Listen Port Number         : 50000
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled

COM frequency = 12000 kHz
Target connection mode: Default
Reading ROM table for AP 0 @0xe00fffd0
Hardware watchpoint supported by the target 
ST-LINK Firmware version : V3J14M5
Device ID: 0x497
PC: 0x8000830
ST-LINK device status: HALT_MODE
ST-LINK detects target voltage = 3.27 V
ST-LINK device status: HALT_MODE
ST-LINK device initialization OK
SwvSrv state change: 0 -> 1
Waiting for debugger connection...
Waiting for connection on port 50000...
Stm32Device, pollAndNotify running...
Waiting for connection on port 50001...
[2024-04-13T20:14:12.590Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

**Expected behavior**

I expect to be able to do source level debugging of Zephyr blinky demo on NUCLEO-WL55 board but instead I'm getting this error:
**Screenshots**
[image]

[... 326 lines cut ...]
Loading section initlevel, size 0x68 lma 0x8003518
--Type <RET> for more, q to quit, c to continue without paging--
Loading section device_area, size 0xa0 lma 0x8003580
Loading section sw_isr_table, size 0x1f0 lma 0x8003620
Loading section rodata, size 0x308 lma 0x8003810
Loading section datas, size 0x5c lma 0x8003b18
Loading section device_states, size 0x10 lma 0x8003b74
Loading section .last_section, size 0x4 lma 0x8003b84
Info : Unable to match requested speed 500 kHz, using 200 kHz
Info : Unable to match requested speed 500 kHz, using 200 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000830 msp: 0x20001040
Start address 0x08000830, load size 15240
Transfer rate: 417 bytes/sec, 1524 bytes/write.
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/XXXXXXX/zephyrproject/zephyr/build/zephyr/zephyr.elf 

Info : Unable to match requested speed 500 kHz, using 200 kHz
Info : Unable to match requested speed 500 kHz, using 200 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000830 msp: 0x20001040
```
_I'm able to execute (gdb) run and see LED blinking_

VeniaminGH, 2024-04-15:
**Update**

I have installed STM32 VS Code Extension for VSCode and tried to load Zephyr build with modified launch.json configuration
```json
			"serverpath": "${config:STM32VSCodeExtension.cubeCLT.path}/STLink-gdb-server/bin/ST-LINK_gdbserver",
			"stm32cubeprogrammer":"${config:STM32VSCodeExtension.cubeCLT.path}/STM32CubeProgrammer/bin",
			"stlinkPath": "${config:STM32VSCodeExtension.cubeCLT.path}/STLink-gdb-server/bin/ST-LINK_gdbserver",   
			"armToolchainPath": "${config:STM32VSCodeExtension.cubeCLT.path}/GNU-tools-for-STM32/bin",
			"gdbPath":"${config:STM32VSCodeExtension.cubeCLT.path}/GNU-tools-for-STM32/bin/arm-none-eabi-gdb",
			"serverArgs": ["-m","0"],

```

and it started to work. I will play more to understand what is going on. But one thing I'm certain, that Cortex-Debug is not happy with Zephyr GDB

haneefdm, 2025-01-17:
Closing this issue. There is nothing special we ask of gdb. However, we need to be able to run it. '~' chars in path names is a no-no. That char is only understood by certain shells. It is not part of VSCode or any OS.

Also, people may be conflating settings from launch.json entries. Look at the settings for this extension and you can have them at the Foler/Workspace/User levels.

closed 2025-01-17 by haneefdm