Marus/cortex-debug #1007: FreeRTOS Symbols Not Loaded when `"loadFiles"` and `"symbolFiles"` are Used

Jake-Carter, 2024-04-20:
First off - thanks for your work on this extension.  We use it at ADI for our [MSDK](https://github.com/analogdevicesinc/msdk), and it's been a much better experience than Microsoft's cpptools.

**Describe the bug**

For a FreeRTOS project, if a launch profile explicitly specifies the executable and symbol files with `"loadFiles"`/`"symbolFiles"` it will fail to find FreeRTOS debug symbols and fail to show a thread-aware call stack.  Ex:

```json
"executable": "${workspaceFolder}/build/file.elf",
"loadFiles": ["${workspaceFolder}/build/file.elf"],
"symbolFiles": [{
    "file": "${workspaceFolder}/build/file.elf"
}],
```

[image]

When _only_ `"executable"` is used, then the FreeRTOS symbols are loaded correctly and the call stack becomes thread-aware as expected.  Ex:

```json
"executable": "${workspaceFolder}/build/file.elf",
// "loadFiles": ["${workspaceFolder}/build/file.elf"],
// "symbolFiles": [{
//     "file": "${workspaceFolder}/build/file.elf"
// }],
```

[image]

**Expected behavior**

FreeRTOS symbols should be loaded correctly if `"loadFiles"`/`"symbolFiles"` are used.  We have dual-core microcontrollers with a secondary RISC-V core and we build binaries for each core separately so that the debug symbols can be loaded correctly.  

Manually using `exec-file`/`symbol-file` on the GDB command-line works (see extra context at bottom), and so does a cpptools debug profile in a similar setup.

**Environment (please complete the following information):**

 - VS Code v1.88.1
 - Cortex-Debug Version: v1.12.1
 - OS: Linux, (replicated on Windows 10 as well)
 - GDB Version: GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
 - Compiler Toolchain Version: (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

**Please include `launch.json`**

- **Note: We load values from `settings.json` (see [VSCode-Maxim project](https://github.com/analogdevicesinc/VSCode-Maxim).**

- For this project, `${config:program_file}` and `${config:symbol_file}` point to the same file.  See [here](https://github.com/analogdevicesinc/msdk/tree/main/Examples/MAX78000/FreeRTOSDemo) for the full project.
**Failed** profile:
```
        {
            "name": "Debug Arm (Cortex-debug)",
            "cwd":"${workspaceRoot}",
            "executable": "${workspaceFolder}/build/${config:program_file}",
            "loadFiles": ["${workspaceFolder}/build/${config:program_file}"],
            "symbolFiles": [{
                "file": "${workspaceFolder}/build/${config:symbol_file}"
            }],
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "linux": {
                "gdbPath": "${config:ARM_GCC_path}/bin/arm-none-eabi-gdb",
                "serverpath": "${config:OCD_path}/openocd",
            },
            "windows": {
                "gdbPath": "${config:ARM_GCC_path}/bin/arm-none-eabi-gdb.exe",
                "serverpath": "${config:OCD_path}/openocd.exe",
            },
            "osx": {
                "gdbPath": "${config:ARM_GCC_path}/bin/arm-none-eabi-gdb",
[... 589 lines cut ...]
[New Thread 536876256]
[New Thread 536880416]

Thread 3 "IDLE" received signal SIGINT, Interrupt.
[Switching to Thread 536881056]
prvIdleTask (pvParameters=<optimized out>) at Source/tasks.c:3467
3467                if( listCURRENT_LIST_LENGTH( &( pxReadyTasksLists[ tskIDLE_PRIORITY ] ) ) > ( UBaseType_t ) 1 )
(gdb) info thread
  Id   Target Id                                            Frame 
* 3    Thread 536881056 "IDLE" (Name: IDLE, State: Running) prvIdleTask (pvParameters=<optimized out>)
    at Source/tasks.c:3467
  4    Thread 536877408 "TickTock" (Name: TickTock)         xTaskDelayUntil (
    pxPreviousWakeTime=pxPreviousWakeTime@entry=0x2000193c <ucHeap+2388>, 
    xTimeIncrement=xTimeIncrement@entry=10000) at Source/tasks.c:1293
  5    Thread 536875616 "Task0" (Name: Task0)               xTaskDelayUntil (
    pxPreviousWakeTime=pxPreviousWakeTime@entry=0x2000123c <ucHeap+596>, 
    xTimeIncrement=xTimeIncrement@entry=1000) at Source/tasks.c:1293
  6    Thread 536876256 "Task1" (Name: Task1)               xTaskDelayUntil (
    pxPreviousWakeTime=pxPreviousWakeTime@entry=0x200014bc <ucHeap+1236>, 
    xTimeIncrement=xTimeIncrement@entry=1000) at Source/tasks.c:1293
  7    Thread 536880416 "CmdLineTask" (Name: CmdLineTask)   ulTaskGenericNotifyTake (
    uxIndexToWait=uxIndexToWait@entry=0, xClearCountOnExit=xClearCountOnExit@entry=1, 
    xTicksToWait=xTicksToWait@entry=4294967295) at Source/tasks.c:4730
(gdb) 
```

whosht, 2024-07-16:
+1
Currently I use another launcher which just attaches to the already flashed target with the ELF-binary as executable.

haneefdm, 2024-07-16:
I think I understand what is going on. OpenOCD (any gdb-server) gets one chance to query gdb for symbol info and that is at the very start of the session. We are not involved in providing this info. In OpenOCD, there are no re-tries if the symbols change, because it doesn't even know that such a thing occurred.

But, if we are loading the elf files very late during the debug cycle and can do something about it, then we will. We may have a chicken and egg situation. I also see this as a common problem and I am sure there are solutions out there. Hope we get lucky and stumble across the solution. Most of this is not documented well btw.

whosht, 2024-07-16:
@haneefdm: nice
I forgot to note: This also happens with J-Link (debug server).

haneefdm, 2024-07-16:
Good to know. *If/When* this gets fixed, I will need help testing it. I will report back here.

This will be a large enough (dangerous) change that it may require a [pre-release](https://github.com/Marus/cortex-debug?tab=readme-ov-file#release-versioning) so it will not destroy development environments for half a million users.

haneefdm, 2025-01-16:
This is actually an issue with `gdb`. We use `add-symbol-file` but that does not provide the same info to gdb-servers as when we use `symbol-file`. Silly and undocumented. This means that you cannot use anything other than an offset when specifying symbol files.

A fix is being tested and RTOS detection should work so long as you don't use other fancy options.

I will be providing a patched build for testing tomorrow.

haneefdm, 2025-01-17:
A fix is available here https://github.com/Marus/cortex-debug/releases  [1.13.0-pre3](https://github.com/Marus/cortex-debug/releases/tag/v1.13.0-pre3)

I will be closing this, but if you are still having issues, we can reopen it.

whosht, 2025-01-20:
> Hi,
> 
> Thanks a lot for your solution. I tried, but I currently use: boot + signed app on my project (both are compiled independently, each has the main function). The application signing is done with an external script and so I have to load 2 applications in my launch.json :
> 
> ```
> "loadFiles": [
> "bootloader.elf",
> "signed_application.mot"
> ],
> "symbolFiles": [
> "bootloader.elf",
> "application.elf"
> ],
> "runToEntryPoint": "main",
> ```
> 
> By doing this, I was previously able to add a breakpoint on both applications, but the Freertos thread-aware was KO. With your update, I can only debug the application file (I can't add breakpoint on my bootloader). Do you think this is normal, do you know a way to debug both applications, with the Freertos thread-aware?
> 
> Thanks in advance,
> 
> Sam

Looks like one thing was fixed by breaking another.
In gdb you need to load symbols using `add-symbol-file` to add/append additional symbols. AFAIR using `symbol-file` clears all symbols previously loaded.
Maybe try `preLaunchCommands` in your launch configuration and use `add-symbol-file` and check if that helps.

sgouraud, 2025-01-20:
Thank you for your quick response. 

As I understand, I therefore used : 

```
            "executable": "application.elf",     // error without: unable to find executable file at undefined.
            "loadFiles": [
                "bootloader.elf",
                "signed_application.mot"
            ],
            "preLaunchCommands": [
                "add-symbol-file bootloader.elf",
                "add-symbol-file application.elf"
            ],
```
This help, and I can now add breakoint on both code.

But I didn't mentionned that both bootloader & application are using Freertos, freertos stack-aware seems to not be reloaded when I jump in application code (Only freertos task from bootloader seems to be loaded, and is still display on call stack windows when the final application is running. Moreover, if no breakpoint is set on bootlader, task aren't displayed on call stack windows).

haneefdm, 2025-01-20:
@sgouraud Yes, you can only get thread info for one item. This is because FreeRTOS can query symbol info just once at the beginning when gdb connects to the gdb server. I don't the FreeRTOS will ever requery gdb. It would not know when to do it either. In most instances, while you can load as many images as you want, you can only get thread info for one executable.

Bootloader is just FW like your main application. It is nothing special.

haneefdm, 2025-01-20:
@whosht is right. symbol-file is erasing any previous symbols. But add-symbol-file is not reading symbols at all and there are no threads (original problem) as far as gdb-server (openocd, JLink, etc.) are concerned.

Back to experimentation and re-opening this issue. Not sure what the solution is yet.

haneefdm, 2025-04-14:
Can you test and give feedback with the latest build

https://github.com/Marus/cortex-debug/releases

I am closing this for now, I can reopen if there are still issues

closed 2025-04-14 by haneefdm