Marus/cortex-debug #1008: Problems with ST-Link multicore debugging of STM32H755ZiQ

i-ilak, 2024-04-21:
I'm having trouble setting up the extension for multicore debugging of the STM32H755ZiQ. I don't think the extension has a bug, I'm more looking for hints on how to debug my issues and adjust the configuration to work correctly...

My reference here is [your wiki page](https://github.com/Marus/cortex-debug/wiki/Multi-core-debugging#st-link-notes) regarding multicore debugging and [this issue](https://github.com/Marus/cortex-debug/issues/837). Debugging the CM7 works just fine, but the attaching the CM4 core crashes the gdb-server immediately. 

I saw this statement on your wiki page
> If you get `Target unknown error 32` on the launch of the second gdbserver, it means the Cortex-M4 did not have time to start the Cortex-M0+.

But unfortunately no hints on how to try to debug this. Would appreciate any pointers on how to debug the issue.

**Expected behavior**
Launch multi-core debug session

**Environment (please complete the following information):**

 - Cortex-Debug Version: v1.12.1
 - OS: macOS, 14.3.1
 - GDB Version: ST-LINK_gdbserver 7.5.0
 - Compiler Toolchain Version: arm-none-eabi 13.2.rel1
 
 **Launch.json**

```json
{
    "version": "0.2.0",
    "configurations": [
      {
        "name": "Debug CM7 - ST-Link",
        "cwd": "${workspaceFolder}",
        "type": "cortex-debug",
        "executable": "${workspaceFolder}/build/CM7_firmware.elf",
        "loadFiles": [
            "${workspaceFolder}/build/CM7_firmware.hex",
            "${workspaceFolder}/build/CM4_firmware.hex",
        ],
        "request": "launch",        
        "servertype": "stlink",
        "device": "STM32H755ZI",
        "interface": "swd",
        "runToEntryPoint": "main",
        "svdFile": "${workspaceFolder}/STM32H755_CM7.svd",
        "v1": false,
        "showDevDebugOutput": "both",
        "serverArgs": [
            "-l", "1",
            "-m", "0",
            "-k",
            "-t","-s"
        ],
    },
    {
        "name": "Debug CM7+CM4 - ST-Link",
        "cwd": "${workspaceFolder}",
        "type": "cortex-debug",
        "executable": "${workspaceFolder}/build/CM7_firmware.elf",
        "loadFiles": [
            "${workspaceFolder}/build/CM7_firmware.elf",
            "${workspaceFolder}/build/CM4_firmware.elf",
        ],
        "request": "launch",
        "servertype": "stlink",
        "device": "STM32H755ZI",
        "interface": "swd",
        "serialNumber": "",
        //"runToEntryPoint": "main",
        "svdFile": "${workspaceFolder}/STM32H755_CM7.svd",
        "v1": false,
        "showDevDebugOutput": "both",
        "serverArgs": [
            "-l", "1",
            "-m", "0",
[... 182 lines cut ...]
```
[2024-04-21T16:44:16.454Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.macos64_2.1.100.202310302101/tools/bin/ST-LINK_gdbserver -p 50007 -cp /Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.macos64_2.1.100.202311100844/tools/bin --swd --attach --halt -l 1 -m 1 -t -s

STMicroelectronics ST-LINK GDB server. Version 7.5.0
Copyright (c) 2023, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 1
        Listen Port Number         : 50007
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled

Info : default port : 7184
Info : Remote address: 127.0.0.1
Info : stlinkserver already running, exit
Target unknown error 32

Error in initializing ST-LINK device.
Reason: Unknown. Please check power and cabling to target.
[2024-04-21T16:44:17.325Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

lowkin18, 2024-06-13:
Were you able to resolve this?

I would also like to debug from vscode the STM32H755ZIT but cannot get to main of the application or anywhere in the application.

get warning "Could not step over: Cannot find bounds of current function (from exec-next --thread 1)

Delzye, 2024-08-27:
Hi, I am using an STM32H755 with the extension, I needed to change "-m", "1" in the CM4 config to "-m", "3". As far as I understood, the CM4 is on port 3.

haneefdm, 2024-08-27:
```
Target unknown error 32

Error in initializing ST-LINK device.
Reason: Unknown. Please check power and cabling to target.
```
Well, you are getting this error from ST-LINK on startup. You see `everything` in that window, so please investigate why this happening. Ask ST, StackOverflow, etc.

People are mixing up issues here. There can be hundreds of reasons and just saying "me too" is not help unless the issue is EXACTLY the same as the original issue

i-ilak, 2024-09-01:
Sorry for the long silence, was not working on with this particular board for a while.

As @Delzye mentioned, the issue here was the `-m` flag for the second core. Once you set that to `3` everything works just fine... 

I will close the issue as resolved, hope this is fine.

closed 2024-09-01 by i-ilak