Marus/cortex-debug #1011: gdb and non-ASCII symbols in filenames problems (again)

Gadicuz, 2024-04-28:
As mentioned [here](https://github.com/Marus/cortex-debug/issues/207#issue-496183590) and [here](https://github.com/eclipse-embed-cdt/eclipse-plugins/issues/581#issue-1799598867) there is a problem with non-ASCII symbols in filenames (vscode installation or project name) on win32 platform.

The problem is that GDB can't execute init scripts because it can't find files whose names seem to be correct. There are some workarounds though.

I have the same problem and I'd like to propose **a real solution**.

<details>
<summary>TL;DR</summary>
<pre>
 - stop blame gdb
 - use system local encoding (win1251, win1252 and so on) to encode/decode command exchange with gdb
 - required encoding could be optional and might be configured in launch.json
</pre>
</details>

Prerequisites:
- platform = win32
- non-ASCII character(s) in a file path (extension path or project path)

Fact: **gdb uses system default encoding for all scripting stuff**
_This is win1251 in my case._

Let's consider the following code (mimics cortex-debug):

```js
var spawn = require('child_process').spawn;
var iconv = require('iconv-lite');

var encoding = 'win1251'; // specify your windows default encoding (gdb uses it for scripting)
var dcd = (data) => iconv.decode(data, encoding);  // transcode incoming data
var raw = (data) => data.toString();  // use raw() to test skipping transcoding incoming data

var gdb = spawn("gdb.exe");
gdb.stdout.on('data', (data) => console.log(dcd(data)));
gdb.stderr.on('data', (data) => console.error(dcd(data)));

var file0 = "file0-ascii.txt";     // ASCII-only filename (exists)
var file1 = "file1-ascii.txt";     // ASCII-only filename (exists)
var file2 = "file2-неаски.txt";    // non-ASCII filename (exists)
var file3 = "file3-нетути.txt";    // non-ASCII filename (doesn't exist)

// use utf-8 encoding
gdb.stdin.write(`interpreter-exec console "source ${file0}"\n`);  // [A]
gdb.stdin.write(`interpreter-exec console "source ${file1}"\n`);  // [B]
gdb.stdin.write(`interpreter-exec console "source ${file2}"\n`);  // [C]
// transcode to gdb-friendly encoding
gdb.stdin.write(iconv.encode(`interpreter-exec console "source ${file2}"\n`, encoding)); // [D]
gdb.stdin.write(iconv.encode(`interpreter-exec console "source ${file3}"\n`, encoding)); // [E]
gdb.stdin.end();
```
Also we have three files (all 'win1251' content encoded, attached below):

<details><summary>file0-ascii.txt</summary><pre>
echo <<< Hello ASCII world >>>
</pre></details>

<details><summary>file1-ascii.txt</summary><pre>
echo <<< У меня есть не-ASCII друг (I have non-ASCII friend) >>>
source file2-неаски.txt
</pre></details>

<details><summary>file2-неаски.txt</summary><pre>
echo <<< Привет не-ASCII мир! >>>
</pre></details>

Running the code we get output:
```
...
For help, type "help".
Type "apropos word" to search for commands related to "word".
[A](gdb) <<< Hello ASCII world >>>
[B](gdb) <<< У меня есть не-ASCII друг (I have non-ASCII friend) >>><<< Привет не-ASCII мир! >>>
[C](gdb)
file2-РЅРµР°СЃРєРё.txt: No such file or directory.
error in command: "source file2-РЅРµР°СЃРєРё.txt".
[D](gdb) <<< Привет не-ASCII мир! >>>
[E](gdb)
file3-нетути.txt: No such file or directory.
error in command: "source file3-нетути.txt".
(gdb) 
```

 - Command [A]: executes simple command file "file0" with ASCII name, no problem at all.
 - Command [B]: executes another command file "file1" with ASCII name. The file outputs some non-ASCII text and executes non-ASCII named file. No problems, everything works!!! If one replaces dcd() with raw() non-ASCII text output will corrupt but all scripts keep to work.

This proves gdb internaly uses local system encoding to work with script files and their contents. Let's run "file2" with interpreter by command, the same file we just ran from within "file1".
 - Command [C]: executes "file2" with non-ASCII name directly and failes. gdb failed to find the file and there is the reason.
<details>
  <summary>Explanation</summary>
<pre>
 non-ASCII part of file2 name is "неаски" = "\u{043D}\u{0435}\u{0430}\u{0441}\u{043A}\u{0438}"
 utf-8 encoding is                          "\xD0\xBD\xD0\xB5\xD0\xB0\xD1\x81\xD0\xBA\xD0\xB8" 
 written to stdin as utf-8 encoding         "\xD0\xBD\xD0\xB5\xD0\xB0\xD1\x81\xD0\xBA\xD0\xB8" (unchanged)
        gdb scripting uses system default encoding so every byte in a string abowe is a symbol, hence failed to locate file
 returned in error message as               "\xD0\xBD\xD0\xB5\xD0\xB0\xD1\x81\xD0\xBA\xD0\xB8" (unchanged)
        by default input stream is decoded to string using utf-8, so filename with raw() transcoding seems ok, BUT...
        ... the real encoding from gdb (including this error message) is system default, so we need to
 transcode name from win1251 bytes          "  Р   Ѕ   Р   µ   Р   °   С   Ѓ   Р   є   Р   ё"  (win1251 decoded)
      This is **the** filename gdb tried to find due to encoding mismatch,
      gdb performs the same actions (does interpret every byte as a filename character, uses win1251) so 
      filenames "file2-неаски.txt" and "file2-РЅРµР°СЃРєРё.txt" are not the same.
</pre>
</details>

Let's encode "file2" name with win1251 and try again.
 - Command [D]: same as command [C] but filename is encoded with local encoding (gdb-friendly). Works perfectly!
 - Command [E]: encoded filename of unexisting file - failed with correct message.

### Conclusion
Use system default encoding to exchange command data with gdb and everything will work!!!

[file0-ascii.txt](https://github.com/Marus/cortex-debug/files/15142374/file0-ascii.txt)
[file1-ascii.txt](https://github.com/Marus/cortex-debug/files/15142375/file1-ascii.txt)
[file2-неаски.txt](https://github.com/Marus/cortex-debug/files/15142376/file2-.txt)

haneefdm, 2024-04-28:
It is not just launching gdb. It involves a whole bunch of other things like source file names, etc. in breakpoints, disassembly, etc.

Not going to fix.

closed 2024-04-28 by haneefdm

Gadicuz, 2024-04-28:
I still have gdb start issue only to be honest. As far as I can see most issues are solved with ```\ooo``` escaping of non-ASCII symbols.

The only tiny (minor, nonblocking) issue I see is: gdb is unable to list source code **in CLI terminal** on breakpoint in non-ASCII named source file. UI works fine anyway.

For me it's negligible, I don't use terminal. Just saying.

For example:
```
Breakpoint 1, main () at C:/Projects/test/src/main.c:39
39	    scProcess(&scCntx);
continue
Continuing.
halted: PC: 0x08003572
Breakpoint 2, t () at C:/Projects/test/src/майн.c:2             -- non-ASCII file
2	C:/Projects/test/src/майн.c: No such file or directory. -- ERROR
continue
Continuing.
halted: PC: 0x0800358e
Breakpoint 1, main () at C:/Projects/test/src/main.c:39
39	    scProcess(&scCntx);
continue
Continuing.
halted: PC: 0x08003572
Breakpoint 2, t () at C:/Projects/test/src/майн.c:2
2	C:/Projects/test/src/майн.c: No such file or directory.
halted: PC: 0x0800358e
```
gdb is going to list some source lines, but can't open non-ASCII source file.

As I said this is minor issue if any. But it can be fixed.

gdb uses sym table to get source file name (it's stored in utf-8) and then try to open file (using winXXXX encoding) with no transcoding at all (this is a HUGE mistake). This can't succeed.

One can use ```substitute-path``` to cope with this tiny issue. They can replace utf-8 file name with correct winXXXX encoded one.
"\354\340\351\355" is win1251 encoded "майн"
"\320\274\320\260\320\271\320\275" is utf-8 encoded "майн"
```
set substitute-path C:/Projects/test/src/майн.c C:/Projects/test/src/\354\340\351\355.c  (for utf-8 console)
set substitute-path C:/Projects/test/src/\320\274\320\260\320\271\320\275.c C:/Projects/test/src/майн.c (for win1251 console)
continue
Continuing.
halted: PC: 0x08003572
Breakpoint 2, t () at C:/Projects/test/src/майн.c:2
2	  return 50;					-- it works now!!!
continue
Continuing.
halted: PC: 0x0800358e
Breakpoint 1, main () at C:/Projects/test/src/main.c:39
39	    scProcess(&scCntx);
continue
Continuing.
halted: PC: 0x08003572
Breakpoint 2, t () at C:/Projects/test/src/майн.c:2
2	  return 50;
```
This substitution fixes tiny issue but breaks UI :)
gdb correctly escapes with ```\ooo``` all non-ASCII symbols, but current version of the App interprets them wrong.
```
-> 71^done,threads=[{
    id="1",
    target-id="Remote target",
    frame={
      level="0",
      addr="0x0800358c",
      func="t",
      args=[],
      file="C:/Projects/test/src/\320\274\320\260\320\271\320\275.c",  - filename form sym table, utf-8 encoded
      fullname="C:\\Projects\\test\\src\\\354\340\351\355.c",  - substituted name, win1251 encoded (used by gdb for listing)
      line="2",
      arch="armv3m"
    },
    state="stopped"
  }]
GDB -> App: {
   "output":"","token":71,"outOfBandRecord":[],
   "resultRecords":{
     "resultClass":"done",
     "results":[
       ["threads",[[
         ["id","1"],
         ["target-id","Remote target"],
         ["frame",[["level","0"],["addr","0x0800358c"],["func","t"],["args",[]],
         ["file","C:/Projects/test/src/майн.c"],  - unescaping and decoding as utf-8 give correct filename
         ["fullname","C:\\Projects\\test\\src\\����.c"],  - unescaping and decoding as **utf-8** give wrong filename 
         ["line","2"],
         ["arch","armv3m"]
        ]],
       ["state","stopped"]
       ]]]
    ]}}
```
Sadly, "fullname" prop is used in UI so after substitution and incorrect decoding UI is unable to show correct file.

Rule of thumb: all relevant filenames inside gdb **must** be in windows encoding (winXXXX) only.
Filename sources for gdb are:
 - CLI (use correct encoding)
 - sym table (use substitution).
 
HTH