Marus/cortex-debug #1014: RTT support for Black Magic Probe

sorrowed, 2024-05-05:
At the moment JLInk and OpenOCD support Segger RTT. It would be really nice to have RTT support for the BMP as the BMP firmware also supports it (although not by default, it is a compile time option).

Any pointer on how to get this done would also be welcome too so i myself may have a crack it it.

emlynkz, 2024-08-13:
You need to build the firmware yourself. Build it on main as per the instructions in the README.

Edit ``cross-file/native.ini`` to enable RTT support. You'll have to remove some targets.

haneefdm, 2024-08-13:
I don't own a BMP (or an MCU board that supports it) and I am not familiar with their SW/FW. If someone from the BMP side or yourself can help add support in Cortex-Debug, I can help merge it. Otherwise, this is a dead end.

You have our source code for both OpenOCD (bad implementation) and JLink (much better example) to serve as templates.

closed 2024-08-13 by haneefdm