Marus/cortex-debug #1015: SEGGER rtt connection ended immediately after lunch the cortex-debug

YinGithub, 2024-05-11:
[image]
Jlink version 6.30
rtt related conifg:
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "inputmode":"none",
                "decoders": [
                    {
                        "port": 0,
                        "type": "console",
                    }
                ]
            },
The  rtt connection lost directly after lunch the debug,
However, it appears stable for jlink 7.XX

PhilippHaefele, 2025-01-13:
Hi @YinGithub,

is this still an issue for you, as in the meanwhile we're at J-Link V8.xx and you already mentioned that V7.xx was working?

Best regards
Philipp

haneefdm, 2025-01-17:
6.30x is from Jan 2018. Closing this issue.

closed 2025-01-17 by haneefdm