Marus/cortex-debug #1022: Peripherals, registers, memory and disassemble all read "There is no data provider registered that can provide view data." when using the attached SVD file.

AndrewCapon, 2024-05-30:
**Describe the bug**

Peripherals, registers, memory and disassemble all read "There is no data provider registered that can provide view data." when using the attached SVD file.

This SVD file works fine in STM32CubeIde.

**To Reproduce**

Start debugger with stlink attached to a stm32h725
Set breakpoint, look at windows.

**Expected behavior**

[comment]: <> A clear and concise description of what you expected to happen.

I Expect to see some data.
[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

Latest versions used.

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
{
  "version": "0.2.0",
  "configurations": [
      {
          "name": "Debug (OpenOCD)",
          "cwd": "${workspaceRoot}",
          "executable": "${workspaceRoot}/build/debug.elf",
          "request": "attach",
          "type": "cortex-debug",
          "servertype": "openocd",
          "interface": "swd",
          "device": "STM32H725IETx",
          "svdFile": "${workspaceRoot}/STM32H725.svd",
          "searchDir": ["/Volumes/X82/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.2.0.202401261111/resources/openocd/st_scripts"],
          "configFiles": [
              "${workspaceRoot}/nts-1_mkii.cfg"
          ],
          "postAttachCommands": [
            "dir ${workspaceRoot}",
            "c"
          ],
      }
  ]
}
 ```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
"configuration": {
    "name": "Debug (OpenOCD)",
    "cwd": "/Volumes/X82/Development/Korg/nts1mkiiExample",
    "executable": "/Volumes/X82/Development/Korg/nts1mkiiExample/build/debug.elf",
    "request": "attach",
    "type": "cortex-debug",
    "servertype": "openocd",
    "interface": "swd",
    "device": "STM32H725IETx",
    "svdFile": "/Volumes/X82/Development/Korg/nts1mkiiExample/STM32H725.svd",
    "searchDir": [
        "/Volumes/X82/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.2.0.202401261111/resources/openocd/st_scripts"
    ],
    "configFiles": [
        "/Volumes/X82/Development/Korg/nts1mkiiExample/nts-1_mkii.cfg"
[... 107 lines cut ...]
14-exec-continue --all
-> 14^running
-> *running,thread-id="all"
mi2.status = running
15-exec-interrupt
-> 15^done
-> ~"\nProgram"

Program
-> ~" received signal SIGINT, Interrupt.\n"
 received signal SIGINT, Interrupt.
-> ~"0x08028690 in ?? ()\n"
0x08028690 in ?? ()
-> *stopped,reason="signal-received",signal-name="SIGINT",signal-meaning="Interrupt",frame={addr="0x08028690",func="??",args=[],arch="armv7e-m"},thread-id="1",stopped-threads="all"
mi2.status = stopped
16-break-insert "/Volumes/X82/Development/Korg/nts1mkiiExample/NtsDebugOut.c:776"
-> 16^done,bkpt={number="1",type="breakpoint",disp="keep",enabled="y",addr="0x000150bc",func="DebugOut",file="NtsDebugOut.c",fullname="/Volumes/X82/Development/Korg/nts1mkiiExample/NtsDebugOut.c",line="778",thread-groups=["i1"],times="0",original-location="/Volumes/X82/Development/Korg/nts1mkiiExample/NtsDebugOut.c:776"}
17-exec-continue --all
-> 17^running
-> *running,thread-id="all"
mi2.status = running

```

[STM32H725.svd.zip](https://github.com/Marus/cortex-debug/files/15496018/STM32H725.svd.zip)

Silver-Ray, 2024-06-21:
same problem, have you solved it already?

AndrewCapon, 2024-06-21:
Nope, I just stopped using Cortex Debug and now use Eclipse which works.

haneefdm, 2024-06-21:
This message is from VSCode and not from us. From Googling, it appears to mean that the extension was improperly installed or was not getting activated fully. If you installed it manually, make sure all the dependencies are also installed before installing this extension. 

If you install via the VSCode GUI, it takes care of all the dependencies. See the full dependencies here

https://github.com/Marus/cortex-debug/blob/695fe9fd9b7ace3a7d9b929ece2ea235092cfd53/package.json#L3012

Note that svd/peripheral viewer is another extension and not part of Cortex-Debug.

AndrewCapon, 2024-06-21:
I had installed using the VSCode GUI.

haneefdm, 2024-06-21:
Did you get any (error) messages during installation or first launch? You can also go to the market place and install all the other extensions manually. I don't know what your setup is exactly. Docker for instance requires manually installing all dependencies when you build the docker image -- even if you install it using the GUI.

Uninstall this extension and any dependencies already installed and start over. Do not ignore any messages as you may not see those messages repeatedly.

AndrewCapon, 2024-06-21:
Not that I remember. I should also add Cortex Debug works with other SVD files, it is just the one I attached that doesn't work. As I said I'm using Eclipse now for this board. 

Maybe I should close this issue, it wasn't looked at for 3 weeks anyway :)

closed 2024-06-21 by AndrewCapon