Marus/cortex-debug #1024: Live watch not updating value on esp32s3 with esp-prog; "Warn : [esp32s3.cpu0] target not halted"

OptifySudarshanPatil, 2024-06-03:
**Describe the bug**
Greetings!
I've successfully tested live-watch with sam3x8e using JTAG early on.
On other hand, live-watch not updating variable with esp32s3 (unless breakpoint halt CPU)
Any missing configuration which i might be missing?

**Expected behavior**

live-watch variable should me updated with current value.

**Screenshots**
[image]

[image]

## Live watch variable updated after breakpoint halt

[image]

[comment]: <> If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version v1.12.1
 - OS: Windows 10 x64

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
{
            // more info at: https://github.com/Marus/cortex-debug/blob/master/package.json
            "name": "Attach",
            "type": "cortex-debug",
            "request": "attach", // attach instead of launch, because otherwise flash write is attempted, but fails
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/.pio/build/esp32-s3-devkitc-1/firmware.elf",
            "servertype": "openocd",
            "interface": "jtag",
            "svdFile": "${workspaceRoot}/.pio/build/esp32-s3-devkitc-1/esp32s3.svd",
            "toolchainPrefix": "xtensa-esp32-elf",
            "armToolchainPath": "C:/Users/jaysa/.platformio/packages/toolchain-xtensa-esp32s3/bin",
            "gdbPath": "C:/Users/user/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gdb.exe",
            // "openOCDPreConfigLaunchCommands": [
            //     "set ESP_RTOS none"
            // ],
            "serverpath": "C:/Users/user/.platformio/packages/tool-openocd-esp32/bin/openocd.exe",
            "configFiles": [
                "board/esp32s3-ftdi.cfg"
            ],
            "overrideAttachCommands": [
                "set remote hardware-watchpoint-limit 2",
                "mon halt",
                "flushregs"
            ],
            "overrideRestartCommands": [
                "mon reset halt",
                "flushregs",
                "c",
            ],
            "showDevDebugOutput": "raw",
            "liveWatch": {
                "enabled": true,
                "samplesPerSecond": 1
            }
        }
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[esp32s3.cpu0] requesting target halt and executing a soft reset
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] requesting target halt and executing a soft reset
[esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Register cache flushed.
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] target not halted
[esp32s3.cpu1] Target halted, PC=0x403740C0, debug_reason=00000000
Set GDB target to 'esp32s3.cpu1'
[esp32s3.cpu0] Target halted, PC=0x42028CD6, debug_reason=00000000

Thread
7 "IDLE0" received signal SIGINT, Interrupt.
[Switching to Thread 1070549328]
0x42028cd6 in esp_pm_impl_waiti () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:182
182	/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h: No such file or directory.
[esp32s3.cpu0] Target halted, PC=0x42028CD6, debug_reason=00000000
Set GDB target to 'esp32s3.cpu0'
[esp32s3.cpu1] Target halted, PC=0x4037D2C0, debug_reason=00000000

Thread
7 "IDLE0" received signal SIGINT, Interrupt.
0x42028cd6 in esp_pm_impl_waiti () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:182
182	in /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] target not halted
[esp32s3.cpu0] Target halted, PC=0x42028CD6, debug_reason=00000000
Set GDB target to 'esp32s3.cpu0'
[esp32s3.cpu1] Target halted, PC=0x4037DBF5, debug_reason=00000000

Thread
7 "IDLE0" received signal SIGINT, Interrupt.
0x42028cd6 in esp_pm_impl_waiti () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:182
182	in /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h
LiveGDB: [Inferior 1 (Remote target) detached]
LiveGDB: /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gdb/gdb/thread.c:95: internal-error: thread_info* inferior_thread(): Assertion `tp' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
LiveGDB: 
Quit this debugging session?
LiveGDB: (y or n) [answered Y; input not from terminal]
LiveGDB: 
This is a bug, please report it.
LiveGDB:   For instructions, see:
LiveGDB: <http://www.gnu.org/software/gdb/bugs/>.
LiveGDB: /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gdb/gdb/thread.c:95: internal-error: thread_info* inferior_thread(): Assertion `tp' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
LiveGDB: 
Create a core file of GDB?
LiveGDB: (y or n) [answered Y; input not from terminal]
[Inferior 1 (Remote target) detached]
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gdb/gdb/thread.c:95: internal-error: thread_info* inferior_thread(): Assertion `tp' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.

Quit this debugging session?
(y or n) [answered Y; input not from terminal]

This is a bug, please report it.
  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gdb/gdb/thread.c:95: internal-error: thread_info* inferior_thread(): Assertion `tp' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.

Create a core file of GDB?
(y or n) [answered Y; input not from terminal]
GDB session ended. exit-code: 3

```

diggit, 2024-06-06:
Hi,
live watch needs access to target memory while the target core is not halted (live memory access/background memory access/...). This is available at Arm Cortex-M cores and possibly some more.
ESP32-S3 XTensa LX7 core probably does not allow such memory access or your GDB server does not support it.

Cortex debug is only trying to ask for memory contents periodically. *I think*, You are out of luck here. Some form of logging would be the best option for you here.

OptifySudarshanPatil, 2024-06-06:
That make sense!

closed 2024-06-06 by OptifySudarshanPatil