Marus/cortex-debug #1026: Project status, is it still actively developed/maintained?

daantimmer, 2024-06-16:
I created an issue about 4mo ago. And noticed there was a severe lack of activity. 

Marus, the original author hasn't provided any work in a long time. And the only work that was done was by another individual who seems to have disappeared as well. 

The last commit was 4mo ago, and responses to (new) issues also seem to have stopped around that same time. 

It is slowly making me a bit nervous, as this is a highly dependent upon extension for any embedded development for vscode.

Marus, 2024-06-16:
I wish I had better news for you in terms of more active development - unfortunately I haven't had time to be involved myself in recent years. I don't actually work in the embedded space - so Cortex-Debug was primarily developed to support my hobbyist development - unfortunately I haven't been doing much in that area recently - which has limited the time I have to support and develop this project.

@haneefdm had been doing most of the maintenance in the last few years - but I haven't heard from him recently - so possible that they don't have time to deal with it on top of their other commitments. 

Am always happy to consider getting other contributors that would like to take on a more major role setup.

I will try to make some time to at least look at open PRs to see if I think they can be safely merged into the project and released.

haneefdm, 2024-06-16:
Please see #1027. I also updated a few PRs and still waiting for replies from the authors.

ssimek, 2024-10-29:
@Marus @haneefdm if you guys are still interested, I'm open to discussion about picking up the maintenance of this extension

closed 2025-01-17 by haneefdm