Marus/cortex-debug #1027: Status of this extension

haneefdm, 2024-06-16:
Hello all, Since there is not good way to distribute this kind of info, I am just using the Issue system. I know people are worried about the activity (or lack thereof) on this extension. Here is what is going on

* I have moved/relocated 2400 miles back in Dec 2023.
* But soon after, in Mar 2024, I became very ill and got hospitalized several times. I am doing better but staring at a long recovery ahead -- maybe several more months.
* **I have been looking at every issue that shows up**. But, most issues are either not urgent or are things users should solve for themselves. So many people don’t read our Wiki and other documents I am not able to do much hand-holding like I used to. I know some users pitch in to help others, but we need more volunteers.
* Please note that we rely on GDB and gdb-servers a LOT, so being able to debug those and read their output can really help out
* Please report real issues after you have done your due diligence to convince yourself

Thanks in advance for your understanding. I will have my family post something here if things turn worse and/or unable to support this extension due to other reasons.

haneefdm, 2024-06-17:
I will be making a release shortly where I can test the PR/change in my limited HW/SW setup or consider it safe to do otherwise.

PhilippHaefele, 2024-06-18:
> Hello all, Since there is not good way to distribute this kind of info, I am just using the Issue system. I know people are worried about the activity (or lack thereof) on this extension. Here is what is going on
> 
First of all thank you very much for your update

> 
> * I have moved/relocated 2400 miles back in Dec 2023.
> 
> * But soon after, in Mar 2024, I became very ill and got hospitalized several times. I am doing better but staring at a long recovery ahead -- maybe several more months.
> 

Best wishes for a speedy recovery!

> * **I have been looking at every issue that shows up**. But, most issues are either not urgent (IMHO) or are things users should solve for themselves. So many people don’t read our Wiki and other documents I am not able to do much hand-holding like I used to. I know some users pitch in to help others, but we need more volunteers.
> 
> * Please note that we rely on GDB and gdb-servers a LOT, so being able to debug those and read their output can really help out. If you see a message that GDB is crashing, then, some investigation is warranted.
> 
> * Please report real issues after you have done your due diligence to convince yourself
> 
> 
> 
> Thanks in advance for your understanding. I will have my family post something here if things turn worse and/or unable to support this extension due to other reasons.
> 
> 
> 
> @PhilippHaefele thanks you for your help. I wish there were more of you
> 

Thanks for that. Trying to do my best in my limited time

J-Rios, 2024-06-23:
Hi @haneefdm,

Just want to thanks a lot for all your work and contribution to this project, I use it every day and I really appretiate it.

Best wishes for you, hope you get better :)

bmcdonnell-fb, 2024-10-17:
@haneefdm, thanks for your work on this project! Best wishes for a speedy and full recovery.

> ... things users should solve for themselves. So many people don’t read our Wiki and other documents I am not able to do much hand-holding like I used to. I know some users pitch in to help others, but we need more volunteers.

Is there a forum for users to help each other? (e.g. with existing features, howto, etc.) Other than GitHub issues on this repo, where you'd presumably want to reduce that?

If not, could you create one? Possibly here, using [GitHub Discussions](https://docs.github.com/en/discussions)?

bmcdonnell-fb, 2024-10-17:
WRT my preceding forum suggestion, I imagine a place where I could make a post, "is there a way to do X?" Possible replies:

- "This is the way" &rarr; mark [SOLVED]
- "No..." &rarr; create GH Issue, feature request
- "It should already do that" &rarr; create GH Issue, bug report

haneefdm, 2025-01-16:
I am back (partially). Made a pre-release of some pending bug fixes and new functionality contributed by others.

I will look into [GitHub Discussions](https://docs.github.com/en/discussions)

https://github.com/Marus/cortex-debug/releases

What we push to the marketplace next will be a pre-release version 1.13.x. I will have far too many changes to call it a patch release of 1.12.x

Thank you all for being patient.