Marus/cortex-debug #1029: GDB session ended. but the debug toolbar won't close.

yueduz, 2024-07-03:
**Screenshots**
[image]

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (this extension):v1.12.1
 - OS:archlunux
 - GDB Version: [e.g. 1.11.1]
 - Compiler Toolchain Version:  arm-none-eabi 14.1.0 

**Please include `launch.json`**
```
{
    // 使用 IntelliSense 了解相关属性。 
    // 悬停以查看现有属性的描述。
    // 欲了解更多信息，请访问: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Debug Jlink",
            "type": "cortex-debug",
            "request": "launch",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/debug/AT32_Audio.elf",
            "servertype": "jlink",
            "device": "AT32F435CMU7",
            "interface": "swd",
            // "runToEntryPoint": "main",
            // "overrideLaunchCommands": [
            //     "monitor halt",
            //     "monitor reset",
            //     "-target-download",
            //     "-enable-pretty-printing"
            // ],
            // "overrideRestartCommands": [
            //     "monitor reset",
            //     "-enable-pretty-printing"
            // ],
            // "showDevDebugOutput": "raw" ,
            "svdFile": "${workspaceRoot}/AT32F435CMU7_WorkBench/project/AT32_IDE/SVD/AT32F435xx_v2.svd",
            "preLaunchTask": "build Debug", //先运行make任务
            "serverArgs": [
                "-nologtofile",
                "-s",
                "-vd",
                "-speed",
                "15000"
            ],
            "rttConfig": {
                "enabled": false,
                "address": "auto",
                // "clearSearch": false    // OpenOCD users may have to un-comment this
                "decoders": [
                    {
                        "port": 0,
                        "type": "console"
                    }
                ]
            }
        }
    ]
}
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
[... 265 lines cut ...]
Read 4 bytes @ address 0x080078DA (Data = 0xF003691B)
Reading register 'msp' = 0x2005FFF8
Reading register 'psp' = 0x00000000
Read 4 bytes @ address 0x0800A4E6 (Data = 0x6430E7FB)
Reading 64 bytes @ address 0x2005FFC0
GDB closed TCP/IP connection (Socket 12)
Restoring target state and closing J-Link connection...
Shutting down...
[2024-07-03T01:42:59.498Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

```
Program
 received signal SIGTRAP, Trace/breakpoint trap.
delay_ms (nms=0x0, nms@entry=0x2) at /home/lei/at32-work-bench/AT32_Audio/AT32F435CMU7_WorkBench/mylib/delay/delay.c:59
59	    } while ((temp & 0x01) && !(temp & (1 << 16)));
Resetting target

Program
 received signal SIGTRAP, Trace/breakpoint trap.
0x080078da in delay_ms (nms=0x0, nms@entry=0x2) at /home/lei/at32-work-bench/AT32_Audio/AT32F435CMU7_WorkBench/mylib/delay/delay.c:58
58	      temp = SysTick->CTRL;
GDB session ended. exit-code: 0
```

haneefdm, 2024-07-03:
> Please enable debug output in your launch.json ("showDevDebugOutput": "raw"). It this is too large, please attach it as a file

See, you did not do that, so I have nothing to look at. Nothing changed on our end in almost six months. Did JLink change? Did you try rolling back a few versions. I really doubt JLink caused an issue but, did you look at what could have gone wrong?

The toolbar does not change state because VSCode (it controls that toolbar) thinks the debugger is still running. You also did not say what you did to get the debugger into this state. How am I supposed to reproduce/debug this?

I'm closing this until you give me a better description.

RolfNoot, 2024-07-05:
Adding some info here, experiencing the same issue since the last update (probably with 1.12, but not 100% sure).

Replication of the issue is easy: 

1. start debug
2. wait at main breakpoint
3. press `restart`
4. press `stop/disconnect`: GDB session ends but toolbar remains

See video:
https://github.com/Marus/cortex-debug/assets/44534542/dc8668cf-562b-49d8-b0e3-f70b4f5eb5e5

See debug info:
[terminal_output.txt](https://github.com/user-attachments/files/16108588/terminal_output.txt)
(not much to see here as the issue happens AFTER the debug session ends).

haneefdm, 2024-07-05:
[image]

That indicates a crash in JS code. If you click on that popup, you may see more info on the source or the problem.

In your system temp directory (indicated by $TMPDIR on Mac/Linux and %TEMP% env. var on Windows), hopefully in that directory there is a filed called `cortex-debug-server-exiting.log`. Please use your native shell to find this file and not something like cygwin/msys. It may contain some useful info as well. I only need the last 100 lines from this file. This file is created by a non GUI backed program and may indicate it crashing instead of the frontend GUI. The backend seems to have exited cleanly, but you never know. Because VSCode is still waiting for the backend to quit which is why the toolbar is in the wrong state.

haneefdm, 2024-07-05:
Please see https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood on how a typical debugger is put together.

RolfNoot, 2024-07-05:
This is the log file
[cortex-debug-server-exiting.log](https://github.com/user-attachments/files/16111893/cortex-debug-server-exiting.log)

This is the output of the Developer Tools console as soon as the server disconnects but the toolbar remains:
[image]

I am also debugging your extension:
[image]

Let me know if you need something more.

haneefdm, 2024-07-05:
I am on holiday (Independence Day in US). Besides, I don't have a board yet. Expecting one soon. Get back to this later.

It is one thing for gdb to exit, but another for us to exit because there is other cleanup that has to occur. It appears that messages are being generated even though the debugger has crashed or destroyed. Need to go up the stack to see what is going on.

@RolfNoot Have you read my Wiki page I attached above? Of course, you are the expert!

One of the problems is that the Heisenberg principle comes in by putting breakpoints. Especially while ending the session. Your best bet is to put temporary debug messages in suspected areas like I tried to do with the `cortex-debug-server-exiting.log` file. I created the `ServerConsoleLog(...)` to help me.

RolfNoot, 2024-07-05:
Hi @haneefdm,

Yes, I've read the wiki. I understand the issues you're facing and that it's even harder to pinpoint the eventual error, especially with 'vague' or limited user comments. I always feel that when the issue is to be reproduced, the solution is near.

Yes putting breakpoints can confuse even more and may completely break the functionality, especially when dealing with multithreaded designs. I mean if you want me to add anything to the log or help, I am here. 

We're dealing with users as well and it's not always easy to help especially when there's not much information.

I am willing to send you a Onethinx LoRaWAN kit which contains PSoC6 if it may help speed up investigations.

Enjoy the free day!

haneefdm, 2024-07-05:
I have a bad feeling about this. It appears that it is VSCode that is totally confused.

Over the last 5+ years, the definition and behavior of a 'Restart' kept changing. Maybe it changed again. At one point they did not trigger preLaunchTask on a Restart. Then they did a preLaunchTask. Then reverted back. Then reverted that decision. No idea what else changed with the Restart. The `preLaunchTask` is just one example.

A few years ago, we got tired of this and added a 'Reset' button because....

* In a lot of cases users did not want a re-build because they could take a LONG time even if nothing changed
* The functionality they desired was actually just a 'lets start from the beginning' without compiling or programming.

Thus, the Reset is fast and what most embedded developers wanted.

It appears from the logs that we have done everything we are supposed to do for a 'Restart' and Disconnect/Stop. But maybe their expectations changed. Need to focus on 'Restart' because it appears that if this was never called, it seems to work fine. Need to find out how debuggers are supposed to respond to a 'Restart' request. Rarely, documented so we have to do this by trial and error and looking at VSCode source. Perhaps play with cppdbg or some other debugger.

See also https://community.platformio.org/t/debugging-why-does-the-green-restart-icon-always-rebuild-entire-project/27410

haneefdm, 2024-07-05:
Fixed it. VSCode was seriously messed up. I found many debuggers have removed the 'Restart' functionality from their adapters. Instead, VSCode not totally restarts from scratch almost like you did a 'Stop' followed by a 'Start'. But I don't think it is re-reading the launch.json

Good riddance, because no one knew what the rules were.

But, now we have to remove *Restart* options from the launch.json (our package.json) because a Restart is now handled by VSCode and we have no involvement in it.

RolfNoot, 2024-07-05:
Good to hear!

I was just close, looking at the differences between 1.4.4 which works and 1.6.0 which doesn't. If you like I can check further to pinpoint the causing issue.

RolfNoot, 2024-07-06:
> Fixed it. VSCode was seriously messed up. I found many debuggers have removed the 'Restart' functionality from their adapters. Instead, VSCode not totally restarts from scratch almost like you did a 'Stop' followed by a 'Start'. But I don't think it is re-reading the launch.json

Confirmed, it works at my side. It is actually re-reading launch.json and applying changes if there are in launch.json.

Thanks so much, this issue was bothering me for some time!

haneefdm, 2024-07-06:
Cc: @AndriiLishchynskyi

Adrii, this may affect you. If you are still working on this, please see https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md

If there are general issues, could you give your feedback here? If you want to chat privately, you have my GitHub handle.

cc-caixf, 2024-08-12:
hello, does anyone konw how to fix it?

tim-hilt, 2024-09-19:
I tested it with the current master branch, looks good to me as well @haneefdm. I'm not deeply informed about the dev-progress, but seems save to release from my POV.

closed 2025-01-16 by haneefdm