Marus/cortex-debug #1036: ram .data section error (all FF) when debug (run to main) armclang build + jlink

137526291, 2024-08-05:
**Describe the bug**
RAM section .data inited error (all 0xFF) when run to main. eg: SystemCoreClock has init value 24Mhz but in debu session its value = 0xFFFFFFFF.  add memory view from ram addr 0x2000 0000 dumped to all 0xFF in .data section.

build by cmake + armclang , got elf output. debug use jlink.
when i use keil (ac6) debug, it works normal.
in linker option, i add :  --datacompressor=off

**To Reproduce**
Steps to reproduce the behavior:
1. Start debug session
2. add global var (with init value) into watch 
3. halt program and find value wrong.

**Expected behavior**
var should have expected value not all FF

**Screenshots**
[image]

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042)
 - OS: Windows 10
 - GDB Version: GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git]
 - Compiler Toolchain Version: [ARMClang ARM Compiler 6.16]

**Please include `launch.json`**

*Note: We are unlikely to look at the issue if you do not supply this*
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "cwd": "${workspaceFolder}",
            "executable": "./build_ac6/bms.elf",
            "name": "Debug with JLink",
            "request": "launch",
            "type": "cortex-debug",
            "device": "CBM8580M_AP_240KB",
            "runToEntryPoint": "__main",
            "showDevDebugOutput": "raw",
            "servertype": "jlink",
            "serverpath": "D:/JLink (2)/JLink/jlinkgdbservercl"        
         }
    ]
}
```

**Attach text from `Debug Console`**

Please enable debug output in your launch.json (`"showDevDebugOutput": "raw"`). It this is too large, please attach it as a file
```
-> 136^done,memory=[{begin="0x20000000",offset="0x00000000",end="0x20000200",contents="ffffffffffffffffffffffffffffffffc409ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"}]
137-var-update --thread 1 --frame 1 --all-values watch_8f520b08425785fad815cbf33e62a6944450e3fda40025c91cf87942242929f5
-> 137^done,changelist=[]
138-exec-next --thread 1
```

**Additional context**
Add any other context about the problem here.

137526291, 2024-08-05:
new debug console log add:
**-> 9^done
10-file-exec-and-symbols "D:/de_bms/build_ac6/bms.elf"
-> &"warning: Loadable section \"RW_IRAM1\" outside of ELF segments\n"
warning: Loadable section "RW_IRAM1" outside of ELF segments**

haneefdm, 2024-08-05:
Sorry to say, but this is between your linker script, linker and gdb. None of them belong to us. Can you debug from the command line? You see all the commands sent ...

Keil does not use gdb. Try to use gdb from the command line and see what you get

137526291, 2024-08-06:
> Sorry to say, but this is between your linker script, linker and gdb. None of them belong to us. Can you debug from the command line? You see all the commands sent ...
> 
> Keil does not use gdb. Try to use gdb from the command line and see what you get

thanks for your suggest! i am trying learn how to debug use gdb in cmd.
its there any possiable cortex debug parse elf have some problem?

haneefdm, 2024-08-06:
We don't parse any ELF files. I am closing this. If you find an issue with cortex-debug, please ask to reopen this issue. If you cannot debug with gdb, neither can we.

closed 2024-08-06 by haneefdm

dungChiBinh, 2024-09-28:
Hi @haneefdm,

I am using Cortex-Debug and OpenOCD to debug stm32f4-discovery board. and I faced the similar issue.

Here is how to reproduce the issue:

``` main.c
uint32_t number_of_element = 5;
uint32_t init_array[5] = {0x11111111, 0x22222222, 0x33333333, 0x44444444, 0x55555555};
const uint32_t const_init_array[5] = {0x55555555, 0x66666666, 0x77777777, 0x88888888, 0x99999999};
uint32_t noinit_array[5];

int main(void)
{
    uint32_t attemp = 0;
    uint32_t local_array[5] = {0xAAAAAAAA, 0xBBBBBBBB, 0xCCCCCCCC, 0xDDDDDDDD, 0xEEEEEEEE};

    for (int i = 0; i < number_of_element; i++)
    {
        noinit_array[i] = local_array[i];
        init_array[i] = const_init_array[i] + 1;
    }

    return 0;
}
```

Compile the code with ArmClang 6.22, then debug the application using Cortex-Debug and OpenOCD.

Map file;
```
Memory Map of the image

  Image Entry point : 0x0800018d

  Load Region LR_ROM (Base: 0x08000000, Size: 0x000002e0, Max: 0x00010000, ABSOLUTE)

    Execution Region ER_ROM (Exec base: 0x08000000, Load base: 0x08000000, Size: 0x000002c8, Max: 0x00010000, ABSOLUTE)

    Exec Addr    Load Addr    Size         Type   Attr      Idx    E Section Name        Object

    0x08000000   0x08000000   0x00000188   Data   RO           27    RESET               startup.o
    0x08000188   0x08000188   0x00000004   Code   RO           21    .text.Default_Handler  startup.o
    0x0800018c   0x0800018c   0x00000088   Code   RO           23  * .text.Reset_Handler  startup.o
    0x08000214   0x08000214   0x0000008a   Code   RO            6    .text.main          main.o
    0x0800029e   0x0800029e   0x00000002   PAD
    0x080002a0   0x080002a0   0x00000014   Data   RO           11    .rodata..L__const.main.local_array  main.o
    0x080002b4   0x080002b4   0x00000014   Data   RO           10    .rodata.const_init_array  main.o

    Execution Region RW_RAM (Exec base: 0x20000000, Load base: 0x080002c8, Size: 0x0000002c, Max: 0x0001f200, ABSOLUTE)

    Exec Addr    Load Addr    Size         Type   Attr      Idx    E Section Name        Object

    0x20000000   0x080002c8   0x00000014   Data   RW            9    .data.init_array    main.o
    0x20000014   0x080002dc   0x00000004   Data   RW            8    .data.number_of_element  main.o
    0x20000018        -       0x00000014   Zero   RW           12    .bss.noinit_array   main.o
```

Issue happens when debug to main function (after Reset_Handler has copied data from load_data section for RAM .data section), all the .data section filled with 0xFF. 
[image]

In my example here, I expect that variable **number_of_element** should be 5 and **init_array** should be {0x11111111, 0x22222222, 0x33333333, 0x44444444, 0x55555555};

Checking the load_data section of the two variables at address **0x080002c8** as mentioned in Map file
[image]
It is actually all 0xFF for this section, this is why the variables in .data section are initialized as 0xFF

But I have debug using arm-none-eabi-gdb and OpenOCD:
load the image into the MCU. reset and halt the MCU. dump data at load_data section address (0x080002c8)
[image]
It is actually the correct data. 

I also do other checking which I flash the app into the MCU without debugging, I add some more code into the for loop to blink an LED, and the LED blinks exactly 5 times

That means the issue only happens when debugging, somehow the flash memory fill with all 0xFF except for .text and .rodata
This causes the code still run but the initialize variables (global and static) are all set to 0xFF.

I know there are only you and Marus to handle a lot of stuff, I am really sorry for bothering you guys of my question.

I have tried a lot to search for a solution but I found nothing. I hope that you can check out this issue or give me some advise.

For more information, I tried with arm-none-eabi-gcc to compile the same code, It debugs/runs perfectly. (the issue does not happen with arm-none-eabi-gcc compiler)

Thank you very much.

Hi @137526291.
Did you find out the root cause and solution for this issue, I realy appriciated if you can share. Thank you.

haneefdm, 2024-09-28:
Btw, did you enter the same gdb command in the VSCode debug console? Enter `monitor mdw 0x000002c8 8`.

monitor commands are gdb-server specific so instead we use gdb commands to do the same. The command `x/w 0x000002c8` I forget exact syntax. https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_55.html

Both should give you similar results. The memory viewer ends up using gdb commands exclusively.

Now, my question is, are you actually debugging/flashing the same executable (elf vs axf) from the command line vs via Cortex-Debug. Since we don't touch your images or even look at them there is no mechanism for us to alter the behavior of your FW. You can see every command send to gdb and its response. See if you can spot something wrong.

haneefdm, 2024-09-28:
In case you didn't know, you can enter any gdb command from the bottom of the Debug Console
[image]

dungChiBinh, 2024-09-28:
Hi @haneefdm ,

Thank you for respond, I use the same executable file for all my checking above.
I just tried with the DEBUG CONSOLE
1. start debug and halt the CPU at Reset_Handler entry
2. dump the memory at load_data section using "monitor mdw 0x000002c8 8"
-> All 0xFF there as in the memory view.

I enable "showDevDebugOutput": "raw" in lauch.json and I got this
[image]

Then I try to re-flash the executable file (/workspaces/Project/embedded/armclang/memory_layout/build/memory_layout.axf which is the same file has been loaded at the begining of this debug section)

after reload it, I dump the memory and got the correct data
[image]

Reload the memory view, I got the correct data as well
[image]

I found that there was a warning at the first load of the debug section,
[image]

do you have any idea why loading the executable file manually makes it work, and what causes the warning -> &"warning: Loadable section \"RW_RAM\" outside of ELF segments\n" .  I really want to investigate more on the warning but I don't know where and how to start.

Thank you very much.

haneefdm, 2024-09-28:
I think you need to submit the whole debug console. Not just snippets/screenshots. I can't tell what happened before.

About the gdb error, no idea what you are seeing. Also, I need to see you complete manual gdb session as to what you are doing different from that Cortex-Debug is doing. The process simple .... halt, program, halt, reset, run, etc. and wait for a breakpoint to hit.
```
start debug and halt the CPU at Reset_Handler entry
dump the memory at load_data section using "monitor mdw 0x000002c8 8"
-> All 0xFF there as in the memory view.
```
This tells me again, that we are not doing the same thing with and without cortex-debug. So, cortex-debug is showing you the right results and so is gdb.

dungChiBinh, 2024-09-28:
Sorry, I attached the whole debug console log. 
[Cortex-Debug VSCode debugger extens.txt](https://github.com/user-attachments/files/17175718/Cortex-Debug.VSCode.debugger.extens.txt)

haneefdm, 2024-09-29:
Just a guess. It appears that gdb is not understanding the image created by Cang/ARM. Which may be why it is giving you get that  warning
```
10-file-exec-and-symbols "/workspaces/Project/embedded/armclang/memory_layout/build/memory_layout.axf"
-> &"warning: Loadable section \"RW_RAM\" outside of ELF segments\n"
warning: Loadable section "RW_RAM" outside of ELF segments
-> 10^done
```

You were using an alternate method to flash the device. You can disable ours and use your method ... when things don't work with gdb, that is a bad sign and should be reported to ARM. Anyways, as a workaround could you to add the following to your launch.json
```JSON
    "overrideLaunchCommands": [
         "monitor reset halt",
         "monitor flash write_image erase your-file-name",
         "monitor reset halt",
    ]
```

Remember that there are many ways of overriding what Cortex-Debug does by default. We cannot guess every possible (millions) way a vendor or user has created a program/debug flow. Please read our docs

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md
https://github.com/Marus/cortex-debug/wiki/Cortex-Debug-Under-the-hood

In this case, the compiler/your-linker-script did not produce a gdb friendly executable. But, that is just a guess

dungChiBinh, 2024-09-29:
Hi @haneefdm, 
overrideLaunchCommands did solve my issue.

Thank you very very much.

137526291, 2024-12-02:
hi @dungChiBinh  for my last struggle trying and limited capabilities (dont know how to use gdb) ， i just can find the clang built elf was downloaded uncompletely。 （like fw.bin size converted from fw.elf is 1024, but just 987 byte downloaded , see from debug output.) that is where i stopped and use gcc as compiler instead. 
btw this plugin works with vscode is so nice thinks to all authors and that is why from first begining i give up use keil.
thinks to you guys all !  your shared method to override, i think it can fix my problem also.!
love from china