Marus/cortex-debug #1044: Stop button does not close Debug toolbar

SashaHalytska, 2024-09-16:
**Describe the bug**
ype: Bug

I am using Cortex-Debug extension to debug Cortex-M based microcontroller's firmware:
https://github.com/Marus/cortex-debug/tree/master
Starting from the VSCode 1.86.2 I am observing an issue with the Stop button not closing the Debug toolbar. The issue reproducible only if the Restart button was pressed before Stop button.

This was not happening prior to 1.86.2.

The last version with which I see the issue 1.93.1

VS Code version: Code 1.93.1
OS version: All: Windows 11, Mac 14, Ubuntu 22

**Expected behavior**
Debug toolbar with buttons doesn't exist

**Actual Result**:
Debug toolbar with buttons exists, some of buttons even clickable
It looks like this

**Screenshots**

![image]
[image]

**Environment (please complete the following information):**

[comment]: <> Whenever possible, please make sure you are using the latest versions of VSCode and our extension

 - Cortex-Debug Version (v 1.12.1)

Add any other context about the problem here.

diggit, 2024-11-12:
Yes, this is happening on all my machines (ArchLinux, WSL, Windows) for several months and it's a annoying. It's difficult to start the new debug session afterwards. In my case, this happens when I restart the target at least once. If I never use restart button, I am usually fine.
It looks like vscode itself thinks, debug session is still active, but cortex debug already stopped GSB server and GDB itself.

haneefdm, 2024-11-12:
@SashaHalytska, are you building cortex-debug from source? I ask because you have a link to our repo. I don't know when I will get a chance to release this --- there are other changes being tested.

In the unreleased version sitting in our repo is a fix/workaround for the Restart button.

See also #1029

dancollins, 2025-01-15:
I am also experiencing this issue - and it seems like you have fixed it in later versions of the code. Would you say the preferred way for users to use this extension is compiling from the repo, rather than using the marketplace? For me, this issue is a little annoying and that's it - everything else seems to work great. I have trouble getting Zephyr RTOS threads recognised - but that appears to be an issue with SEGGER/JLink/GDB rather than this extension.

I have the "latest" extension (1.12.1), but I can't figure out how that relates back to this github repo. It states it was released 2023-09-14, however I don't see that in the list of github releases. I'm not familiar with how VS Code extensions are distributed - perhaps it's a manual task where you push compiled data to Microsoft?

In any case, I'd love to run with any important new fixes/features and if I need to compile locally to do so then I will!

haneefdm, 2025-01-15:
Feel free to compile from the repo but that is not my intent. I have been out of commission for a while and just now coming back online. We will have a pre-release distributed by the marketplace (if you subscribe to pre releases)  out shortly but I cannot promise a date.

We also provide pre-pre-releases which are available as I deem fit. There is a new one I just uploaded

https://github.com/Marus/cortex-debug/releases

If there are still issues, please report it here or else, I am closing this issue. Never heard back from @SashaHalytska either

jnz86, 2025-01-18:
Same issue, been terribly annoying. Greatly anticipating a fix.

I’d ask that more “official” versions be pushed in at least the pre-channel. We’re using docker images and dealing with it outside of the automated Vs Code extension system is not sustainable for us.

haneefdm, 2025-01-18:
> Same issue, been terribly annoying. Greatly anticipating a fix.

@jnz86   So, you are saying the issue isn't been fixed? Maybe what you are seeing is different from what others are seeing. Please help me duplicate it. There are thousands of users using Openocd but only a few have reported this. I could duplicate this in conjunction with the restart button. See #1029

> I have the "latest" extension (1.12.1), but I can't figure out how that relates back to this github repo. It states it was released 2023-09-14, however I don't see that in the list of github releases.

MS/VSCode has a market place (like an app store) for extensions. Only those extensions that are published there are pushed out to users. Periodiacally, we put unofficial releases on github for experimental purposes The official ones in the Market place are also the ones displayed in VSCode Extensions tab

https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug

Click on Version History in the page above. Yes, depending on your TZ it was released on Sep 13, 2023

haneefdm, 2025-01-18:
Can anyone confirm or deny that this issue has been addressed. If not fixed, can you help me delicate it. What you are seeing may not be caused for the same reason as someone else.

Please note that this issue (#1029) was caused by a VSCode release breaking something and only happened if you ever used the 'Restart' Button. 'Reset' but which we own was fine.

dancollins, 2025-01-18:
I believe this was addressed in your pre-release - it was for me - and I believe your other update fixed the Windows bug that was introduced.

Do you need me to test anything else so you can start closing off issues? I have a SEGGER JLink Pro, nRF9151 and Zephyr RTOS and can also run on STM32 (though I think that wont make any difference to you as that's purely a SEGGER difference).

dancollins, 2025-01-18:
Oh I should note I usually use the restart button which, I understand, you're not a fan of - but it does work! It's just a habit as I've used the debug toolbar across multiple IDEs and that's the button to use haha.

haneefdm, 2025-01-18:
> I believe this was addressed in your pre-release

Thanks for confirming.

> Oh I should note I usually use the restart button which, I understand, you're not a fan of - but it does work! It's just a habit as I've used the debug toolbar across multiple IDEs and that's the button to use haha.

You got me wrong. It is not that I am not a fan of it. What it does changes every few months and MS owns it. I have no control over what it does before, during or after. When we first started the project, what it did was simple and made sense. Over time it kept changing (and sometimes broke our extension) and I got tired of reverse-engineering what it actually did. So, I now tell VSCode that I don't support Restart so it treats it like a stop followed by an optional build (`preLaunchTask`) and then a start from scratch. That is fine for most people.

The Reset button we added was due to MS changing their mind every few months and to avoid a build+program.

When I upload `pre4` today or tomorrow, cortex-debug now supports setting values in the Watch panel. MS finally added the API and the UI element to do this.

Closing this **again** until someone proves **with data** that it is still a problem with the pre-releases.

closed 2025-01-18 by haneefdm

dancollins, 2025-01-18:
Love your work, mate. Thanks for the awesome extension.

Your described restart behaviour is what I would expect it to do - glad you found a workaround for constantly changing requirements from MS.