Marus/cortex-debug #1046: Cannot start debug

GlowingBrick, 2024-09-24:
here's my launch.json

> {
>     // Use IntelliSense to learn about possible attributes.
>     // Hover to view descriptions of existing attributes.
>     // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
>     "version": "0.2.0",
>     "configurations": [
>         {
>             "name": "Cortex Debug",
>             "cwd": "${workspaceRoot}",
>             "executable": "${workspaceRoot}/build/${workspaceFolderBasename}.elf",
>             "request": "launch",
>             "type": "cortex-debug",
>             "servertype": "openocd",
>             "showDevDebugOutput": "raw",
>             "interface": "swd",
>             "configFiles": [
>                 "interface/stlink-v2.cfg",
>                 "target/stm32f1x.cfg"
>             ],
>             "svdFile": "STM32F103x.svd",
>             "preLaunchTask": "build"
>         }
>     ]
> }
And my tasks.json

> 
> {
>     // See https://go.microsoft.com/fwlink/?LinkId=733558
>     // for the documentation about the tasks.json format
>     "version": "2.0.0",
>     "tasks": [
>         {
>             "label": "cmake configure",
>             "type": "shell",
>             "command": "cmake",
>             "args": [
>                 "-DCMAKE_BUILD_TYPE=Debug",
>                 ".."
>             ],
>             "group": {
>                 "kind": "build",
>                 "isDefault": true
>             },
>             "presentation": {
>                 "reveal": "always"
>             },
>             "problemMatcher": [
>                 "$gcc"
>             ],
>             "options": {
>                 "cwd": "${workspaceFolder}/build"
>             }
>         },
>         {
>             "label": "build",
>             "type": "shell",
>             "command": "make",
>             "group": {
>                 "kind": "build",
>                 "isDefault": true
>             },
>             "presentation": {
>                 "reveal": "always"
>             },
>             "problemMatcher": [
>                 "$gcc"
>             ],
>             "options": {
>                 "cwd": "${workspaceFolder}/build"
>             },
>             "dependsOn": "cmake configure"
>         },
>         {
>             "label": "download",
>             "type": "shell",
>             "command": "openocd",
>             "args": [
>                 "-f",
>                 "interface/stlink-v2.cfg",
>                 "-f",
>                 "target/stm32f1x.cfg",
>                 "-c",
>                 "program ${workspaceRoot}/build/${workspaceFolderBasename}.elf verify reset exit"
>             ],
>             "group": "build"
>         }
>     ]
> }

when I tried to debug ,I can see it building but cannot start debug and download,and no errors,just no respond.

Here's the output

> [proc] 执行命令: /opt/st/stm32cubeclt_1.15.1/CMake/bin/cmake --version
> [proc] 执行命令: /opt/st/stm32cubeclt_1.15.1/CMake/bin/cmake -E capabilities
> [kit] 已成功从 /home/brick/.local/share/CMakeTools/cmake-tools-kits.json 加载 15 工具包
> [presetController] 读取并验证预设“文件 /mnt/drive-d/code/embedded_dev/STM32/uart-test/CMakePresets.json”
> [presetController] 已成功针对预设架构验证 /mnt/drive-d/code/embedded_dev/STM32/uart-test/CMakePresets.json
> [presetController] 正在展开预设文件 /mnt/drive-d/code/embedded_dev/STM32/uart-test/CMakePresets.json
> [proc] 执行命令: uname -s
> [presetController] 已成功展开预设文件 /mnt/drive-d/code/embedded_dev/STM32/uart-test/CMakePresets.json
> [main] 正在配置项目: uart-test 
> [proc] 执行命令: /opt/st/stm32cubeclt_1.15.1/CMake/bin/cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE=/mnt/drive-d/code/embedded_dev/STM32/uart-test/cmake/gcc-arm-none-eabi.cmake -S/mnt/drive-d/code/embedded_dev/STM32/uart-test -B/mnt/drive-d/code/embedded_dev/STM32/uart-test/build/Debug -G Ninja
> [cmake] Build type: Debug
> [cmake] -- Configuring done (0.1s)
> [cmake] -- Generating done (0.0s)
> [cmake] -- Build files have been written to: /mnt/drive-d/code/embedded_dev/STM32/uart-test/build/Debug
> 

and the debug console

> [proc] 执行命令: /opt/st/stm32cubeclt_1.15.1/CMake/bin/cmake --version
> [proc] 执行命令: /opt/st/stm32cubeclt_1.15.1/CMake/bin/cmake -E capabilities
> [kit] 已成功从 /home/brick/.local/share/CMakeTools/cmake-tools-kits.json 加载 15 工具包
> [presetController] 读取并验证预设“文件 /mnt/drive-d/code/embedded_dev/STM32/uart-test/CMakePresets.json”
> [presetController] 已成功针对预设架构验证 /mnt/drive-d/code/embedded_dev/STM32/uart-test/CMakePresets.json
> [presetController] 正在展开预设文件 /mnt/drive-d/code/embedded_dev/STM32/uart-test/CMakePresets.json
> [proc] 执行命令: uname -s
> [presetController] 已成功展开预设文件 /mnt/drive-d/code/embedded_dev/STM32/uart-test/CMakePresets.json
> [main] 正在配置项目: uart-test 
> [proc] 执行命令: /opt/st/stm32cubeclt_1.15.1/CMake/bin/cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE=/mnt/drive-d/code/embedded_dev/STM32/uart-test/cmake/gcc-arm-none-eabi.cmake -S/mnt/drive-d/code/embedded_dev/STM32/uart-test -B/mnt/drive-d/code/embedded_dev/STM32/uart-test/build/Debug -G Ninja
> [cmake] Build type: Debug
> [cmake] -- Configuring done (0.1s)
> [cmake] -- Generating done (0.0s)
> [cmake] -- Build files have been written to: /mnt/drive-d/code/embedded_dev/STM32/uart-test/build/Debug
> 
What's wrong....It used to be running normally.
And openocd can run independently normally
:(

haneefdm, 2024-09-24:
I am not seeing anything related to cortex-debug in the debug console output. This seems to be your build output. Maybe your build is failing -- we do not handle builds. VSCode does

Closing this for lack of any actionable data....

closed 2024-09-24 by haneefdm

GlowingBrick, 2024-09-25:
> I am not seeing anything related to cortex-debug in the debug console output. This seems to be your build output. Maybe your build is failing -- we do not handle builds. VSCode does
> 
> Closing this for lack of any actionable data....

well,Because it has already been compiled before, cmake did not recompile it. Actually, the build was successful, I can manually download and run it:)

haneefdm, 2024-09-25:
Lets not mix build and debug. Remove the following
```
"preLaunchTask": "build"
```

We don't handle this. VSCode does and it has nothing to do with our debugger.

After removing the  `preLaunchTask` show me what is failing. I can't see a single line of text that is coming from. If I am not seeing output from Cortex-Debug and/or gdb, then I can't help

GlowingBrick, 2024-09-25:
> Lets not mix build and debug. Remove the following
>
> ```
> "preLaunchTask": "build"
> ```
>
> We don't handle this. VSCode does and it has nothing to do with our debugger.
>
> After removing the [...]

I tried it, but it seems like it just skipped the build and nothing happened
Sorry, I found that I pasted the wrong information earlier. The output in the debugging console is
```
"configuration": {
    "name": "Cortex Debug",
    "cwd": "/mnt/drive-d/code/embedded_dev/STM32/uart-test",
    "executable": "/mnt/drive-d/code/embedded_dev/STM32/uart-test/build/uart-test.elf",
    "request": "launch",
    "type": "cortex-debug",
    "servertype": "openocd",
    "showDevDebugOutput": "raw",
    "interface": "swd",
    "configFiles": [
        "interface/stlink-v2.cfg",
        "target/stm32f1x.cfg"
    ],
    "svdFile": "/mnt/drive-d/code/embedded_dev/STM32/uart-test/STM32F103x.svd",
    "__configurationTarget": 6,
    "gdbServerConsolePort": 55878,
    "pvtAvoidPorts": [],
    "chainedConfigurations": {
        "enabled": false
    },
    "debuggerArgs": [],
    "swoConfig": {
        "enabled": false,
        "decoders": [],
        "cpuFrequency": 0,
        "swoFrequency": 0,
        "source": "probe"
    },
    "rttConfig": {
        "enabled": false,
        "decoders": []
    },
    "graphConfig": [],
    "preLaunchCommands": [],
    "postLaunchCommands": [],
    "preAttachCommands": [],
    "postAttachCommands": [],
    "preRestartCommands": [],
    "postRestartCommands": [],
    "preResetCommands": [],
    "postResetCommands": [],
    "searchDir": [],
    "toolchainPrefix": "arm-none-eabi",
    "extensionPath": "/home/brick/.vscode/extensions/marus25.cortex-debug-1.12.1",
    "registerUseNaturalFormat": true,
    "variableUseNaturalFormat": true,
    "pvtVersion": "1.12.1",
    "__sessionId": "9d9a1b88-ba7d-451f-ad3f-c68d732f0228",
    "pvtShowDevDebugOutput": "raw"
}
Reading symbols from arm-none-eabi-objdump --syms -C -h -w /mnt/drive-d/code/embedded_dev/STM32/uart-test/build/uart-test.elf
Reading symbols from arm-none-eabi-nm --defined-only -S -l -C -p /mnt/drive-d/code/embedded_dev/STM32/uart-test/build/uart-test.elf
Launching GDB: arm-none-eabi-gdb -q --interpreter=mi2
1-gdb-version
```

haneefdm, 2024-09-25:
I have a feeling that your gdb is not properly installed. This happens when gdb is missing other things like python, curses, etc. Arms' version of gdb is notorious for making an incomplete package.

Make sure you can run gdb from the command line

GlowingBrick, 2024-09-25:
> I have a feeling that your gdb is not properly installed. This happens when gdb is missing other things like python, curses, etc. Arms' version of gdb is notorious for making an incomplete package.
> 
> Make sure you can run gdb from the command line

I checked arm non nr eabi gdb and it can be used normally. It can also be debugged using opeocd. 
Discovered an issue with gdb in stmcubeclt, as it is no longer usable due to system updates, and the project defaults to using it. 
Already resolved. Thank you very much for your help!